lvgl/lvgl #682: Inconsistently named enums

rreilink, 2018-12-30:
The following enum constants are not named consistently with their lv_xxx_t: (this was found using [this script](https://github.com/rreilink/pylvgl/blob/unify_with_mpy/check_decls.py)

```
lv_border_part_t :  LV_BORDER_NONE,LV_BORDER_BOTTOM,LV_BORDER_TOP,LV_BORDER_LEFT,LV_BORDER_RIGHT,LV_BORDER_FULL
lv_shadow_type_t :  LV_SHADOW_BOTTOM,LV_SHADOW_FULL
lv_design_mode_t :  LV_DESIGN_DRAW_MAIN,LV_DESIGN_DRAW_POST,LV_DESIGN_COVER_CHK
lv_anim_builtin_t :  LV_ANIM_NONE,LV_ANIM_FLOAT_TOP,LV_ANIM_FLOAT_LEFT,LV_ANIM_FLOAT_BOTTOM,LV_ANIM_FLOAT_RIGHT,LV_ANIM_GROW_H,LV_ANIM_GROW_V
lv_hal_indev_type_t :  LV_INDEV_TYPE_NONE,LV_INDEV_TYPE_POINTER,LV_INDEV_TYPE_KEYPAD,LV_INDEV_TYPE_BUTTON,LV_INDEV_TYPE_ENCODER
lv_label_long_mode_t :  LV_LABEL_LONG_EXPAND,LV_LABEL_LONG_BREAK,LV_LABEL_LONG_SCROLL,LV_LABEL_LONG_DOT,LV_LABEL_LONG_ROLL,LV_LABEL_LONG_CROP
lv_preloader_type_t :  LV_PRELOAD_TYPE_SPINNING_ARC
lv_cursor_type_t :  LV_CURSOR_NONE,LV_CURSOR_LINE,LV_CURSOR_BLOCK,LV_CURSOR_OUTLINE,LV_CURSOR_UNDERLINE,LV_CURSOR_HIDDEN
```

I am not yet sure whether (and if so, how) these inconsistencies should be fixed. However, they currently cause uncertainty as to how these enums should be called when generating bindings for another language, e.g. should one use:

`lvgl.BORDER.BOTTOM` or `lvgl.BORDER_PART.BOTTOM`

In case of the `lv_preloader_type_t`, I think the constant should be `LV_PRELOADER_TYPE_SPINNING_ARC`

kisvegabor, 2018-12-30:
These inconsistencies can be fixed in v6.0 because they might affect the backward compatibility in v5.3.
v5.3 will be released soon and the `dev-6.0` branch will be created where the fixes can be added. 

Until that, I'd prefer `lvgl.BORDER.BOTTOM`-like names. So the enums should be extracted from the definition itself. Is it okay for you?

kisvegabor, 2018-12-30:
I've just read your detailed questions/suggestions about naming in #557. I suggest discussing it there. I will answer there today as I can!

closed 2018-12-30 by kisvegabor