lvgl/lvgl #684: Image rotation support

doode, 2019-01-01:
I see activity on screen rotation, but wondering if there are plans to support rotating an image as would be needed to implement analog clock with fancy indicators etc... One option is to create multiple copies of the image and load/swap them at run time as needed, but this is coarse. Transforming the image data would be cleaner approach.

I looked at implementing a custom draw function for image objects, but there is certainly a lot of heavy lifting going on in the image object that will take some time to understand so as not to break anything.

Thanks

embeddedt, 2019-01-01:
@doode Doing rotations in increments of 90 degrees is simpler to implement because it is just some X and Y coordinate adjusting. That's what the screen rotation would do when implemented. Is that what you are trying to do? From the way you wrote your suggestion it looks like you want more accurate increments than that, which isn't something on the roadmap at this point as far as I know.

kisvegabor, 2019-01-01:
@doode @embeddedt 
Hopefully, image rotation will be supported in v6.0.  
My idea about implementation of image rotation is to have a `lv_img_src_rotate(src_in, buf_out, degree) function which creates a new, rotated image into RAM.  I was thinking about a render-time rotation too but:
1. It would be very difficult to implement because of the optimizations and complexity of drawing
2. It would be slow

For example, let's imagine a clock handle like this:
[image]

It has 200x200 size, on a 16-bit display it needs 200*200*2 = 78 kB which is a lot of RAM. But if the rotated image is encoded as indexed on 2 bits then  200*200 * (2/8) = 10kB which is, in my opinion, acceptable.

doode, 2019-01-01:
@kisvegabor @embeddedt

The lv_img_src_rotate function is the right approach since it will allow arbitrary angles of rotation. May want to consider the ability to allow for specifying the axis of rotation. Something like...

`lv_img_src_rotate(src_in, buf_out, degree, x,y) where x,y are points within the frame buffer describing the axis of rotation.

Looking forward to v6.0. Please advise if you need help testing it.

Thanks

kisvegabor, 2019-01-03:
@doode It's reasonable to specify the axis of the rotation but it changes the dimension of the image. For example, rotating a 10x100 image by 90-degree results in 100x10, or by 45 degrees in ~70x70. It makes the alignment quite difficult. 

Anyway, in this concept,  new and more advanced rotation function can be added in a modular way.

doode, 2019-01-03:
@kisvegabor, good point on the alignment complexity.  Image dimensions will change for all but square images that are rotated at any angle that is not a multiple of 90. 

From a naive point regarding object and layer management, what if images that are to be rotated adhere to alignment spec when they are first drawn (original image). Can the rotation api copy (move?) pixels to the target computed x,y within the frame buffer ignoring alignment? The new image will move within the frame buffer depending on selection of the rotation axis.

If alignment is critical, will imposing or creating constraints on the dimensions of an image that can be rotated help?  Can only rotate square images about its center for example. Maybe this constraint is imposed regardless to make things simpler?

doode, 2019-01-07:
OK, will start digging into this after the lib comes up on my target platform.

kisvegabor, 2019-01-08:
Great! Thank you!

Pablo2048, 2019-03-01:
Hi @doode . I want to ask how far did You get into implementing image rotation. Can I help You with something or I have to start from beginning?

manison, 2019-03-01:
I'm also interested in image transformations – specifically I need to mirror an image.
If you are thinking of image rotation in arbitrary angles then the generic [matrix transformation](https://en.wikipedia.org/wiki/Transformation_matrix#Examples_in_2D_computer_graphics) may be taken into account.

Then we can have something like:

```c
struct lv_transform_t {
    lv_coord_t a11, a12, a21, a22;
};

/*Main transform function*/
lv_img_dsc_t * lv_img_transform(void *src, const lv_transform_t *transform);

/*Predefined transformation*/
const lv_transform_t LV_TRANSFORM_MIRROR_X = { ... };
const lv_transform_t LV_TRANSFORM_MIRROR_Y = { ... };

/*Transformation matrix builder functions*/
void lv_transform_make_scale(lv_transform_t *transform, lv_coord_t scale_x, lv_coord_t scale_y);
void lv_transform_make_rotation(lv_transform_t *transform, int degree);
```

doode, 2019-03-01:
Hello Pablo

Unfortunately nothing implemented. 

Feel free to innovate. ;)

Norik

kisvegabor, 2019-03-01:
My idea is to use `lv_canvas` for this purpose because it already has a user-defined buffer and some useful features (e.g. set pixel with indexed color formats too). 

So IMO the most simple thing would be to add a function which can calculate the rotated image to the buffer of the canvas. For example: `lv_img_rotate(&image_original, buffer, angle)`;

Pablo2048, 2019-03-01:
So I will start (because my very little knowledge about LVGL) with simple nearest match rotation. `lv_draw_img.c/h` are my targets for now...

kisvegabor, 2019-03-01:
I suggest trying to work with simple buffers first. 

Just convert an image with the [converter](https://littlevgl.com/image-to-c-array) and rotate its buffer to a big array and replace the original image's data:
```c
lv_color_t rotated_buf[image_width * image_height];

rotate(original_image.data, image_width, image_height, rotated_buffer, angle);

original_image.data = rotated_buffer;      /*Replace the original image data with the rotated*/

lv_obj_t * img = lv_img_create(lv_scr_act(), NULL);
lv_img_set_src(img, &original_img);

```

Pablo2048, 2019-03-01:
Ok - great idea...

kisvegabor, 2019-03-02:
@Pablo2048 I'm working on better organizing issues and assigning people to them. Can I assign you to this one? After the basic implementation (with buffers) is ready I can inegrate it to LittlevGL.

Pablo2048, 2019-03-02:
Yes. Actually I'm getting familiar with SDL simulator to made development easier...

embeddedt, 2019-03-02:
@kisvegabor It looks like only contributors can be assigned to issues.

kisvegabor, 2019-03-03:
@Pablo2048 
> Yes. Actually I'm getting familiar with SDL simulator to made development easier...

Nice!

@embeddedt 
> It looks like only contributors can be assigned to issues.

Oh, then I assign myself.

Pablo2048, 2019-03-03:
So I have first result with the simplest nearest match rotation working:
[image]
this is lvgl logo rotated by 10 degree. Pivot is the center of the image, but it can be easily modified to any point if there is need for it IMHO. Non-existent source pixels are filled with pre-defined color (LV_COLOR_BLACK used in demo). Now I have to solve sin & cos computing without use of double/float math...

kisvegabor, 2019-03-04:
Awesome! :clap:  :clap:  :clap: 

What is the black border around the blueish squares?

Pablo2048, 2019-03-04:
To be honest I did not take a closer look. The image is just this one
[image]
I have added just an white background and converted by online tool. The code was also pretty easy:
```c
static void image_rotate(lv_color_t *src, int width, int height, int pivotx, int pivoty, lv_color_t *target, int angle, lv_color_t defaultBckgr)
{

  angle %= 360;
  double angleradians = angle * M_PI / 180.0;
  double sinma = sin(-angleradians);
  double cosma = cos(-angleradians);

  for (int x = 0; x < width; x++) {
    for (int y = 0; y < height; y++) {
      int xt = x - pivotx;
      int yt = y - pivoty;

      int xs = (int)round((cosma * xt - sinma * yt) + pivotx);
      int ys = (int)round((sinma * xt + cosma * yt) + pivoty);

      if (xs >= 0 && xs < width && ys >= 0 && ys < height) {
        target[(y * width) + x] = src[(ys * width) + xs]; /* set target pixel (x,y) to color at (xs,ys) */
      } else {
        target[(y * width) + x] = defaultBckgr; /* set target pixel (x,y) to default background */
      }
    }
  }
}
```

kisvegabor, 2019-03-04:
It seems the algorithm maps the source pixels to destination pixels. This way some destination pixels might be left out. Try to prefill the array with `LV_COLOR_RED` to see if really that is the case.

To avoid this usually, the destination pixels are mapped to the nearest source pixels.
https://stackoverflow.com/a/583801

Pablo2048, 2019-03-04:
:-) no - it travels over every destination pixel (x and y variable). It is nearest match algorithm...

kisvegabor, 2019-03-04:
Ohh really, it's clear from "target[(y * width) + x] = ..."

kisvegabor, 2019-03-04:
There is a small gradient on the edges but I can't anything in the algorithm which can do it.

[image]

Pablo2048, 2019-03-04:
Yes - I did the same right now (enlarge the pixels). When I return from work I check my converted image if https://pinta-project.com/pintaproject/pinta/ and his color fill can be the source of this...

Pablo2048, 2019-03-04:
So I do have an explanation for this - its because of Pinta background fill. See:
[image]
It's direct image output from Pinta after filling transparent background with white solid color. It seems like Pinta left some alpha channel artifacts in image (it can be seen clearly if you put this image in black background).

kisvegabor, 2019-03-05:
I see. I edited it with GIMP:
[image]

How does the rotation look like with this image?

Pablo2048, 2019-03-05:
Here it is:
[image]
[image]
I just changed default background color to LV_COLOR_RED to clarify...

embeddedt, 2019-03-05:
@kisvegabor @Pablo2048 It seems that we would need to do antialiasing or a similar effect on the rotated image to avoid this pixelation. But this is looking great so far!

Pablo2048, 2019-03-05:
I think that antialiasing can be substituted with bilinear interpolation, but this is way before me and my knowledge so far. Anyway this is my last result:
[image]
It is rotated using sin&cos table with 10 bit precision using only integer arithmetics (and no division - only shifts)...

Pablo2048, 2019-03-05:
So here is the result within all 360 degree rotation:
[image]
So I'm thinking that my next step is to use canvas as @kisvegabor suggested. Unfortunately I'm not sure that this can be used to rotate character glyphs (at least the antialiased ones)... Suggestions anyone?

kisvegabor, 2019-03-05:
Very nice!

I update the algorithm to make anti-aliasing. The quality is surpisingly good!
[image]

```c

void image_rotate(lv_color_t *src, int width, int height, int pivotx, int pivoty, lv_color_t *target, int angle, lv_color_t defaultBckgr)
{

  angle %= 360;
  double angleradians = angle * M_PI / 180.0;
  double sinma = sin(-angleradians);
  double cosma = cos(-angleradians);

  for (int x = 0; x < width; x++) {
    for (int y = 0; y < height; y++) {

      /*Get the ??? coordinates and upscale them*/
      int xt = (x - pivotx) * 256;
      int yt = (y - pivoty) * 256;

      /*Get the source pixel fro mteh upcaled image*/
      int xs = (int)((cosma * xt - sinma * yt) + pivotx * 256);
      int ys = (int)((sinma * xt + cosma * yt) + pivoty * 256);

      /*Get the integer part of teh source pixel*/
      int xs_int = xs >> 8;
      int ys_int = ys >> 8;

      /*Get the fractional part of teh source pixel*/
      int xs_fract = xs & 0xff;
      int ys_fract = ys & 0xff;

      /* If the fractional < 0x70 mix the sorce pixel with the left/top pixel
       * If the fractional > 0x90 mix the sorce pixel with the right/bottom pixel
       * In the 0x70..0x90 range use the unchanged source pixel */

      int xn;           /*x neightboor*/
      lv_opa_t xr;      /*x mix ratio*/
      if(xs_fract < 0x70) {
          xn = xs_int - 1;
          xr = xs_fract * 2;
      }
      else if(xs_fract > 0x90) {
          xn = xs_int + 1;
          xr = (0xFF - xs_fract) * 2;
      }
      else {
          xn = xs_int;
          xr = 0xFF;
      }

      /*Handle under/overflow*/
      if(xn >= width) xn = width - 1;
      else if(xn < 0) xn = 0;

      int yn;            /*y neightboor*/
      lv_opa_t yr;       /*y mix ratio*/
      if(ys_fract < 0x70) {
          yn = ys_int - 1;
          yr = ys_fract * 2;
      }
      else if(ys_fract > 0x90) {
          yn = ys_int + 1;
          yr = (0xFF - ys_fract) * 2;
      }
      else {
          yn = ys_int;
          yr = 0xFF;
      }

      /*Handle under/overflow*/
      if(yn >= height) yn = height - 1;
      else if(yn < 0) yn = 0;

      /*Get the mixture of the orignal source and the neightboor pixels in both directions*/
      lv_color_t x_dest = lv_color_mix(src[(ys_int * width) + xs_int], src[(ys_int * width) + xn], xr);
      lv_color_t y_dest = lv_color_mix(src[(ys_int * width) + xs_int], src[(yn * width) + xs_int], yr);

      if (xs_int >= 0 && xs_int < width && ys_int >= 0 && ys_int < height) {
          /*The result color as the avarage of the x/y mixed colors*/
        target[(y * width) + x] = lv_color_mix(x_dest, y_dest, LV_OPA_50);
      } else {
        target[(y * width) + x] = defaultBckgr; /* set target pixel (x,y) to default background */
      }
    }
  }
}
```

kisvegabor, 2019-03-05:
@Pablo2048 
> So I'm thinking that my next step is to use canvas as @kisvegabor suggested. 

I'll examine where we can add a good interface for it.

> Unfortunately I'm not sure that this can be used to rotate character glyphs (at least the antialiased ones)... Suggestions anyone?

If you store the letters as images you can rotate them. With the updated algorithm it looks quite well:
[image]

Just to make it clear for others too, the goal is this: https://forum.littlevgl.com/viewtopic.php?f=12&t=10

Pablo2048, 2019-03-06:
Wow - that is great. My integer math look like this:
```c
  angle = 360 - (angle % 360);

  int32_t sinma = SINTABLE[angle];
  int32_t cosma = SINTABLE[(angle + 90) % 360]; /* cos */

...

      int xs = ((cosma * xt - sinma * yt) >> 10) + pivotx;
      int ys = ((sinma * xt + cosma * yt) >> 10) + pivoty;
```
so if i reduce 10 bit sin table to just 8 bit (and try if it still work without significant errors), we can use just one division by 256.

Pablo2048, 2019-03-06:
I feel very stupid now :-( - I found right now that there is already the sine table in LVGL. So I change the math to use the `lv_trigo_sin()` instead...

kisvegabor, 2019-03-06:
I just wanted to suggest it! :) You are not stupid just know all parts of LittlevGL which are not a sin :)

I'm curiously waiting how the integer math will affect the quality.

Pablo2048, 2019-03-06:
So I end with this:
```c
static void image_rotate(lv_color_t *src, int32_t width, int32_t height, int32_t pivotx, int32_t pivoty, lv_color_t *target, int16_t angle, lv_color_t defaultBckgr)
{

  int32_t sinma = lv_trigo_sin(-angle);
  int32_t cosma = lv_trigo_sin(-angle + 90); /* cos */

  for (int32_t x = 0; x < width; x++) {
    for (int32_t y = 0; y < height; y++) {
      /*Get the target point relative coordinates to the pivot*/
      int32_t xt = x - pivotx;
      int32_t yt = y - pivoty;

      /*Get the source pixel from the upscaled image*/
      int32_t xs = ((cosma * xt - sinma * yt) >> (LV_TRIGO_SHIFT - 8)) + pivotx * 256;
      int32_t ys = ((sinma * xt + cosma * yt) >> (LV_TRIGO_SHIFT - 8)) + pivoty * 256;

      /*Get the integer part of the source pixel*/
      int xs_int = xs >> 8;
      int ys_int = ys >> 8;

      /*Get the fractional part of the source pixel*/
      int xs_fract = xs & 0xff;
      int ys_fract = ys & 0xff;

      /* If the fractional < 0x70 mix the sorce pixel with the left/top pixel
       * If the fractional > 0x90 mix the sorce pixel with the right/bottom pixel
       * In the 0x70..0x90 range use the unchanged source pixel */

      int xn;           /*x neightboor*/
      lv_opa_t xr;      /*x mix ratio*/
      if(xs_fract < 0x70) {
          xn = xs_int - 1;
          xr = xs_fract * 2;
      }
      else if(xs_fract > 0x90) {
          xn = xs_int + 1;
          xr = (0xFF - xs_fract) * 2;
      }
      else {
          xn = xs_int;
          xr = 0xFF;
      }

      /*Handle under/overflow*/
      if(xn >= width) xn = width - 1;
      else if(xn < 0) xn = 0;

      int yn;            /*y neightboor*/
      lv_opa_t yr;       /*y mix ratio*/
      if(ys_fract < 0x70) {
          yn = ys_int - 1;
          yr = ys_fract * 2;
      }
      else if(ys_fract > 0x90) {
          yn = ys_int + 1;
          yr = (0xFF - ys_fract) * 2;
      }
      else {
          yn = ys_int;
          yr = 0xFF;
      }

      /*Handle under/overflow*/
      if(yn >= height) yn = height - 1;
      else if(yn < 0) yn = 0;

      /*Get the mixture of the orignal source and the neightboor pixels in both directions*/
      lv_color_t x_dest = lv_color_mix(src[(ys_int * width) + xs_int], src[(ys_int * width) + xn], xr);
      lv_color_t y_dest = lv_color_mix(src[(ys_int * width) + xs_int], src[(yn * width) + xs_int], yr);

      if (xs_int >= 0 && xs_int < width && ys_int >= 0 && ys_int < height) {
          /*The result color as the average of the x/y mixed colors*/
        target[(y * width) + x] = lv_color_mix(x_dest, y_dest, LV_OPA_50);
      } else {
        target[(y * width) + x] = defaultBckgr; /* set target pixel (x,y) to default background */
      }
    }
  }
}
```
I've removed Your xt/yt multiplication because of lv_trigo_sin() naturality and it ends with this:
[image]
Which look very good IMHO...

kisvegabor, 2019-03-07:
Wow, that looks amazing! Still great quality with integer math too!

I've cheked the canvas object and it seems it can be simply used with `LV_IMG_CF_TRUE_COLOR` images.  
Just `target[(y * width) + x] = lv_color_mix(x_dest, y_dest, LV_OPA_50);` need to be replaced with  
`lv_canvas_set_px(canvas, x, y, lv_color_mix(x_dest, y_dest, LV_OPA_50))`

It becomes trickier when `LV_IMG_CF_INDEXED_1/2/4/8BIT` is used because indexed colors can't be mixed this way.

I'll make some experiments with it soon but I need to finish a few other things first.

Pablo2048, 2019-03-07:
Ok, I can do some minor speed enhancements, but I'll wait for Your advice on how to continue...

kisvegabor, 2019-03-20:
I updated the canvas object to support rotation. 

Here is the image I tested with: [logo_small.c.zip](https://github.com/littlevgl/lvgl/files/2988969/logo_small.c.zip)

And here is my test code:
```c
    /*Create a "source" canvas from an image*/
    LV_IMG_DECLARE(logo_small);
    lv_obj_t * can_src = lv_canvas_create(lv_scr_act(), NULL);
    lv_canvas_set_buffer(can_src, logo_small.data,
            logo_small.header.w, logo_small.header.h, LV_IMG_CF_TRUE_COLOR);
    lv_obj_set_hidden(can_src, true);          /*It doesn't need to be visible*/

    /*Create a "destination" canvas*/
    static lv_color_t can_dest_buf[200 * 200];
    lv_obj_t * can_dest = lv_canvas_create(lv_scr_act(), NULL);
    lv_canvas_set_buffer(can_dest, can_dest_buf, 200, 200, LV_IMG_CF_TRUE_COLOR);

    uint32_t angle = 0;
    while(1) {
        /* Periodically call the lv_task handler.
         * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        usleep(10*1000);

        /*Clear the canvas*/
        memset(can_dest_buf, 0xAA, sizeof(can_dest_buf));

        /*Rotate the source and store it on the  destination*/
        lv_canvas_rotate(can_dest, can_src, angle % 360,
                                            50, 50, /*Offset*/
                                            0, 50); /*Pivot*/
        angle++;
}
```

[image]

embeddedt, 2019-03-20:
@kisvegabor Nice! I think it would also be nice to have a simpler API that allows you to set the rotation without requiring two canvas objects; is this feasible? If so I would be happy to work on it.

Pablo2048, 2019-03-20:
@kisvegabor Great! So it seems like we can have image of clock hour hand very small. Guys who made an analog watchface with LVGL as an example first? ;-)

kisvegabor, 2019-03-21:
Quick test:
[image]

I was cheating because alpha channel is not supported yet:
[image]

@embeddedt 
It really would better to make possible to use images directly. By having an `lv_canvas_get_img_dsc()` function canvas could be used too with new interface.

To do this the content of `lv_canvas_set/get_px` should be moved to more generic place, e.g. `lv_draw_img.c` and named like `lv_img_buf_set/get_px`. These `set/get_px` functions are not complete yet (e.g. the alpha support is missing as mentioned above) but once they will be on their correct place I can add support for the missing parts.

Could you reorganize it this way?

embeddedt, 2019-03-21:
@kisvegabor Sure. I'll do that today.

seyyah, 2019-03-22:
@kisvegabor @embeddedt 

I meet Ne10 project.

> An open optimized software library project for the ARM® Architecture
https://github.com/projectNe10/Ne10

The Ne10 has a lot of modules

- FFT
- FIR
- IIR
- Math: matrix, vector
- Physics

The most important for us is [imgproc](https://github.com/projectNe10/Ne10/tree/master/modules/imgproc)

- Rotate
- Resize
- Boxfilter

The Ne10 is highly inspiring. Using this implementation ideas in the Ne10, we can accomplish the following future/actions:

- image zoom / rescale
- image filter: blur, contrast enhancement etc
- maybe image affine transform via math matrix operations or convolution

kisvegabor, 2019-03-22:
@seyyah 
It really looks like a very useful resource. Actually exactly these image processing features were in mind too to improve canvas. Fortunately, these can be added in a modular way at any time once we establish the environment via the rotate feature.

kisvegabor, 2019-03-22:
[Here](https://github.com/littlevgl/lvgl/issues/87#issuecomment-475077487) we had a discussion with @turoksama about adding interface for advanced 2D GPU functions, such as rotate or zoom. 

The 2D GPU functions don't fit well into LittlevGL's internal drawing mechanisms but can be used to make operations on simple buffers.

I've just realized that we can use a GPU interface here to replace software operations. What do you think?

turoksama, 2019-03-23:
Software renderer(by default): Please take a look at stb or "3d math" project, it may inspire us with matrix caculations, and also, it is a subset of 3D math.
Hardware renderer(if exist): Advanced 2D GPU pipeline, which we are working on.

embeddedt, 2019-03-23:
@kisvegabor Should I still move the canvas functions or wait for the GPU discussion first?

seyyah, 2019-03-23:
Maybe also helpful the ol3d open source 3d graphic library.

A tiny portable 3D graphics lib for micro controllers.
Github: https://github.com/avem-labs/ol3d

kisvegabor, 2019-03-24:
@embeddedt 
> Should I still move the canvas functions or wait for the GPU discussion first?

Yes, please. This reorganization is required to move forward with the software rendering but independent from hardware acceleration. 

@turoksama @seyyah 
I was thinking more about the GPU interface and I've just realized that no dedicated interface is required at all. It's because the canvas's buffer is a plain array of colors and the GPU functions can use it directly. Once the rendering is done only `lv_obj_invalidate(canvas)` needs to be called to redraw the screen with the new canvas.

Following this approach, [ol3d](https://github.com/avem-labs/ol3d) can be integrated with LittelvGL as an external library which actually uses the canvas. 

What do you think?

embeddedt, 2019-03-24:
@kisvegabor Just for clarification: the functions should be modified to accept `lv_img_dsc_t` right?

turoksama, 2019-03-24:
Well, what I meant is that it may worth giving a try with GPU acceleration for the library, not just canvas object itself.

kisvegabor, 2019-03-24:
@embeddedt 
I merged the PR about moving `set/get_px` to `lv_draw_img.c` and update `lv_canvas_rotate` accordingly. Here is the new example code:
```c
    /*Create a "source" canvas from an image*/
    LV_IMG_DECLARE(handle);

    /*Create a "destination" canvas*/
    static lv_color_t canvas_buf[200 * 200];
    lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);
    lv_canvas_set_buffer(canvas, canvas_buf, 200, 200, LV_IMG_CF_TRUE_COLOR);

    uint32_t angle = 0;

    while(1) {
        /* Periodically call the lv_task handler.
         * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        usleep(10*1000);

        /*Clear the canvas*/
        memset(canvas_buf, 0xFF, sizeof(canvas_buf));

        /*Rotate the source and store it on the  destination*/
        lv_canvas_rotate(canvas, &handle, angle % 360,
                                            96, 9, /*Offset*/
                                            8, 91); /*Pivot*/
        angle++;
}
```

kisvegabor, 2019-03-24:
@turoksama 
> Well, what I meant is that it may worth giving a try with GPU acceleration for the library, not just canvas object itself.

It really would be great but:
1. LittlevGL's drawing interface is not designed to adopt such high-level GPU features. It's very well optimized, therefore not modular, not flexible, but runs fast and has some important features like opacity and anti-aliasing. So the integration there is reasonable now.
2. Even it could be integrated into the drawing mechanisms it would mean to rotate the images on demand. It could mean to rotate an image 3 times if it is refreshed in 3 parts and rotate it every time when drawing occurs. It doesn't seem effective.
3. By integrating it to the canvas the images could be "pre-rotated" and even reused by several objects. It requires RAM indeed, but once you have a high-level GPU I suppose you have some RAM too.

kisvegabor, 2019-03-26:
I've updated the `lv_img_buf_set/get_px` functions to supprt all color formats and opacity.
For example, a clock can be build from:
1. a normal image as `lv_img` as the background of the clock
2. a canvas with `LV_IMG_CF_ALPHA_2BIT` format to display the hands of the clock.

I've added macros the easily get the required buffer size: `LV_CANVAS_BUF_SIZE_...`

I've tested some configurations and different color formats (the canvas and the image to rotate can have different color format) and they worked well for me.

As the image rotation is implemented and seems working I close this issue. If you had a chance to test it
and found a bug please open a new dedicated issue for it.

Thank you very much for your contribution! It really made LittelvGL more complete.

closed 2019-03-26 by kisvegabor