lvgl/lvgl #685: line and arce shadow

ali-rostami, 2019-01-01:
Hi 
It seems that changing the shadow part of the style of a line doesn't change anything, does it?
I need to add a shadow to a line and more than that I have to specify the direction of it. 
Is it possible now?
I think that we need to add more options to lv_shadow_type_t in order to specify the direction of shadow?

kisvegabor, 2019-01-03:
Hi,

`body.shadow` affects only shadow of rectangle object.

What is your use case? Maybe we can suggest a workaround.

ali-rostami, 2019-01-04:
As you saw this picture in #679 :
https://user-images.githubusercontent.com/9710249/50571786-58720b00-0dc8-11e9-9ab3-344e9aa15cff.jpg
I draw that white frame with line and arc. I want to give a shadow to the right lines toward the right-bottom side.

kisvegabor, 2019-01-04:
I suggest using images instead, You can use 1px wide images too. If you set the larger size for the object then the image source size then the image source will be repeated like a mosaic. So it won't consume a lot of memory but gives you great flexibility and quality.

embeddedt, 2019-01-04:
@ali-rostami I assume its just that shadows haven't been implemented for lines or arcs yet. They are drawn with different code than other objects.

kisvegabor, 2019-01-04:
@ali-rostami 
> You mean make the shade by setting the width of an img object for example 4 pixels while its source has 1 pixel width?

Exactly!

@embeddedt 
>  I assume its just that shadows haven't been implemented for lines or arcs yet. They are drawn with different code than other objects.

Yes, shadow drawing is different for every primitive. And it's quite difficult the implement a well optimizes shadow drawing for lines: 
1. The shadow on the perpendicular edges become rounded 
2. Unlike the rectangle, the shadow for the lines can be skew

Because of these, it would be very complicated to render the shadow for lines. If we had a lot of resources, shadow drawing would be trivial. But on small embedded systems it isn't. 

So implementing a shadow drawing for lines would require a lot of effort. And it's not a common requirement so I think it's not worth it.

ali-rostami, 2019-01-04:
ok, thanks for your reply. I will try it.
another question I have is: Does the lvgl use anti-aliasing for drawing arcs? Again with respect to the above picture, the round parts that are created with arc objects seem to be not anti-aliased?

embeddedt, 2019-01-04:
@ali-rostami Antialiasing is not supported on arcs right now, because they are drawn with different code.

embeddedt, 2019-01-05:
@ali-rostami Line and arc drawing is done separately from everything else, in a different part of the code. This is why they don't support antialiasing and shadows.

kisvegabor, 2019-01-05:
@ali-rostami 
Adding anti-aliasing to Arc is planned. It's much more simple than the line shadow. 
Hopefully, I will find some time to do it in `v6.0`.

ali-rostami, 2019-01-06:
thanks, dear @kisvegabor 
Is the v6.0 the next version after v5.3?
Does it have any reason, like some prerequisites that aren't in ver 5 and will be added in ver6.0, that will make implementing this feature possible?

kisvegabor, 2019-01-06:
Yes v6.0 is the next version after v5.3. v5.3 will be released in a few weeks and I'm not sure that I will have time to implement it.

kisvegabor, 2019-01-10:
As the implementation of the original request, the line shadow, is not planned I close this issue.

closed 2019-01-10 by kisvegabor