lvgl/lvgl #690: Compiling lv_log.c fails due to type limits error

Ferruck, 2019-01-02:
Hi!

When compiling LittlevGL with the GCC 7.3.1 provided by Arm (`arm-bare-none-eabi`) with `-Wall`, `-Wextra` and `-Werror` activated and `LV_LOG_LEVEL` set to `LV_LOG_LEVEL_TRACE`, compilation fails with the following error

```
lv_log.c:61:14: error: comparison is always true due to limited range of data type [-Werror=type-limits]
     if(level >= LV_LOG_LEVEL) {
              ^~
cc1: all warnings being treated as errors
```

because `level` is a `lv_log_level_t` (which is a typedef for `uint8_t`) and `LV_LOG_LEVEL_TRACE` equals `0`.

There are multiple possibilities how to fix this (in case you think that this is an error, too):

* Start enumerating log levels with `1`
* Enumerate log levels in a way that bitwise comparisons could be done
* Streamline the definitions of the log macros in a way that the comparison is completely superfluous (though this would imply that `lv_log_add()` is kinda internal and shall be used _exclusively_ through the log macros)
* Add compiler pragmas for "downgrading" this to a warning
* Your obvious solution that I didn't think of

I'd gladly provide a patch for any of the above if you like and – again – if you think this is an issue, too, otherwise feel free to simply close this issue.

Regards

Ferruck, 2019-01-02:
I forgot one option: Use a non-unsigned integer type for `lv_log_level_t`.

embeddedt, 2019-01-02:
@Ferruck Considering that we aren't likely to have 128 log levels anytime soon, I think using a signed integer type might be the cleanest type.

kisvegabor, 2019-01-03:
@Ferruck Thank you for the clear report and suggestion!

I merged your patch!

closed 2019-01-03 by kisvegabor

Ferruck, 2019-01-03:
No problem, you're welcome.

PR #691 (2019-01-03): lv_log: Fix type limits error
When building LittlevGL with a rather recent compiler (tested with GCC
7.3.1 provided by Arm), `-Wall`, `-Wextra` and `-Werror` activated and
`LV_LOG_LEVEL` set to `LV_LOG_LEVEL_TRACE`, compilation fails due to a
type limits error in lv_log.c

     if(level >= LV_LOG_LEVEL) {
              ^~

because the comparison is always true, since level is a `lv_log_level_t`
(that is a typedef for `uint8_t`) and `LV_LOG_LEVEL_TRACE` equals 0.

Fix that by making `lv_log_level_t` a typedef to `int8_t`.

Fixes #690
commit 75f3198c (2019-01-03): lv_log: Fix type limits error

When building LittlevGL with a rather recent compiler (tested with GCC
7.3.1 provided by Arm), `-Wall`, `-Wextra` and `-Werror` activated and
`LV_LOG_LEVEL` set to `LV_LOG_LEVEL_TRACE`, compilation fails due to a
type limits error in lv_log.c

     if(level >= LV_LOG_LEVEL) {
              ^~

because the comparison is always true, since level is a `lv_log_level_t`
(that is a typedef for `uint8_t`) and `LV_LOG_LEVEL_TRACE` equals 0.

Fix that by making `lv_log_level_t` a typedef to `int8_t`.

Fixes #690