lvgl/lvgl #694: #warning pre-processor command breaks Visual Studio builds

MCF, 2019-01-02:
A #warning pre-processor command was added recently that breaks any building of the lvgl code with Visual Studio (e.g. in the Simulator).  The Visual Studio compiler does not support the #warning command.  The offending line is here in lv_vdb.h:

https://github.com/littlevgl/lvgl/blob/a114a42b74e8d0d4889d21a7660fa455e460d73e/lv_core/lv_vdb.h#L34

There are two ways to fix this problem.

1. The easiest (and best IMHO) is to simply remove the warning line.  The warning seems extraneous in this case - either you supply "defaults" for macro definitions or not - the warning just seems like noise.   To put it in perspective, when building the simulator that warning will be shown around 45 times in the compiler/linker output. 

2. You can build some kind of platform independent pro-processor command for this.  The visual studio equivalent is shown here:
http://goodliffe.blogspot.com/2009/07/c-how-to-say-warning-to-visual-studio-c.html

As I mentioned above I would vote for option 1.  But I can prepare a pull request for either depending on how others feel.

embeddedt, 2019-01-03:
Number 1 sounds cleaner and simpler. I agree that it seems extraneous.

kisvegabor, 2019-01-03:
I also vote for 1)

@MCF 
Can you send a PR?

PR #697 (2019-01-03): Remove unneeded #warning pre-processor command in lv_vdb.h
That directive breaks non-gcc builds (e.g. for the pc simulator) and is
not very valuable from a user point of view.

Fixes: #694
closed 2019-01-03 by embeddedt