lvgl/lvgl #701: how Hide flashlight icon when press a external key

majidsoltani, 2019-01-04:
hi
how i can hide flashlight icon when press a external key?

[image]

embeddedt, 2019-01-04:
@majidsoltani Try `lv_obj_set_hidden`.

majidsoltani, 2019-01-04:
this my code to draw flashlight icon :

#define GT_SYSTEM_ICONS_Y  2
#define GT_SYSTEM_ICONS_MARGIN_X  3

void gt_show_System_Icons(uint8_t buttery_state,bool flashLight)
{
	uint8_t x=LV_HOR_RES - GT_SYSTEM_ICONS_MARGIN_X;

	if (buttery_state==3)
	{
		LV_IMG_DECLARE(gt_ico_battery_3);
		lv_obj_t *img_battery_3 = lv_img_create(lv_scr_act(),NULL);
		lv_img_set_src(img_battery_3,&gt_ico_battery_3);
		lv_obj_set_pos(img_battery_3,x-26,GT_SYSTEM_ICONS_Y);

		x=x-25-GT_SYSTEM_ICONS_MARGIN_X;
	}
	if (flashLight==true){
		LV_IMG_DECLARE(gt_ico_flash_light);
		lv_obj_t *img_flash_light = lv_img_create(lv_scr_act(),NULL);
		lv_img_set_src(img_flash_light,&gt_ico_flash_light);
		lv_obj_set_pos(img_flash_light,x-25,GT_SYSTEM_ICONS_Y);

		x=x-25-GT_SYSTEM_ICONS_MARGIN_X;
	}

}

and use top function in main code two draw system icons.

        gt_show_System_Icons(3,true);

	while(1){
		lv_task_handler();
		delay_us(10000);		
	}

now i want show/hide flash light icon when user pressed an external key.

i don't now how use lv_obj_set_hidden() to hide/show flashlight icon :-P

turoksama, 2019-01-04:
Hi,
 You may use ```lv_obj_set_hidden(img_flash_light, true);``` in your step. But the object pointer ```img_flash_light ``` cannot be seen outside of function ```gt_show_System_Icons```, you may consider to make it global or others.

seyyah, 2019-01-04:
@majidsoltani,

> [...] when press a external key

For this purpose, you can use LV_INDEV_TYPE_BUTTON [1, 2, 3].

## :clipboard: Reference
1. https://github.com/littlevgl/lvgl/issues/567#issuecomment-446586421
2. https://www.youtube.com/watch?v=dk772McmJs4
3. https://github.com/littlevgl/lvgl/wiki/Porting#button

majidsoltani, 2019-01-04:
i define img_flash_light as global pointer. when use lv_obj_set_hidden(img_flash_light, true) before run lv_task_handler();  The flashlight icon is hidden but if the lv_obj_set_hidden(img_flash_light, true) run after the lv_task_handler(); command, the flashlight icon will not be hidden and will still be displayed in the display. how i can refresh current screen?

embeddedt, 2019-01-04:
@majidsoltani Please verify that you are actually entering the body of that `if` statement with a debugger. This sounds like a problem with your setup, rather than LittlevGL.

seyyah, 2019-01-04:
@majidsoltani Please try the `volatile`,

```c
volatile int c = 0;
[...]
```

majidsoltani, 2019-01-05:
> @majidsoltani Please verify that you are actually entering the body of that `if` statement with a debugger. This sounds like a problem with your setup, rather than LittlevGL.

i checked...i think GUI screen most be refresh after change gt_ico_flash_light hidden attribute.
but i don't khow how refresh GUI !

embeddedt, 2019-01-05:
@majidsoltani The GUI is refreshed by `lv_task_handler`. If the screen is not being refreshed it means that `lv_task_handler` is never executed a second time.

Given the code sample you provided I would first check that `delay_us` actually returns, and then check whether you enter the `if` statement.

majidsoltani, 2019-01-05:
> @majidsoltani The GUI is refreshed by `lv_task_handler`. If the screen is not being refreshed it means that `lv_task_handler` is never executed a second time.
>
> Given the code sample you provided I [...]

yes enter in if statement....I found the reason for my problem, but I still could not solve it. this problem is form ex_disp_flush (My LCD Driver Code).

embeddedt, 2019-01-05:
@majidsoltani What is the problem in `ex_disp_flush`?

majidsoltani, 2019-01-05:
is use nokia 1661 LCD , this lcd use ST7735 driver. 
this is my flush code :

int32_t x;
	int32_t y;
	setAddrWindow(x1,y1,x1,y2);

	for(y = y1; y <= y2; y++) {
	    for(x = x1; x <= x2; x++) {
	        /* Put a pixel to the display. For example: */
     	    nlcdSendPixel(color_p->full);
	        color_p++;
	    }
	}
	nlcdSendCmd(0x00);
	/* IMPORTANT!!!
	 * Inform the graphics library that you are ready with the flushing*/
	lv_flush_ready();
}

but i think this code not work currently for draw pixels in x,y position

majidsoltani, 2019-01-05:
Video result :

[image]

embeddedt, 2019-01-05:
@majidsoltani It looks like there are artifacts appearing on the display. This could be memory corruption or a problem with your display driver.

majidsoltani, 2019-01-05:
I finally solved my problem

for solve this problem  i do :
1- use lv_tick_inc(3); in main while(1){}
2- in ex_disp_flush function before  send setAddrWindow command , send  NOP command 
   (after send NOP command , setAddrWindow Command work as correctly)

[image]

embeddedt, 2019-01-05:
@majidsoltani I think that your problem was simply that you didn't call `lv_tick_inc` before. Glad to hear it is fixed.

`lv_tick_inc` shouldn't be called in the main `while(1)` loop. It needs to be called from a thread or an interrupt handler. This is because you cannot guarantee the timing of `lv_task_handler` in all scenarios.