lvgl/lvgl #702: how can i draw a line with multiple color

greedyhao, 2019-01-04:
I want to port a NES emulator to stm32. 
The PPU need a function to write a line buffer to the screen. I had look through the [LittlevGL documentation](https://docs.littlevgl.com). But I can't find the tutorial to write the line buffer to the screen.

kisvegabor, 2019-01-04:
Please, describe your software architecture in more detail:
1. By NES  you mean this? http://www.emulator-zone.com/doc.php/nes/
2. If it's really a Nintendo emulator how comes LittelvGL in the picture? And how the emulator and LittlevGL work together?

greedyhao, 2019-01-04:
1. Yes, NES is Nintendo NES Emulators.
2. My board is based on stm32f767. I use a RTOS named [RT-Thread](https://github.com/RT-Thread/rt-thread). The LittlevGL had ported to RT-Thread to provide GUI service. 
A important part of NES emulator named Picture Processing Unit(PPU), needed to write line buffer (read from .nes file) to the screen. I notice that the image object in LittlevGL can partly meets my demand. I don't know if I can define a image object to display the line buffer. If it works ,then the PPU can works well, and the NES can also works.
```c
lv_img_dsc_t line_buffer = {
  .header.always_zero = 0,
  .header.w = 256,
  .header.h = 0,
  .data_size = size* LV_COLOR_SIZE / 8,
  .header.cf = LV_IMG_CF_TRUE_COLOR,
  .data = line_buffer_map,
};
```

kisvegabor, 2019-01-05:
I think it would be better to use:
```c
lv_img_dsc_t nes_fb_img_src= {
  .header.always_zero = 0,
  .header.w = NES_HOR_RES,         //2D buffer
  .header.h = NES_VER_RES,          //2D buffer
  .data_size = size* LV_COLOR_SIZE / 8,
  .header.cf = LV_IMG_CF_TRUE_COLOR,
  .data = nes_frame_buffer,
};

...

//Create an image with the NES frame buffer image
lv_obj_t * nes_fb = lv_img_create(lv_scr_act(), NULL);
lv_img_set_src(nes_fb, &nes_fb_img_src);
```

And when NES tries to write a line to the display you should:
1. copy the line  `nes_frame_buffer`
2. call `lv_obj_invalidate(nes_fb)` to refresh the image

This way you can do anything with the rendered image similarly to any other objects n LittlevGL. You can create objects on top of it, set its position, hide it etc.

The only drawback is you need some extra memory.

greedyhao, 2019-01-06:
Thank you so much for your suggestion. I will try it later. I'll give you feedback on any progress.

closed 2019-01-06 by greedyhao

embeddedt, 2019-01-13:
@greedyhao Wow! That looks really cool! Are you planning to make it open-source?

kisvegabor, 2019-01-13:
Awesome! Congratulation!

Recently I encourage every people with a great project to write about it in LittlevGL's blog. Would you be interested in it?

greedyhao, 2019-01-14:
@embeddedt You can find it on https://github.com/greedyhao/nes-lvgl-rtt

embeddedt, 2019-01-14:
@greedyhao I looked at your repository. Where do I find the code relating to the NES emulation itself?

greedyhao, 2019-01-14:
@embeddedt  I didn't even notice until you told me. Some files were not uploaded due to. gitignore. The project is updated. NES emulator is under bsp\qemu-vexpress-a9\packages\nes-old