lvgl/lvgl #706: Using images as resource file

mharizanov, 2019-01-05:
Hello,

Instead of having multipple image resources, it is sometimes useful to combine several images into one, and then visualise a "window" of it. 
As an example, WiFi strength indicator (credit to Moddable), but same will apply to battery levels etc:

[image]

Is there a way that this sort of image resource can be used in LittlevGL?

seyyah, 2019-01-05:
This idea is known as image sprite at web tech [1].

For example, sprites.png (132x44px),

[image]

```css
#home {
  height: 44px;
  width: 46px;
  background: url('sprites.gif') 0 0;
}

#prev {
  height: 44px;
  width: 43px;
  background: url('sprites.gif') -47px 0;
}

#next {
  height: 44px;
  width: 43px;
  background: url('sprites.gif') -91px 0;
}
```

`height` and `width` parameters are subimage's size. `0px, -47px, -91px` are shift left amount.

I don't know any practical way in the LittlevGL.

## :clipboard: Reference
1. https://www.w3schools.com/css/css_image_sprites.asp

embeddedt, 2019-01-05:
@kisvegabor I think `offset` is a better solution, the other one seems too much like a hack and would consume more memory.

mharizanov, 2019-01-06:
Thanks for the `lv_obj` + `lv_obj_set_pos(img, x, y)` idea, I think it will work for me. I'm hesitant regarding the payoff of developing the `offset` functionality, as this is a very specific use case that I had. 
Thanks!

kisvegabor, 2019-01-07:
Making sprites more easy to use would require a good interface too to set the width, height, x and y data of the images. So a new object type, derived from `lv_img` would be even better. 

But if you are happy with the `lv_obj + lv_obj_set_pos(img, x, y)` then leave it for later. :)

closed 2019-01-07 by mharizanov