lvgl/lvgl #707: VDB size STM32

Simof92, 2019-01-05:
Hi,

I'm trying to implement the demo application on the development board DISCO-STM32F746.
Screen 480x272 4.3"
Internal RAM 64KB
External SDRAM 64Mbit

The demo code work with 16bit color depth with the correct size of screen but the color of button is wrong (Horizontal Blu Lines)

I want to use 32bit color depth, I have configured the lv_conf.h and adapt the lcd driver with ARGB8888. Now the colors are correct but the image on the screen is divide in 4 equal parts and I don't understand what is the problem.

The VDB size is 480*272 and is allocated in External SDRAM.
[image]

[image]

[C_File.zip](https://github.com/littlevgl/lvgl/files/2729445/C_File.zip)

embeddedt, 2019-01-05:
@Simof92 What a lucky coincidence! I just implemented 32-bit display support for that board earlier. Now you should just be able to set `LV_COLOR_DEPTH` to 32 and everything should work! Or is that what you already tried?

Simof92, 2019-01-05:
With LV_COLOR_DEPTH to 16 the colors of the button are wrong (Horizontal Blue Lines) (Image 1)
With LV_COLOR_DEPTH to 32 the colors are correct but the size of screen is reduced (Image 2)

embeddedt, 2019-01-05:
@Simof92 Please run `git pull` and `git submodule update`. I think you may not have my latest changes.

Simof92, 2019-01-05:
@embeddedt I have updated my Atollic project with your last changes and now the colors are perfect in 16bit depth. What was the problem in the previus version? Thank you a lot.
Now I will try with 32bit depth!

embeddedt, 2019-01-05:
@Simof92 I'm not sure what your problem was, 16-bit color has always worked as far as I know. Maybe you happened to have an old version of the project.

Simof92, 2019-01-05:
`LV_VDB_ADR (SDRAM_ADDR)+(LV_HOR_RES*LV_VER_RES*2)`
`LV_COLOR_DEPTH 32` 
DMA2D configured for ARGB8888
LTDC configured for ARGB8888

and the result is the same of the second picture.

The screen area is divide in 4 parts, in the first 2 parts is draw only half screen and in the other parts there are only noise

embeddedt, 2019-01-05:
@Simof92 If you use the default `lv_conf.h` supplied with the F746 project do you have problems?

Simof92, 2019-01-05:
@embeddedt after the update i don't have problem with the default configuration but LV_COLOR_DEPTH is configured 16 bit. 
With the changes described above don't work correctly

embeddedt, 2019-01-05:
@Simof92 Use the default configuration but change `LV_COLOR_DEPTH` to 32. That should work (it is working for me).

embeddedt, 2019-01-05:
@Simof92 My mistake! I had made the pull request but the changes weren't actually in `master` yet; they were in a separate branch! Please `git pull` and `git submodule update` again.

Simof92, 2019-01-06:
Great! Wonderful! Now is perfect! My next step will be to change the tft with 800x480 display! thank you a lot!

embeddedt, 2019-01-06:
@Simof92 Glad to hear it helped you!

closed 2019-01-06 by embeddedt

Simof92, 2019-01-11:
Can you update the .ioc file created by STMCubeMX?
I have some problem to recreate the same project in Atollic Studio.

embeddedt, 2019-01-11:
@Simof92 I never changed any `.ioc` files but I think @kisvegabor created the original project so he probably knows more about it than I do.

Simof92, 2019-01-12:
This is the problem, Do you have some idea about the problem?

[image]
[STM32F746_TFTDriver_v2.zip](https://github.com/littlevgl/lvgl/files/2751891/STM32F746_TFTDriver_v2.zip)

Simof92, 2019-01-12:
The problem was in the SystemClockConfig function because it initialized the LTDC frequency

kisvegabor, 2019-01-12:
@Simof92 STM32F746G project was created by someone else but I don't remember who. Anyway glad to hear that you solved the issue.
I was thinking about improving how the projects are added to LittlevGL. See [this](https://github.com/littlevgl/lvgl/issues/543#issuecomment-453749574),
It would be easier to see who created a project.

Simof92, 2019-01-12:
I was able to rebuild the correct configuration in STMCubeMx and now everything works correctly in Atollic TrueSTUDIO.
I can upload the whole project if it will help anyone

kisvegabor, 2019-01-12:
Great! Let's wait what @embeddedt thinks about this new project organization. If he also likes it I create a repo and give you write access there.

Simof92, 2019-01-12:
Thanks for everything

embeddedt, 2019-01-12:
@kisvegabor It's a much better solution and there will be less sub modules per project (and thus a smaller download size).

kisvegabor, 2019-01-12:
I can imagine 3 solutions:
1. use 2 separate repos -> results in a lot of repos, and need to keep up to date both repos
2. use 1 repo with two folders for the 2 project -> need to update both repos
3. use 1 repo with two folders for the 2 project but they use the same code base -> might be difficult to copy a project

I vote for the 3rd option. What do you think?

embeddedt, 2019-01-12:
@kisvegabor I've never used Atollic. But if it were possible to keep the IDE-specific stuff in two different directories that would be the simplest way.

@Simof92 Do you think this is possible?

kisvegabor, 2019-01-12:
@embeddedt Atollic is also Eclipse-based and it has the same project settings to add source files.
So I think it's possible

Simof92, 2019-01-12:
I don't now well how work the repos.
Atollic allows to work in different workspace

kisvegabor, 2019-01-12:
I've imagined the structure like this:
- stm32f746_disco_no_os
  - atollic
    - main.c
    - other project realted things
  - sw4stm32
    - main.c
    - other project realted things
  - lvgl
  - lv_examples
  - hal_stm_lvgl

So you only need to link som eexternal foledrs (lvgl, lv_eamples, hal_stm_lvgl) to the project.

You an do it in the project properties in Atollic and SW4STM32 too:
[image]

embeddedt, 2019-01-12:
@kisvegabor Perfect!

kisvegabor, 2019-01-12:
Awesome! :)

kisvegabor, 2019-01-13:
I played with the "Link folder"  thing for a while and it's not so good as it seemed.
- more complicated project structure 
- more complex project settings (might be hard to follow for other projects)
- more complicated workspace structure because you will have an unused project.

In the light of it'd rather prefer separate repos for every IDE. 
I rename the current repo to `..._sw4stm32` and the create a new with `..._atollic`. Do you agree?

kisvegabor, 2019-01-13:
I moved the f429 disco project: https://github.com/littlevgl/stm32f429_disco_no_os_sw4stm32

It was very easy to move the project this way.

Simof92, 2019-01-13:
Yes i think is the better solution, when you will create the stm32f746_disco_no_os_atollic repo I will commit the project

embeddedt, 2019-01-13:
@kisvegabor It is a better solution  in my opinion as well.

kisvegabor, 2019-01-13:
Great!

Here are the repos:
- Atollic: https://github.com/littlevgl/stm32f746_disco_no_os_atollic 
- SW4STM32: https://github.com/littlevgl/stm32f746_disco_no_os_sw4stm32

embeddedt, 2019-01-13:
@kisvegabor I've uploaded a working project to `stm32f746_disco_no_os_sw4stm32`.

kisvegabor, 2019-01-13:
@embeddedt  Great thank you! I've added the git command to the README to clone the repo and the submodules.

kisvegabor, 2019-01-15:
Awesome! Thanks you guys!

seyyah, 2019-01-16:
Thanks a lot