lvgl/lvgl #708: Method of returning error from lv_task

embeddedt, 2019-01-05:
While refactoring the Windows driver I have found that I need a way to cleanly return a value from `lv_task_handler`. Would this be a reasonable feature for a PR?

kisvegabor, 2019-01-05:
Why not!? Returning `int`, `bool`, or `lv_res_t`?

embeddedt, 2019-01-05:
@kisvegabor Probably `lv_res_t` to be consistent with the rest of the library. I suppose `LV_RES_INV` is treated as an error in that case. Is that okay?

kisvegabor, 2019-01-13:
I added this update. Finally, I used `bool` return value because `lv_res_t` is introduced n `lv_obj.h` and I didn't want to add object related things to tasks.

kisvegabor, 2019-01-13:
Great! :)
Waiting for that!

embeddedt, 2019-01-13:
@kisvegabor I've just realized that the current implementation still doesn't let me return a value from the task itself, it only allows `lv_task_handler` to report an error trying to execute the task.

kisvegabor, 2019-01-13:
> I need a way to cleanly return a value from lv_task_handler

I interpreted it as `lv_task_handler` needs a return value. 
Returning from the tasks with a value can go only into `v6.0` because it might break backward compatibility.

embeddedt, 2019-01-13:
@kisvegabor I see. In that case the Windows port will have to wait, but it should only be 2 weeks at most.

kisvegabor, 2019-01-13:
Actually, I don't really understand it. How Windows things are related to `lv_task_handler`? How does the current implementation look like where it causes an issue?

embeddedt, 2019-01-13:
@kisvegabor When the user closes the window I need to return from `WinMain`. That is the only clean way to exit. Thus I need to break out of the `while(1)`  loop that calls `lv_task_handler`.

kisvegabor, 2019-01-13:
I have 2 ideas/questions:
1. Simply calling `exit(0);` on window close would cause an issue? Similarly to [monitor.c](https://github.com/littlevgl/lv_drivers/blob/master/display/monitor.c#L249).
2. Setting a global flag like `win_closed = true` and 
```c
while(!win_closed) {
  lv_task_handler();
  usleep(10*1000);
}
```

embeddedt, 2019-01-13:
@kisvegabor 

> Simply calling `exit(0);` on window close would cause an issue?

That's not supposed to be done on Windows. It would work but it is a hack.

> Setting a global flag . . .

Yes, I thought of that, and I could easily do it that way. I was just trying to make it a useful feature for someone else's project at the same time. Would you like me to just use a global flag?

kisvegabor, 2019-01-14:
> Yes, I thought of that, and I could easily do it that way. I was just trying to make it a useful feature for someone else's project at the same time. Would you like me to just use a global flag?

Yes, I'd prefer it because it's very cleary. For me, it's better than adding a return value to all `lv_task` only for one, special thing.  
So I'll revert the return value of `lv_task_handler`. Do you agree?

embeddedt, 2019-01-14:
@kisvegabor Sounds good. Sorry for the trouble!

closed 2019-01-14 by embeddedt

kisvegabor, 2019-01-14:
No problem! :)
I reverted the canges.

commit 088936e7 (2019-01-13): add return value to lv_task_handler. resolves #708