lvgl/lvgl #710: Issues detected by Clang's static analyzer

embeddedt, 2019-01-06:
In light of the problems discovered in #687, I ran a build of LittlevGL through Clang's static analyzer. Here's a list of the problems it reported (after removing the false positives I could find). I'm not familiar with some of this code so it's possible that there are still false positives in this list.

---

`lv_draw_img.c`: line 718: there is a potential zero-length VLA declaration if `decoder_header.cf` is not initialized to a valid value above it. This applies to `dev-5.3` and `master`. Can we guarantee that `decoder_header.cf` is valid whenever this function is called?

https://github.com/littlevgl/lvgl/blob/a6714bbe4715da4a8a25a07be6d4daa9d907d4dd/lv_draw/lv_draw_img.c#L686-L718

This problem is also present on line 631. See [here](https://github.com/littlevgl/lvgl/blob/a6714bbe4715da4a8a25a07be6d4daa9d907d4dd/lv_draw/lv_draw_img.c#L600-L631).

---

`lv_color.c`: lines 136-140: Clang reports potential division by zero here. This applies to `dev-5.3` and `master`.

https://github.com/littlevgl/lvgl/blob/a6714bbe4715da4a8a25a07be6d4daa9d907d4dd/lv_misc/lv_color.c#L136-L140

---

`lv_chart.c`: line 803: Clang's analyzer says that `p1.y` is never initialized before its use. It appears to be right.

https://github.com/littlevgl/lvgl/blob/167e26a243d2384478b48de91166b5de74dfef59/lv_objx/lv_chart.c#L803

kisvegabor, 2019-01-06:
I never run a static analyser on LittelvGL so I'm very happy to did it! 

### lv_draw_img.c
I've fixed it in `dev-5.3`.

### lv_color.c
if `rgbMax` and `rgbMin` are equal then the function will return earlier:
https://github.com/littlevgl/lvgl/blob/a6714bbe4715da4a8a25a07be6d4daa9d907d4dd/lv_misc/lv_color.c#L129-L133

### lv_chart.c
I've fixed it in `dev-5.3`.

embeddedt, 2019-01-06:
@kisvegabor It looks like there are no more real problems reported by the analyzer. Thanks for the fixes!

closed 2019-01-06 by embeddedt

kisvegabor, 2019-01-07:
Thanks for the report! :)