lvgl/lvgl #715: Zephyr Integration

vanwinkeljan, 2019-01-07:
All,

The [Zephyr RTOS project](https://github.com/zephyrproject-rtos/zephyr) just has [merged](https://github.com/zephyrproject-rtos/zephyr/pull/6826) a glue logic layer integrating lvgl v5.2 with Zephyr.

The glue logic layer provides:
 * Configuration of the majority of the lv_conf flags through KConfig
 * Integration with Zephyr memory management model (kernell, user space & dedicated memory pool)
 * Integration with Zephyr FS API
 * Integration with  Zephyr Display API.
    This includes a simulation port of Zephyr, with SDL based display driver, that runs on linux
 * Integration of lvgl log messages into Zephyr log framework

Further a development environment is available as a [docker container](https://github.com/zephyrproject-rtos/docker-image) which can exposes the SDL based simulation display through a VNC server.

kisvegabor, 2019-01-08:
Hi,

Glad to hear that you have chosen LittlevGL for Zephyr! And thank you for your work with the integration!

We have a guest blogging opportunity exactly for these kinds of great things. Would you be interested to write a post here about:
- promote and introduce Zephyr  
- how to make work Zephyr with LittlevGL in a simulator

Here you can see how the guest blogging works: 
https://blog.littlevgl.com/2018-12-06/contributing#writing-a-blog-post-about-your-experiences

vanwinkeljan, 2019-01-08:
>Glad to hear that you have chosen LittlevGL for Zephyr!

Just to clarify I need a GUI for my application and I was already using Zephyr. I then bumped into LittlevGL, great work btw, and thought it was useful to contribute the integration code back to Zephyr.

>We have a guest blogging opportunity...

I will have a look into this but can't promise anything, writing texts is not my strongest point

kisvegabor, 2019-01-08:
> I will have a look into this but can't promise anything, writing texts is not my strongest point

No problem. It doesn't have to be a literary masterpiece. Only a step-by-step guide to try out Zephyr with LittlevGL. The required commands, some screenshots etc. A guide like this would help a lot to move bare metal GUI project to Zephyr.  Anyway, I don't wanna push you just asked.

vanwinkeljan, 2019-02-04:
@kisvegabor sorry for the delay I would like to take up job on writing a step-by-step guide once zephyr reaches 1.14, this release will be a long term support release and includes a major overhaul of the preferred tooling, as such I would like to wait for this release and write the blog once.

Zephyr PR: https://github.com/zephyrproject-rtos/zephyr/pull/6826#

kisvegabor, 2019-02-05:
Awesome! Do you know what is the schedule for 1.14?

I mark the issue as "pinned" to be not closed automatically if no activity.

vanwinkeljan, 2019-02-06:
Full overview of the road map can be found at https://github.com/zephyrproject-rtos/zephyr/wiki/Program-Management

Currently release is planned for 05-April-19 but feature merge window closes this week so that looks like a good starting point.

kisvegabor, 2019-02-07:
I see, thank you!

closed 2019-06-13 by stale[bot]