lvgl/lvgl #717: [Help needed]How to make the size of table cells fixed even contain long text value?

turoksama, 2019-01-09:
Hi,
 Is there any way to solve this issue since cell text are not actually a label object?
 Cell will change its height automaticlly with long text value.
Regards.

kisvegabor, 2019-01-09:
Hi,
It's not supported now but it might be added. 
What would you expect if the text is longer? Just crop it?

turoksama, 2019-01-09:
I suggest to crop it but not to delete the rest, simply not display in that line, should look like oneline mode.
Because we may need them for future use.

kisvegabor, 2019-01-10:
I update the table with `lv_table_set_cell_crop`. Adding dots would be quite complex so it just crops the text outside of the table

Can you share the case where you need this crop feature?

turoksama, 2019-01-14:
Hi,
 Sorry for the delay because of email related issues.
 I'll test it and share my case when available.Thanks.
Regards.

turoksama, 2019-01-14:
Hi,
 It works fine except get_row_height. Thanks!
 Please review the PR for that bug fix.

closed 2019-01-14 by turoksama

embeddedt, 2019-01-14:
@turoksama Merged, thanks for catching that!