lvgl/lvgl #718: Draw text with transparent label over existing image

hallmatthewp, 2019-01-09:
My current system reads images from external flash storage (in a proprietary format) and writes the image to the LCD. I want to use LVGL to write text on top of the existing image. How is this performed?

Here is the relevant code:
`
static lv_style_t style;
lv_obj_t *        label = lv_label_create(lv_scr_act(), NULL);

// My custom image-to-LCD function
load_flash_img_to_lcd(asset_to_load);

// Set screen to transp style
lv_obj_set_style(lv_scr_act(), &lv_style_transp);

// Set label style as desired
lv_style_copy(&style, &lv_style_transp);
style.text.color = LV_COLOR_WHITE;
style.text.font  = &graphik_semibold_digits_80;
lv_label_set_style(label, &style);
lv_label_set_text(label, "Hello world");
lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);
`

The above code loads my image for a brief moment, before LVGL draws a black screen w/ my white text in the middle-center.

hallmatthewp, 2019-01-10:
I have also enabled `LV_COLOR_SCREEN_TRANSP` in lv_conf.h

embeddedt, 2019-01-10:
Try `lv_style_screen_transp`.

kisvegabor, 2019-01-10:
You can simply put an image to an object. No backgrond will be drawn. So:
1. You don't need to set `lv_obj_set_style(lv_scr_act(), &lv_style_transp);` to achieve this.
2. Instead of `lv_style_copy(&style, &lv_style_transp);` try `lv_style_copy(&style, &lv_style_plain);` 
3. `LV_COLOR_SCREEN_TRANSP` also not required for this

embeddedt, 2019-01-10:
@kisvegabor I suspect @hallmatthewp is using custom image drawing code which doesn't use LittlevGL images. How will `lv_style_plain` work in that case?

kisvegabor, 2019-01-10:
@embeddedt 
By default, no label background is drawn so it really doesn't matter which style is used, but `lv_style_plain` is the most simple style. Transparent styles have `glass == 1` property which might be trick in some cases.

embeddedt, 2019-01-10:
@kisvegabor What about the `lv_scr_act()` object?

kisvegabor, 2019-01-10:
I think I got idea! Do you draw the image directly to the framebuffer and would like to add only a label with littlevgl?
If yes it's not so simple because by default littlevgl needs a background to draw things there. However, there are some features to support it but I need more info about your hardware:
- color depth?
- do you have an LCD interface (RGB lines) or an external display controller?
- how much RAM do you have?
- resolution and image size?
- is it fast decode the image?

hallmatthewp, 2019-01-10:
@kisvegabor and @embeddedt thank you for the quick responses. 

I'm using a Nordic nRF52840 SoC with a 240x240 LCD. 

The LCD controller is communicated to via a parallel interface. It uses 16bit 565 colors. Full images are 115200 bytes (240 horizontal * 240 vertical * 2 bytes per pixel).

We have enough RAM to keep a VDB and a full frame buffer (115200 bytes) which we use to read our images from external flash (QSPI) and write them to the LCD controller.

To pull an image from flash and complete the write to the LCD is ~15ms.

hallmatthewp, 2019-01-10:
@embeddedt is correct, we are not using Little VGL images, but instead our own proprietary format. The goal is to draw an image to the screen and use LVGL to draw characters, lines, circles, etc. on top of our image.

hallmatthewp, 2019-01-10:
In essence, I want LVGL to update only those pixels which are part of the characters, and not touch the pixels that are next to the characters.

kisvegabor, 2019-01-11:
I see, If you have enough RAM o store the raw image I suggest to use it as a LittlevGL image.

Initialize an image structure like:
```c

lv_img_dsc_t img_bg = {
  .header.always_zero = 0,
  .header.w = 240,
  .header.h = 240,
  .data_size = 240 * 240 * LV_COLOR_SIZE / 8,,
  .header.cf = LV_IMG_CF_TRUE_COLOR,
  .data = buffer_with_your_image,
};
```

And create an image with the descriptor:
```c
lv_obj_t * img = lv_img_create(lv_scr_act(), NULL);
lv_img_set_src(img, &img_bg);
```

This way LittlevGLwill handle the image and draw everything correctly.

hallmatthewp, 2019-01-14:
I can try that for our static images, and it may be performant enough. 

We also play animations, which are sequences of animations, played back to back. Performance is very important here, and the lv_img_create method may not be fast enough to achieve the framerate we need. 

Is there any other way to have LVGL not touch the pixels behind the text/labels? I.e. just draw the characters?

embeddedt, 2019-01-14:
@hallmatthewp We added `LV_COLOR_SCREEN_TRANSP` for purposes like that. I'm not sure how to configure it though, because I haven't gotten a chance to try that feature yet.

kisvegabor, 2019-01-15:
@hallmatthewp You don't need to create an image every time just change `buffer_with_your_image` and call `lv_obj_invalidate(img)` to refresh the image. You can even use that buffer directy to load the image there.

@embeddedt  `LV_COLOR_SCREEN_TRANSP` and be used only with `ARGB8888` color format. It uses the Alpha channel to make some parts transparent. It can be used when you have 2 layers (2 frame buffer on each other). E.g. when you play a video on the background layer and you create a menu with LittelvGL on the top layer.  You can test it the PC simulator with the `monitor` driver if you uncomment these lines: https://github.com/littlevgl/lv_drivers/blob/master/display/monitor.c#L318
and set a transparent style for the screen.

embeddedt, 2019-01-15:
@kisvegabor Now I understand. Thanks for the explanation.

kisvegabor, 2019-01-15:
@embeddedt You are welcome! :)

closed 2019-02-28 by stale[bot]