lvgl/lvgl #719: how to set tabview's label width and pos?

HinsunRan, 2019-01-10:
hi, kisvegabor:
how to set tabview's label width and pos?
The widht of automatic label is divided equally from tabview's width.
I set style_tv_btn_rel.body.padding.hor is not effect.
My expectations is that tabview can set label width and position(e.g. two tab labels is in the middle of the tabview top.)
thanks very much!
[image]

kisvegabor, 2019-01-10:
Hi,
it's automatically divided equally right now. What you can do is to add hidden tabs like this:
```c

    lv_tabview_add_tab(tv1, "\211");
    lv_tabview_add_tab(tv1, "First");
    lv_tabview_add_tab(tv1, "Second");
    lv_tabview_add_tab(tv1, "\211");
```

It uses the control byte feature of the button matrix (the buttons are created with the button matrix object). https://docs.littlevgl.com/#Button-matrix

HinsunRan, 2019-01-11:
hi,
the method you said  is not effect. like this:
[image]
the first tab content move to the blank tab content, and the second content move to the first tab.
i see the code:  lv_tabview_add_tab(tv1, "\211");
this first will be create a page, and tab_cnt++, so it is not effect.

kisvegabor, 2019-01-11:
Yes, it will create two dummy tabs.

But a better solution just came to my mind:

```c
    static lv_style_t btns_bg;
    lv_style_copy(&btns_bg, &lv_style_transp_fit);
    btns_bg.body.padding.hor = 80;
    lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BTN_BG, &btns_bg);
```

HinsunRan, 2019-01-14:
@kisvegabor , perfect solution to the problem.
i used this solution, maybe i set the padding.hor is small, and the effect is not so obvious.
Thank you very much!

closed 2019-01-14 by HinsunRan

kisvegabor, 2019-01-14:
Great! :)