lvgl/lvgl #720: hi, how to set different font for the button in num or text keyboard?

HinsunRan, 2019-01-10:
e.g.
the point button in the number keyboard is not very clear，so i want only to set the point button's font,
making sure the point look clearly.
thanks very much!

kisvegabor, 2019-01-11:
So do you want to set a different color only for some button's text in `lv_kb` (keyboard)?

HinsunRan, 2019-01-11:
hi, kisvegabor, the way you said is a good idea for attention.
But, the point (" . ') is very small in a button in keyboard, so for eye-catching，the bettery way is make the point char bigger in button. 
This is why i want to set the point button's font in the number keyboard.
In littleVgl，the number keyboard is created by lv_btnm_set_map(kb, kb_map_num),
i find no way to get the pointer of the point(" . ') button,so how to set different font for the button in num or text keyboard?
thanks very much!

kisvegabor, 2019-01-11:
I see. There is no way to set the font of the button's text separately.
I suggest creating a new font and  editing the font by hand to make the `.` larger.

kisvegabor, 2019-01-11:
For example:

Modfify `0xC0` to `0xE0`: 
https://github.com/littlevgl/lvgl/blob/master/lv_fonts/lv_font_dejavu_20.c#L350

And set the width to `3`:
https://github.com/littlevgl/lvgl/blob/master/lv_fonts/lv_font_dejavu_20.c#L8781

Note that the font in the example contains 1, 2, 4 and 8 bpp too so you need to edit the part you are using.

embeddedt, 2019-01-11:
@kisvegabor I think adding a feature like this would be implemented similarly to image support. Maybe we could use an ID system for button fonts as well.

kisvegabor, 2019-01-12:
@embeddedt Good idea! We can have let's say 4 (2 bit) type of buttons with 5 images + and 5 styles for each type.

kisvegabor, 2019-01-12:
@embeddedt I agree. 

The control byte is getting difficult to manage by setting bits. We should use a macro somehow to help the use of control bytes. However, I don't know (and didn't find) a way to convert a number to string value, like:
```c
LV_BTNM_CTRL(LV_BTNM_WIDTH_2 | LV_BTNM_HIDDEN)"my_btn"  -> "\x43" "my_btn"
```

embeddedt, 2019-01-12:
@kisvegabor Maybe we should find an alternative way of storing the same data, but in a more user-friendly way. I don't think it is possible to turn a number into a string with the C preprocessor.

kisvegabor, 2019-01-12:
> Maybe we should find an alternative way of storing the same data, but in a more user-friendly way.

I agree

Let's figure it out when we get there in `v6.0`. There are other questions too related to Button matrix (e.g. store the data from a simple string).

HinsunRan, 2019-01-14:
> For example:
>
> Modfify `0xC0` to `0xE0`:
> https://github.com/littlevgl/lvgl/blob/master/lv_fonts/lv_font_dejavu_20.c#L350
>
> And set the width to `3`: [...]

@kisvegabor 
This solution solves the point( . ) display problem, and the point shows bigger.
But this solution lead that all points( . ) in other string is also bigger.
Looking forward to the v6.0. Thanks very much.
And thank @embeddedt @hydra @MCF, too.

closed 2019-01-14 by HinsunRan

kisvegabor, 2019-01-14:
@HinsunRan You can generate a new font with only the required characters and use this font in the style.
But in v6.0 we will have better solution for the issue.

HinsunRan, 2019-01-14:
@kisvegabor When does the v6.0 come out?

kisvegabor, 2019-01-14:
@HinsunRan  Probably in this summer