lvgl/lvgl #725: FreeRTOS & littlevGL

EvgeniiaDZ, 2019-01-14:
Hello!
First of all, thank you for LVGL! It's amazing. 
I' ve been working with LVGL and STM32 + HAL, and it's worked perfect!
But now I try no use FreeRTOS and there are some problems. When I use STM32+HAL+FREERTOS it works, when I use STM32+HAL+LVGL it works, but when I started to use LVGL + FREERTOS it falls!
Is there something important in settings that I can miss?
Now I just do some initializations of widget without drawing it on lcd, and It falls in proccess of initialization of gui.
In lv_conf I do:

/* Memory size which will be used by the library
 * to store the graphical objects and other data */
#define LV_MEM_CUSTOM      1                /*1: use custom malloc/free, 0: use the built-in lv_mem_alloc/lv_mem_free*/
#if LV_MEM_CUSTOM == 0
#define LV_MEM_SIZE    (32U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#define LV_MEM_ATTR                         /*Complier prefix for big array declaration*/
#define LV_MEM_AUTO_DEFRAG  1               /*Automatically defrag on free*/
#else       /*LV_MEM_CUSTOM*/
#define LV_MEM_CUSTOM_INCLUDE "FreeRTOS.h"   /*Header for the dynamic memory function*/
#define LV_MEM_CUSTOM_ALLOC   pvPortMalloc       /*Wrapper to malloc*/
#define LV_MEM_CUSTOM_FREE    vPortFree         /*Wrapper to free*/
#endif     /*LV_MEM_CUSTOM*/

embeddedt, 2019-01-14:
@EvgeniiaDZ Where are you calling `lv_task_handler` and `lv_tick_inc`?

Also if you are calling API functions from more than one thread you need to use mutexes.

turoksama, 2019-01-14:
Please pay more attention to the stack memory size of your tasks.
Try to creat a very very compact UI first to make sure FREERTOS is working.
You have to make sure that every block is fine and minimize 2 blocks combine is fine when you building blocks. After that, you can build more blocks and check.

BesitzeRuf, 2019-01-14:
If STM32+HAL+FREERTOS is working, so probably the reason, why it is not working, is the missing of configuration of some timers or interrupts.. Please check it...

BrianPugh, 2019-01-14:
expanding off of what @embeddedt said, it's the easiest and least error prone to create a *recursive mutex* that you take at the beginning of every lvgl portion of code, and give at the end.

It might be abusing macros more than some people are comfortable with, but I'm using a pseudo-context to manage my recursive mutex in my lvgl sections of code using [mp.h](https://www.chiark.greenend.org.uk/~sgtatham/mp/)

See: https://github.com/joltwallet/jolt_wallet/blob/joltos/jolt_os/jolt_gui/jolt_gui.h#L162

Basically all this does is takes a mutex at the beginning of a codeblock and gives it at the end, like:
```
JOLT_GUI_CTX{
    lv_obj_t *obj = lv_obj_create ....
}
```

embeddedt, 2019-01-14:
@BrianPugh That is a very readable solution. I never thought of (ab)using the C preprocessor in that way before. Maybe we could use a construct like that if/when we implement threaded support...

EvgeniiaDZ, 2019-01-15:
Due to my inattention I absolutely forgot about task's stack size . Looks like it works now. I will take your advices about mutex and so on.
Thanks for all of you!

kisvegabor, 2019-01-15:
It would be great to have an example project with FreeRTOS and an STM Dev Board. If you already an example would you upload it to a repository here?

EvgeniiaDZ, 2019-01-17:
So this project not for a standart stm discovery board. If it's ok, I will test  and prepare it for uploading.

closed 2019-02-28 by stale[bot]