lvgl/lvgl #726: Signature capture

JSmolders, 2019-01-15:
Hi Gabor,

can you please explain on how to capture a signature on the newly created lv_canvas object?

Many thanks.

With best regards,

Jan

kisvegabor, 2019-01-15:
Hi Jan,

I added a test file for the canvas to show how it works: https://github.com/littlevgl/lv_examples/blob/master/lv_tests/lv_test_objx/lv_test_canvas/lv_test_canvas.c

To write a signature to the canvas you need to extend its signal function.
Here you can read about the signal functions: https://blog.littlevgl.com/2018-12-13/extend-lvgl-objects

You need to use the `LV_SIGNAL_PRESSING` signal and either set the pixel of the canvas or copy a small circle pattern to it. (example in the test file liked above)

To get the pressed point in `LV_SIGNAL_PRESSING` you can use:
```c
    /*Get triggering input device*/
    lv_indev_t * indev = lv_indev_get_act();
    if(indev) {
        /*Get the pressed point*/
        lv_point_t p;
        lv_indev_get_point(indev, &p);

        /*Make the point relative to the canvas*/
        p.x -= canvas->coords.x1;
        p.y -= canvas->coords.y1;

        /*Set a pixel on the canvas*/
        lv_canvas_set_px(canvas, p.x, p.y, LV_COLOR_BLACK);
    }
```

JSmolders, 2019-01-16:
Hi Gabor,

many thanks for the information.

I gave it a try and the outcome was as expected. There is a need for interpolation to end up with a smooth line.

With best regards,

Jan

seyyah, 2019-01-23:
Hi @kisvegabor,

Is `lv_canvas_create` function implemented? I don't find it.

We make draw simple object on the canvas.

JSmolders, 2019-01-23:
Hi Seyyah,

yes, it is implemented. You need to checkout the dev-5.3 branch.

With best regards,

Jan

seyyah, 2019-01-23:
We try to Github search but it don't find any result. We find [manually](https://github.com/littlevgl/lvgl/blob/dev-5.3/lv_objx/lv_canvas.c#L45). Thanks a lot @JSmolders 

[image]

embeddedt, 2019-01-23:
@seyyah GitHub will take a while to re-index the repository, and I'm not sure if it searches in any branch other than `master`.

seyyah, 2019-01-24:
How can we use the `lv_draw` functions?

This code doesn't draw any thing on the display.

```c
void lv_test_canvas_1(void)
{

    /*Create a buffer for the canvas set its pixels to light blue*/
    static lv_color_t cbuf[sizeof(lv_color_t) * 100 * 80];
    uint32_t i;
    for(i = 0; i < sizeof(cbuf) / sizeof(cbuf[0]); i++) {
        cbuf[i] = LV_COLOR_HEX3(0xacf);
    }

    /*Create the canvas object*/
    lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);

    /*Assign the buffer to the canvas*/
    lv_canvas_set_buffer(canvas, cbuf, 100, 80, LV_IMG_CF_TRUE_COLOR);

    lv_area_t coords = {
      .x1 = 10, .y1 = 10,
      .x2 = 50, .y2 = 50
    };

    lv_draw_rect(&coords, (const lv_area_t *) canvas, &lv_style_plain, LV_OPA_COVER);
}
```

seyyah, 2019-01-24:
@kisvegabor 

What do you think that we add `lv_canvas_get_px`, `drawcircle`, `floodFill` etc. functionality in `lv_canvas`?

kisvegabor, 2019-01-24:
It's okay from my side. As the performance of drawing to the canvas is not so critical (compared to drawing to the display), adding some simple drawing function should be easier.

Would you like send PR?

seyyah, 2019-01-24:
We realize this functions.

```c
// https://www.thecrazyprogrammer.com/2016/12/bresenhams-midpoint-circle-algorithm-c-c.html
void drawcircle(lv_obj_t * canvas, lv_coord_t x0, lv_coord_t y0, lv_coord_t radius, lv_color_t color);

// https://www.thecrazyprogrammer.com/2017/02/flood-fill-algorithm-in-c.html
void floodFill(lv_obj_t * canvas, lv_coord_t x, lv_coord_t y, lv_color_t oldcolor, lv_color_t newcolor);

// https://www.geeksforgeeks.org/boundary-fill-algorithm/
void boundaryFill4(lv_obj_t * canvas, lv_coord_t x, lv_coord_t y, lv_color_t fill_color, lv_color_t boundary_color);

// https://rosettacode.org/wiki/Category:C
void lv_canvas_draw_line(lv_obj_t * canvas, lv_point_t point1, lv_point_t point2, lv_color_t color);

void lv_canvas_draw_triangle(lv_obj_t * canvas, lv_point_t * points, lv_color_t color);

void lv_canvas_draw_rect(lv_obj_t * canvas, lv_point_t * points, lv_color_t color);

void lv_canvas_draw_polygon(lv_obj_t * canvas, lv_point_t * points, size_t sz, lv_color_t color);

void lv_canvas_fill_polygon(lv_obj_t * canvas, lv_point_t * points, size_t sz, lv_color_t color, lv_color_t bg_color);
```

We want to send a PR as ASAP.

seyyah, 2019-01-24:
@kisvegabor 

How to use/draw the image data?

```c
lv_img_dsc_t img_flower_icon = {
    .header.always_zero = 0,
    .header.w = 40,
    .header.h = 40,
    .data_size = 1600 * LV_IMG_PX_SIZE_ALPHA_BYTE,
    .header.cf = LV_IMG_CF_TRUE_COLOR_ALPHA,
    .data = img_flower_icon_map,
};
```

embeddedt, 2019-01-24:
@seyyah Those websites you linked to don't give the license for their code. All code entering LittlevGL has to be MIT-licensed.

seyyah, 2019-01-24:
We try this code. It works

```c
/*Create the canvas object*/
lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);

/*Assign the buffer to the canvas*/
lv_canvas_set_buffer(canvas, cbuf, 100, 80, LV_IMG_CF_TRUE_COLOR_ALPHA);

LV_IMG_DECLARE(img_flower_icon);
lv_canvas_copy_buf(canvas, img_flower_icon.data, img_flower_icon.header.w, img_flower_icon.header.h, 0, 0);
```

seyyah, 2019-01-25:
> Would you like to send a PR?

https://github.com/littlevgl/lvgl/pull/758

kisvegabor, 2019-01-25:
@seyyah Merged, thank you!
The license question was solved?

seyyah, 2019-01-29:
@kisvegabor
> Why is this line commented out? https://github.com/littlevgl/lvgl/blob/dev-5.3/lv_objx/lv_canvas.c#L416

There are two method for filling: lv_canvas_flood_fill, lv_canvas_boundary_fill4

seyyah, 2019-01-29:
@kisvegabor 
> The license question was solved?

@embeddedt 
> Those websites you linked to don't give the license for their code. All code entering LittlevGL has to be MIT-licensed.

I don't know enough about license issue.

I adapt [1-4] code bases for LittlevGL.

I write/ask a comment to author [1-3],

```
Would you allow us to add the codes to LittlevGL under the MIT license?
LittlevGL: https://github.com/littlevgl/lvgl/
```

[4] has a "GNU Free Documentation License 1.2" [5].

## :clipboard: Reference
1. ttps://www.thecrazyprogrammer.com/2016/12/bresenhams-midpoint-circle-algorithm-c-c.html
2. https://www.thecrazyprogrammer.com/2017/02/flood-fill-algorithm-in-c.html
3. https://www.geeksforgeeks.org/boundary-fill-algorithm/
4. https://rosettacode.org/wiki/Category:C
5. http://www.gnu.org/licenses/fdl-1.2.html

embeddedt, 2019-01-29:
We cannot put FDL code into LittlevGL under any circumstances. Sorry to sound harsh, but that license is not compatible with MIT.

As for the other three, if the authors approve, then we can use their code.

If code doesn't have an explicit license on it, that means that it is under full copyright and can't be used anywhere.

embeddedt, 2019-01-29:
@kisvegabor I think both pull requests should be reverted until we know about the licensing issue. Having non-MIT licensed code in a release cannot happen.

seyyah, 2019-01-29:
We may use [1] for "Bresenham's line algorithm". So FDL license problem is eliminated. Codes are same

[3] is MIT license

```c
void bresenham_line(void *data, int x0, int y0, int x1,
                    int y1, int (*setPixel)(void *, int, int)) {
  int dx = abs(x1-x0), sx = x0<x1 ? 1 : -1;
  int dy = abs(y1-y0), sy = y0<y1 ? 1 : -1;
  int err = (dx>dy ? dx : -dy)/2, e2;

  for(;;){
    if (setPixel(data, x0, y0) == BH_STOP_LINE) break;
    if (x0==x1 && y0==y1) break;
    e2 = err;
    if (e2 >-dx) { err -= dy; x0 += sx; }
    if (e2 < dy) { err += dx; y0 += sy; }
  }
}
```

[4] is FDL license

```c
void line(int x0, int y0, int x1, int y1) {
 
  int dx = abs(x1-x0), sx = x0<x1 ? 1 : -1;
  int dy = abs(y1-y0), sy = y0<y1 ? 1 : -1; 
  int err = (dx>dy ? dx : -dy)/2, e2;
 
  for(;;){
    setPixel(x0,y0);
    if (x0==x1 && y0==y1) break;
    e2 = err;
    if (e2 >-dx) { err -= dy; x0 += sx; }
    if (e2 < dy) { err += dx; y0 += sy; }
  }
}
```

## :clipboard: References
1. https://github.com/jb55/bresenham-line.c/blob/master/bresenham_line.c
2. https://rosettacode.org/wiki/Bitmap/Bresenham%27s_line_algorithm#C
3. https://github.com/jb55/bresenham-line.c/blob/master/bresenham_line.c

embeddedt, 2019-01-29:
@seyyah You can't use FDL code in LittlevGL. It's just not compatible with MIT.

seyyah, 2019-01-29:
@embeddedt Ok. We use the code from [3] in LittlevGL. This code is MIT license. So there is not problem.

embeddedt, 2019-01-29:
@seyyah The following functions are based off the FDL code as per your comment up above.

```c
void lv_canvas_draw_line(lv_obj_t * canvas, lv_point_t point1, lv_point_t point2,
  lv_color_t color);

void lv_canvas_draw_triangle(lv_obj_t * canvas, lv_point_t * points,
  lv_color_t color);

void lv_canvas_draw_rect(lv_obj_t * canvas, lv_point_t * points,
  lv_color_t color);

void lv_canvas_draw_polygon(lv_obj_t * canvas, lv_point_t * points, size_t sz,
  lv_color_t color);

void lv_canvas_fill_polygon(lv_obj_t * canvas, lv_point_t * points, size_t sz,
  lv_color_t color, lv_color_t bg_color);
```

Can you review [your commit](https://github.com/littlevgl/lvgl/pull/758/commits/98b6e446f0d6c635baf2bef7496f303626f0f43c) and make sure that the only functions that are in LittlevGL are from the MIT code base?

closed 2019-02-01 by kisvegabor