lvgl/lvgl #727: how to remove chart default line plot

himsha, 2019-01-16:
after initializing chart as below, there is automatically a line in mid of chart area, is this line can be removed and how?

chart init. code
`	/*Create a style for the chart*/
	static lv_style_t style;
	lv_style_copy(&style, &lv_style_btn_ina);
	style.body.shadow.width = 0;
	style.body.shadow.color = LV_COLOR_GRAY;
	style.line.color = LV_COLOR_GRAY;

	/*Create a chart*/
	lv_obj_t * chart;
	chart = lv_chart_create(lv_scr_act(), NULL);
	lv_obj_set_size(chart, 300, 150);
	lv_obj_set_style(chart, &style);
	lv_obj_align(chart, NULL, LV_ALIGN_IN_RIGHT_MID, 0, 10);
	lv_chart_set_type(chart, LV_CHART_TYPE_POINT | LV_CHART_TYPE_LINE);   /*Show lines and points too*/
	lv_chart_set_div_line_count(chart, 10, 5);
	lv_chart_set_range(chart, 0, 100);
	lv_chart_set_point_count(chart, 20);

	/*Add two data series*/
	lv_chart_series_t * ser1 = lv_chart_add_series(chart, LV_COLOR_RED);`

himsha, 2019-01-16:
where should i download dev-5.3?

himsha, 2019-01-16:
i had updated to dev-5.3 - line is gone.

but graph starts from x endpoint then come near to 0 is it possible to start plot from (x,y) = (0,0) and increase thereafter (x,y graph plotting)

himsha, 2019-01-18:
i found the function but how to use it?

static void lv_chart_draw_lines(lv_obj_t * chart, const lv_area_t * mask)

embeddedt, 2019-01-18:
@himsha I think @kisvegabor was referring to the `lv_line` object. Documentation is available here: https://docs.littlevgl.com/#Line

closed 2019-03-03 by stale[bot]