lvgl/lvgl #728: image as a button

himsha, 2019-01-16:
hi,

i had tried image on button and put the button background as transparent, till button not pressed user only see the  image part but once it get clicked background of button also get visible is there is any code I am missing? or is image can also used as a button?

himsha, 2019-01-16:
/*Create anew style*/
	static lv_style_t menu_style_txt;
	lv_style_copy(&menu_style_txt, &lv_style_plain);
	menu_style_txt.text.font = &font12;
	menu_style_txt.text.letter_space = 1;
	menu_style_txt.text.line_space = 1;
	menu_style_txt.text.color = LV_COLOR_GREEN;

	for(uint8_t i = 0; i < 6; i++) {
		lv_obj_t * holder = lv_cont_create(box1, NULL);
		lv_cont_set_fit(holder, true, true);
		lv_obj_set_style(holder, &lv_style_transp);
		lv_cont_set_layout(holder, LV_LAYOUT_COL_M);

		lv_obj_t * btn1 = lv_btn_create(holder, NULL);
		lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, btn_click_action);
		lv_obj_set_style(btn1, &lv_style_transp);
		lv_obj_set_size(btn1, 65, 65);
		lv_obj_set_free_num(btn1, i);   /*Set a unique number for the button*/

		lv_obj_t * btnStartTest_img = lv_img_create(btn1, NULL);                         /*Create the image on 'btn1'*/
		lv_obj_set_click(btnStartTest_img, false);                                  /*Don't let to click to the image (click the background button instead)*/

		if (i==0)		lv_img_set_src(btnStartTest_img, &ImageEditTest);
		else if (i==1)	lv_img_set_src(btnStartTest_img, &ImageStartTest);
		else if (i==2)	lv_img_set_src(btnStartTest_img, &ImageSetup);
		else if (i==3)	lv_img_set_src(btnStartTest_img, &ImageReport);
		else if (i==4)	lv_img_set_src(btnStartTest_img, &ImageDataStorage);
		else if (i==5)	lv_img_set_src(btnStartTest_img, &ImageAbout);

		char buf[32];
		lv_obj_t * label = lv_label_create(holder, NULL);
		lv_label_set_text(label, menu_label[i]);
		lv_obj_set_style(label, &menu_style_txt);
	}

	uckey = NOP;
	while (1)
    {
		if(uckey!=NOP){
			if (uckey==1){
				lv_obj_del(window);
				menu_RunTest();
			}
			uckey = NOP;
		}

  		lv_task_handler();
  		HAL_Delay(100);
    }

mheranco, 2019-01-16:
Hello himsha,

a button has more than one style.
Try to use `lv_btn_set_style(btn, LV_BTN_STYLE_..., &style)` instead of `lv_obj_set_style`.

Link to Documentation:
https://docs.littlevgl.com/#Button

himsha, 2019-01-18:
yes thanks

closed 2019-01-18 by himsha