lvgl/lvgl #729: Set display resolution at runtime

manison, 2019-01-16:
Is it possible to set display resolution at runtime? It seems that the resolution is defined by `LV_HOR_RES` and `LV_VER_RES` symbols at compile time, but I only know the resolution of the connected display at runtime.
If it's not possible right now is this considered as a future feature? Probably related to #329?

embeddedt, 2019-01-16:
@manison Since rotating the display is effectively a change of display size, this is really a duplicate of #329 and will be implemented in `v6.0` (planned to be released in June).

manison, 2019-01-17:
OK, thanks, I thought so.

I tried to define `LV_xxx_RES` constants as variables that I set at runtime, but since my compiler does not support VLAs, I also needed some constants for array size definitions. So I defined new constants `LV_HOR_RES_MAX` and `LV_VER_RES_MAX` (maximum resolution I can set at runtime) and used them instead.

Seems to work for me pretty satisfactory. I can share if interested.

kisvegabor, 2019-01-17:
> Probably related to #329?
Yes, it's related. In `v6.0` multi-display support will be added and the display-related settings (resolution, anti-aliasing, orientation etc.) will be in the display driver (`lv_disp_drv_t`) instead of defines. 

Glad to hear that you solved it! Your solution might help other so if have a link to your repo, please share it.

manison, 2019-01-18:
Commited the change here: https://github.com/manison/lvgl/commit/51b63deb5cafc047ec48ad71db1be55de4eff522

kisvegabor, 2019-01-18:
Thank you! 
It will help others until others until `v6.0` is released.

closed 2019-01-18 by kisvegabor

embeddedt, 2019-01-18:
@kisvegabor We could start off this patch to implement multi-display support.

acsplayon, 2019-01-19:
I just had a need to do this exact thing, run the same code on two different platforms with different screen resolution.  Pretty simple to hack up a solution.
1) create two functions in your code one that returns the X resolution for the running platform, and one that returns the Y resolution. These functions return different numbers based on platform currently running.
2) in your lv_conf.h for your project, define LV_HOR_RES and LV_VER_RES as calls to your functions. #define LV_HOR_RES get_xres()
3) define in lv_conf.h MAX_LV_HOR_RES and MAX_LV_VER_RES as the maximum values that all of your platforms would support.
4) in lv_conf.h change the define LV_VDB_SIZE to use MAX_LV_HOR_RES instead of LV_HOR_RES

When you compile, you will get two preprocessor errors relating to checks about min/max VDB size in lv_obj.h.  I just commented these out, but you could reference MAX_LV_HOR_RES here as well.

Now, everywhere in your code that you use LV_HOR_RES and LV_VER_RES, the getter functions will be called and your resolution can change based on running platform.

acsplayon, 2019-01-19:
I just posted this, then saw manison's patch.  His is probably much better than mine, but it requires a lot more modification to the LVGL basic source.  I do like his mods to the to preprocessor checks in lv_obj.h

kisvegabor, 2019-01-21:
@embeddedt The mutli-display support will be a very big update and we will mean changing the most basic things in the library. I think we should start is from starch. Let's continue it in the dedicated thread: https://github.com/littlevgl/lvgl/issues/329

@acsplayon Great idea to change `LV_HOR/VER_RES` to macros! Thanks for sharing.