lvgl/lvgl #730: Moved (bigger) first line on the screen

zavovi, 2019-01-17:
Hi,
why I see the first line bigger or moved (only in some themes - alien, night, nemo)? Like on this picture: https://scontent-ams3-1.xx.fbcdn.net/v/t1.0-9/50549186_2056416067779937_8128424365881556992_o.jpg?_nc_cat=106&_nc_ht=scontent-ams3-1.xx&oh=2969e48226e8bfb159bf9cb24b792fbc&oe=5CB850C5

Thank you.

embeddedt, 2019-01-17:
@zavovi This looks like it might be a bug in your display driver.

kisvegabor, 2019-01-18:
I agree with @embeddedt it looks like a driver issue.

Try to create a button, make it draggable and drag it on the screen to see you driver works fine:
```c
    lv_obj_t * btn = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_drag(btn, true);

    lv_label_create(btn, NULL);
```

zavovi, 2019-01-18:
It seems, that it is not in driver. If I use default theme, then it is ok. Now, I tried add header status bar and tabs was moved down. This seems be ok without issue. 

About drag a button - on the ESP32 kit v4 is not touch screen. 

Thank you.

kisvegabor, 2019-01-18:
> About drag a button - on the ESP32 kit v4 is not touch screen.

Then try an animations
```c
lv_anim_t a;
a.var = btn;
a.start = 0;
a.end = 200;
a.fp = (lv_anim_fp_t)lv_obj_set_x;
a.path = lv_anim_path_linear;
a.end_cb = NULL;
a.act_time = 0;
a.time = 5000;
a.playback = 1;
a.playback_pause = 0;
a.repeat = 1;
a.repeat_pause = 0;
lv_anim_create(&a);

a.fp = (lv_anim_fp_t)lv_obj_set_y;
a.time = 1000;
lv_anim_create(&a);
```

zavovi, 2019-01-18:
The animation is ok. And the animation works like rubber (eraser) :-).  After animated button move over the bad place, then the place is now ok. It seems, that something is initialized too late...

Thank you.

kisvegabor, 2019-01-18:
Hm, interesting.

Just to be sure, can you share you `disp_flush` function?

zavovi, 2019-01-18:
My disp_flush function:

`void ex_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_p)
{
	int16_t height = (int16_t)(y2 - y1 + 1);
	int16_t width = (int16_t)(x2 - x1 + 1);
	color_t *color;

	color = heap_caps_malloc(height*width*sizeof(color_t), MALLOC_CAP_DMA);
	if(color == NULL)
		return;

	for(int i=0; i<height*width; i++)
	{
		color[i].b = color_p[i].blue;
		color[i].g = color_p[i].green;
		color[i].r = color_p[i].red;
	}

	disp_select();
	wait_trans_finish(1);
	send_data(x1, y1, x2, y2, height*width, (color_t *)color);
	disp_deselect();

	free(color);
    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
    lv_flush_ready();
}`

It is depend on TFT SPI driver.

kisvegabor, 2019-01-18:
Maybe you `free` the color earlier than the transaction is ready?

zavovi, 2019-02-07:
Hi, when I moved tab view below by 20px, then It is OK. The first 20px is ok too. I don't know why.

closed 2019-02-07 by zavovi