lvgl/lvgl #732: littlevgl not work well with cmake

KaungZawHtet, 2019-01-17:
This is my error
---------------
Undefined symbols for architecture x86_64:
  "_SDL_CreateThread", referenced from:
      _hal_init in main.c.o
  "_SDL_Delay", referenced from:
      _tick_thread in main.c.o
  "_SDL_PollEvent", referenced from:
      _main in main.c.o
  "_demo_create", referenced from:
      _main in main.c.o
  "_encoder_handler", referenced from:
      _main in main.c.o
  "_keyboard_handler", referenced from:
      _main in main.c.o
  "_lv_disp_drv_init", referenced from:
      _hal_init in main.c.o
  "_lv_disp_drv_register", referenced from:
      _hal_init in main.c.o
  "_lv_img_create", referenced from:
      _hal_init in main.c.o
  "_lv_img_set_src", referenced from:
      _hal_init in main.c.o
  "_lv_indev_drv_init", referenced from:
      _hal_init in main.c.o
  "_lv_indev_drv_register", referenced from:
      _hal_init in main.c.o
  "_lv_indev_set_cursor", referenced from:
      _hal_init in main.c.o
  "_lv_init", referenced from:
      _main in main.c.o
  "_lv_mem_monitor", referenced from:
      _memory_monitor in main.c.o
  "_lv_scr_act", referenced from:
      _hal_init in main.c.o
  "_lv_task_create", referenced from:
      _hal_init in main.c.o
  "_lv_task_handler", referenced from:
      _main in main.c.o
  "_lv_tick_inc", referenced from:
      _tick_thread in main.c.o
  "_monitor_fill", referenced from:
      _hal_init in main.c.o
  "_monitor_flush", referenced from:
      _hal_init in main.c.o
  "_monitor_init", referenced from:
      _hal_init in main.c.o
  "_monitor_map", referenced from:
      _hal_init in main.c.o
  "_mouse_cursor_icon", referenced from:
      _hal_init in main.c.o
  "_mouse_handler", referenced from:
      _main in main.c.o
  "_mouse_init", referenced from:
      _hal_init in main.c.o
  "_mouse_read", referenced from:
      _hal_init in main.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [pc_simulator] Error 1
make[2]: *** [CMakeFiles/pc_simulator.dir/all] Error 2
make[1]: *** [CMakeFiles/pc_simulator.dir/rule] Error 2
make: *** [pc_simulator] Error 2

This is my cmake configuration
-------------------------------
cmake_minimum_required(VERSION 3.12)
project(pc_simulator C)

set(CMAKE_C_STANDARD 11)

include_directories(.)
include_directories(lv_drivers)
include_directories(lv_drivers/display)
include_directories(lv_drivers/indev)
include_directories(lv_examples)
include_directories(lv_examples/lv_apps/benchmark)
include_directories(lv_examples/lv_apps/demo)
include_directories(lv_examples/lv_apps/sysmon)
include_directories(lv_examples/lv_apps/terminal)
include_directories(lv_examples/lv_apps/tpcal)
include_directories(lv_examples/lv_tests)
include_directories(lv_examples/lv_tests/lv_test_group)
include_directories(lv_examples/lv_tests/lv_test_misc)
include_directories(lv_examples/lv_tests/lv_test_obj)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_arc)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_bar)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_btn)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_btnm)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_cb)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_chart)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_cont)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_ddlist)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_gauge)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_img)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_imgbtn)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_kb)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_label)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_led)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_line)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_list)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_lmeter)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_mbox)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_page)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_preload)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_roller)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_slider)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_sw)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_ta)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_tabview)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_win)
include_directories(lv_examples/lv_tests/lv_test_stress)
include_directories(lv_examples/lv_tests/lv_test_theme)
include_directories(lv_examples/lv_tutorial/0_porting)
include_directories(lv_examples/lv_tutorial/10_keyboard)
include_directories(lv_examples/lv_tutorial/1_hello_world)
include_directories(lv_examples/lv_tutorial/2_objects)
include_directories(lv_examples/lv_tutorial/3_styles)
include_directories(lv_examples/lv_tutorial/4_themes)
include_directories(lv_examples/lv_tutorial/5_antialiasing)
include_directories(lv_examples/lv_tutorial/6_images)
include_directories(lv_examples/lv_tutorial/7_fonts)
include_directories(lv_examples/lv_tutorial/8_animations)
include_directories(lv_examples/lv_tutorial/9_responsive)
include_directories(lvgl)
include_directories(lvgl/lv_core)
include_directories(lvgl/lv_draw)
include_directories(lvgl/lv_fonts)
include_directories(lvgl/lv_hal)
include_directories(lvgl/lv_misc)
include_directories(lvgl/lv_objx)
include_directories(lvgl/lv_themes)

add_executable(pc_simulator main.c)
find_package(SDL2 REQUIRED)
include_directories(${SDL2_INCLUDE_DIR})
target_link_libraries(pc_simulator ${SDL2_LIBRARY})

I install sdl2 with this cmd
-------------------------
brew install sdl2

embeddedt, 2019-01-17:
@KaungZawHtet It seems that the only C file you are compiling is `main.c`, based on your `add_executable` line.

KaungZawHtet, 2019-01-17:
Another error occours. It is working partaily and in the middle, that err is found :

This is err:
---------
Undefined symbols for architecture x86_64:
  "_SDL_CreateRenderer", referenced from:
      _monitor_sdl_init in monitor.c.o
  "_SDL_CreateTexture", referenced from:
      _monitor_sdl_init in monitor.c.o
  "_SDL_CreateThread", referenced from:
      _monitor_init in monitor.c.o
      _hal_init in main.c.o
  "_SDL_CreateWindow", referenced from:
      _monitor_sdl_init in monitor.c.o
  "_SDL_Delay", referenced from:
      _monitor_sdl_refr_core in monitor.c.o
      _tick_thread in main.c.o
  "_SDL_DestroyRenderer", referenced from:
      _monitor_sdl_clean_up in monitor.c.o
  "_SDL_DestroyTexture", referenced from:
      _monitor_sdl_clean_up in monitor.c.o
  "_SDL_DestroyWindow", referenced from:
      _monitor_sdl_clean_up in monitor.c.o
  "_SDL_Init", referenced from:
      _monitor_sdl_init in monitor.c.o
  "_SDL_PollEvent", referenced from:
      _main in main.c.o
  "_SDL_Quit", referenced from:
      _monitor_sdl_clean_up in monitor.c.o
  "_SDL_RenderClear", referenced from:
      _monitor_sdl_refr_core in monitor.c.o
  "_SDL_RenderCopy", referenced from:
      _monitor_sdl_refr_core in monitor.c.o
  "_SDL_RenderPresent", referenced from:
      _monitor_sdl_refr_core in monitor.c.o
  "_SDL_SetEventFilter", referenced from:
      _monitor_sdl_init in monitor.c.o
  "_SDL_SetTextureBlendMode", referenced from:
      _monitor_sdl_init in monitor.c.o
  "_SDL_UpdateTexture", referenced from:
      _monitor_sdl_init in monitor.c.o
      _monitor_sdl_refr_core in monitor.c.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [pc_simulator] Error 1
make[2]: *** [CMakeFiles/pc_simulator.dir/all] Error 2
make[1]: *** [CMakeFiles/pc_simulator.dir/rule] Error 2

I change add_executables to this:
---------------------------------
add_executable(pc_simulator
        lv_drivers/display/fbdev.c
        lv_drivers/display/fbdev.h
        lv_drivers/display/monitor.c
        lv_drivers/display/monitor.h
        lv_drivers/display/R61581.c
        lv_drivers/display/R61581.h
        lv_drivers/display/SSD1963.c
        lv_drivers/display/SSD1963.h
        lv_drivers/display/ST7565.c
        lv_drivers/display/ST7565.h
        lv_drivers/indev/AD_touch.c
        lv_drivers/indev/AD_touch.h
        lv_drivers/indev/encoder.c
        lv_drivers/indev/encoder.h
        lv_drivers/indev/evdev.c
        lv_drivers/indev/evdev.h
        lv_drivers/indev/FT5406EE8.c
        lv_drivers/indev/FT5406EE8.h
        lv_drivers/indev/keyboard.c
[... 257 lines cut ...]
        lvgl/lv_themes/lv_theme.c
        lvgl/lv_themes/lv_theme.h
        lvgl/lv_themes/lv_theme_alien.c
        lvgl/lv_themes/lv_theme_alien.h
        lvgl/lv_themes/lv_theme_default.c
        lvgl/lv_themes/lv_theme_default.h
        lvgl/lv_themes/lv_theme_material.c
        lvgl/lv_themes/lv_theme_material.h
        lvgl/lv_themes/lv_theme_mono.c
        lvgl/lv_themes/lv_theme_mono.h
        lvgl/lv_themes/lv_theme_nemo.c
        lvgl/lv_themes/lv_theme_nemo.h
        lvgl/lv_themes/lv_theme_night.c
        lvgl/lv_themes/lv_theme_night.h
        lvgl/lv_themes/lv_theme_templ.c
        lvgl/lv_themes/lv_theme_templ.h
        lvgl/lv_themes/lv_theme_zen.c
        lvgl/lv_themes/lv_theme_zen.h
        lvgl/lv_conf_templ.h
        lvgl/lvgl.h
        lv_conf.h
        lv_drv_conf.h
        lv_ex_conf.h
        main.c
        mouse_cursor_icon.c)

embeddedt, 2019-01-17:
@KaungZawHtet I think that it is not linking SDL because you never set `SDL2_INCLUDE_DIR` and `SDL2_LIBRARY`. Maybe this can help: https://stackoverflow.com/questions/28395833/using-sdl2-with-cmake

Also, as a side note, you don't need to compile the header files.

KaungZawHtet, 2019-01-17:
OMG OMG OMG! It worksssssss !!!!!
Very Very Very thx and I m now so emotional to see this happening lol

embeddedt, 2019-01-17:
@KaungZawHtet No problem!

KaungZawHtet, 2019-01-17:
I think i can close this issue now

closed 2019-01-17 by KaungZawHtet

3038922, 2019-02-12:
@[KaungZawHtet](/KaungZawHtet)
Can you share your CMakeLists. txt?

KaungZawHtet, 2019-02-13:
@3038922 : Just an unoptimized CMakeLists.txt . You had better modify this :
------- 
cmake_minimum_required(VERSION 3.12)
project(pc_simulator C)

set(CMAKE_C_STANDARD 11)

include_directories(.)
include_directories(lv_drivers)
include_directories(lv_drivers/display)
include_directories(lv_drivers/indev)
include_directories(lv_examples)
include_directories(lv_examples/lv_apps/benchmark)
include_directories(lv_examples/lv_apps/demo)
include_directories(lv_examples/lv_apps/sysmon)
include_directories(lv_examples/lv_apps/terminal)
include_directories(lv_examples/lv_apps/tpcal)
include_directories(lv_examples/lv_tests)
include_directories(lv_examples/lv_tests/lv_test_group)
include_directories(lv_examples/lv_tests/lv_test_misc)
include_directories(lv_examples/lv_tests/lv_test_obj)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_arc)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_bar)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_btn)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_btnm)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_cb)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_chart)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_cont)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_ddlist)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_gauge)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_img)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_imgbtn)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_kb)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_label)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_led)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_line)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_list)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_lmeter)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_mbox)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_page)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_preload)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_roller)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_slider)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_sw)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_ta)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_tabview)
include_directories(lv_examples/lv_tests/lv_test_objx/lv_test_win)
include_directories(lv_examples/lv_tests/lv_test_stress)
include_directories(lv_examples/lv_tests/lv_test_theme)
include_directories(lv_examples/lv_tutorial/0_porting)
include_directories(lv_examples/lv_tutorial/10_keyboard)
include_directories(lv_examples/lv_tutorial/1_hello_world)
include_directories(lv_examples/lv_tutorial/2_objects)
include_directories(lv_examples/lv_tutorial/3_styles)
include_directories(lv_examples/lv_tutorial/4_themes)
include_directories(lv_examples/lv_tutorial/5_antialiasing)
include_directories(lv_examples/lv_tutorial/6_images)
include_directories(lv_examples/lv_tutorial/7_fonts)
include_directories(lv_examples/lv_tutorial/8_animations)
include_directories(lv_examples/lv_tutorial/9_responsive)
include_directories(lvgl)
include_directories(lvgl/lv_core)
include_directories(lvgl/lv_draw)
include_directories(lvgl/lv_fonts)
include_directories(lvgl/lv_hal)
include_directories(lvgl/lv_misc)
include_directories(lvgl/lv_objx)
include_directories(lvgl/lv_themes)

add_executable(pc_simulator
[... 280 lines cut ...]
        lvgl/lv_themes/lv_theme_default.c
        lvgl/lv_themes/lv_theme_default.h
        lvgl/lv_themes/lv_theme_material.c
        lvgl/lv_themes/lv_theme_material.h
        lvgl/lv_themes/lv_theme_mono.c
        lvgl/lv_themes/lv_theme_mono.h
        lvgl/lv_themes/lv_theme_nemo.c
        lvgl/lv_themes/lv_theme_nemo.h
        lvgl/lv_themes/lv_theme_night.c
        lvgl/lv_themes/lv_theme_night.h
        lvgl/lv_themes/lv_theme_templ.c
        lvgl/lv_themes/lv_theme_templ.h
        lvgl/lv_themes/lv_theme_zen.c
        lvgl/lv_themes/lv_theme_zen.h
        lvgl/lv_conf_templ.h
        lvgl/lvgl.h
        lv_conf.h
        lv_drv_conf.h
        lv_ex_conf.h
        main.c
        mouse_cursor_icon.c)

find_package(SDL2 REQUIRED)
include_directories(${SDL2_INCLUDE_DIRS})
target_link_libraries(pc_simulator ${SDL2_LIBRARIES})

kisvegabor, 2019-02-13:
@embeddedt Do you think it would be worth to add a `cmake` support?

embeddedt, 2019-02-13:
@kisvegabor Not at the moment. We have Makefiles, and support for a wide range of IDEs and platforms, so I don't think `cmake` is necessary (yet).

3038922, 2019-02-13:
@kisvegabor 
```
#设置最低的编译版本
cmake_minimum_required(VERSION 3.10)
#项目名称
project(lvgl)
#设置c/c++标准
set(CMAKE_C_STANDARD 11)#C11
set(CMAKE_CXX_STANDARD 17)#C17
set(CMAKE_CXX_STANDARD_REQUIRED ON)
#设置编译器选项

#add_compile_options("-Wnarrowing")
#头文件路径
INCLUDE_DIRECTORIES(${PROJECT_SOURCE_DIR})
#枚举头文件
file(GLOB_RECURSE INCLUDES "lv_drivers/*.h" "lv_examples/*.h"  "lvgl/*.h" "./ncrapi/*.hpp" "./*.h" "./*.hpp")
#枚举源文件
file(GLOB_RECURSE SOURCES "lv_examples/*.c" "ncrapi/*.cpp" "ncrapi/*.c"  )

#输出路径
SET(EXECUTABLE_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/bin) #设置二进制文件存放路径 
SET(LIBRARY_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/firmware) #设置各种库存放路径
#要建立的静态库
# file(GLOB_RECURSE LIBLVGL "lvgl/*.c")
# file(GLOB_RECURSE LIBDRIVERS "lv_drivers/*.c")
# add_library(lvgl STATIC ${LIBLVGL})
# add_library(drivers STATIC ${LIBDRIVERS})

#添加需要链接的库文件目录 新建好了 链接的时候用
LINK_DIRECTORIES(${LIBRARY_OUTPUT_PATH})
# #寻找sdl2
find_package(SDL2 REQUIRED SDL2)
include_directories(${SDL2_INCLUDE_DIRS})
#生成可执行文件
add_executable(main main.cpp mouse_cursor_icon.c ${SOURCES} ${INCLUDES})
#链接SDL2的静态库
target_link_libraries(main PRIVATE SDL2 liblvgl.a libdrivers.a)
#添加一个run命令
add_custom_target (run COMMAND ${EXECUTABLE_OUTPUT_PATH}/main)

```

KaungZawHtet, 2019-02-14:
btw, i did this on CLion.

kisvegabor, 2019-02-15:
@3038922 
> You have provided me with a lot of valuable experience, and I hope to contribute to the development of lvgl.

Very happy to hear that! :)

I tried your script and it worked well. :+1: 

I'm wondering if it would be possible to avoid manually editing the file to build libraries first.

3038922, 2019-02-15:
@kisvegabor
It is recommended that these two plug-ins be installed under vscode
[image]
Then select this option:`CMkae :clean rebuild`.
[image]
Then select this option:`CMkae :Build a target ` `run`
[image]
[image]
`
cmake_minimum_required(VERSION 3.10)
project(lvgl)
set(CMAKE_C_STANDARD 11)#C11
set(CMAKE_CXX_STANDARD 17)#C17
set(CMAKE_CXX_STANDARD_REQUIRED ON)

INCLUDE_DIRECTORIES(${PROJECT_SOURCE_DIR})

file(GLOB_RECURSE INCLUDES "lv_drivers/*.h" "lv_examples/*.h"  "lvgl/*.h"  "./*.h" )
file(GLOB_RECURSE SOURCES  "lv_drivers/*.c" "lv_examples/*.c"  "lvgl/*.c" )

SET(EXECUTABLE_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/bin) 

find_package(SDL2 REQUIRED SDL2)
include_directories(${SDL2_INCLUDE_DIRS})
add_executable(main main.c mouse_cursor_icon.c ${SOURCES} ${INCLUDES})
target_link_libraries(main PRIVATE SDL2 )
add_custom_target (run COMMAND ${EXECUTABLE_OUTPUT_PATH}/main)
`

kisvegabor, 2019-02-18:
@3038922 Thank you!