lvgl/lvgl #733: lv_obj_set_pos() fails for first item added to container

rrroonn, 2019-01-17:
Hi,

I have a fault, where the first item added to a container (in this case a tab's page) is not responding to the lv_obj_set_pos() call.  I have traced into the library and it does get changed but then gets reset by signals to the parent.

```
// =======================================================================
static void createZonesPage(lv_obj_t * parent)
{
    lv_cont_set_layout(lv_page_get_scrl(parent), LV_LAYOUT_OFF);
    char buff[24];
    lv_obj_t * txt;

    for (int i=0; i<Z_COUNT; i++)
    {
    	lv_style_t * style;
    	int row,col;
    	if (i%3==0)
    	{
    		style = &styleZoneOpen;
        	sprintf(buff,"Zone %02d [open]",i);
    	}
    	else if (i%5==0)
    	{
    		style = &styleZoneGreyed;
        	sprintf(buff,"Zone %02d [bypassed]",i);
    	}
    	else
    	{
    		style = &styleZoneClosed;
        	sprintf(buff,"Zone %02d [closed]",i);
    	}
    	txt = lv_label_create(parent, NULL);
    	lv_label_set_text(txt, buff);
    	col = 10+(i%2)*350;
    	row = 100+(i/2)*100;
    	styleZone(parent, txt,style);
    	lv_obj_set_pos(txt,col,row);
    	}
}
```
produces this output, where "Zone 00" is positioned at the top, rather than at offset of 100:

[image]

The first item's Y position has been reset back to the top of the screen.  Every other item is ok.  If I add a dummy label with text "" before adding the first item, then the layout works ok.

The problem occurs in this call:

```

    /*Send a signal to the parent too*/
    par->signal_func(par, LV_SIGNAL_CHILD_CHG, obj);

```

in lv_obj.c

The tabview itself is nothing special:

  ```
void gui_create(void)
{

//STYLES
    static lv_style_t style_tv_btn_bg;
    lv_style_copy(&style_tv_btn_bg, &lv_style_plain);
    style_tv_btn_bg.body.main_color = LV_COLOR_HEX(0x487fb7);
    style_tv_btn_bg.body.grad_color = LV_COLOR_HEX(0x487fb7);
    style_tv_btn_bg.body.padding.ver = 0;

    static lv_style_t style_tv_btn_rel; // released
    lv_style_copy(&style_tv_btn_rel, &lv_style_btn_rel);
    style_tv_btn_rel.body.empty = 1;
    style_tv_btn_rel.body.border.width = 0;

    static lv_style_t style_tv_btn_pr; // pressed
    lv_style_copy(&style_tv_btn_pr, &lv_style_btn_pr);
    style_tv_btn_pr.body.radius = 0;
    style_tv_btn_pr.body.opa = LV_OPA_50;
    style_tv_btn_pr.body.main_color = LV_COLOR_WHITE;
    style_tv_btn_pr.body.grad_color = LV_COLOR_WHITE;
    style_tv_btn_pr.body.border.width = 0;
    style_tv_btn_pr.text.color = LV_COLOR_GRAY;

//TABVIEW

    lv_obj_t * tv = lv_tabview_create(lv_scr_act(), NULL);
    lv_obj_t * tabHome = lv_tabview_add_tab(tv, "Home");
    lv_obj_t * tabZones = lv_tabview_add_tab(tv, "Zones");
    lv_obj_t * tabSwitches = lv_tabview_add_tab(tv, "Switches");
    lv_tabview_set_anim_time(tv,0);
    lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BTN_BG, &style_tv_btn_bg);
    lv_tabview_set_style(tv, LV_TABVIEW_STYLE_INDIC, &lv_style_plain);
    lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BTN_REL, &style_tv_btn_rel);
    lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BTN_PR, &style_tv_btn_pr);
    lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BTN_TGL_REL, &style_tv_btn_rel);
    lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BTN_TGL_PR, &style_tv_btn_pr);
//    lv_tabview_set_btns_pos(tv,LV_TABVIEW_BTNS_POS_BOTTOM);

	setupStyles();

	// MAIN PAGE

    createMainPage(tabHome);
    createZonesPage(tabZones);
    createSwitchesPage(tabSwitches);
    /*Blue bg instead of wallpaper*/
    lv_tabview_set_style(tv, LV_TABVIEW_STYLE_BG, &style_tv_btn_bg);
}
```
I have also tried the same using just a container (no tabview).

Maybe I am doing something wrong, but I cannot understand what.

thank you.

Ron

embeddedt, 2019-01-17:
@rrroonn From a quick read of https://docs.littlevgl.com/#Container:

> You can't set the child position with hor/ver fit enabled. 

Try adding:

```c
lv_cont_set_fit(lv_page_get_scrl(parent), false, false);
```

Maybe that is the problem. I will test your code later when I get a chance.

rrroonn, 2019-01-18:
Thank you, that works, provided I set the object size big enough.

Seems like an odd restriction though.

thanks again,

Ron

embeddedt, 2019-01-18:
@rrroonn The link I put explains why.

kisvegabor, 2019-02-03:
As the issue seems to be solved I close it.

If it's still not working please reopen the issue and comment here.

closed 2019-02-03 by kisvegabor