lvgl/lvgl #734: Indexed 256 color image created with Online Image to C Array Converter

fishermr, 2019-01-18:
Version 5.2.1.dev
My source image file is a PNG file with some transparent pixels.  
I load this file into your online image to c-array converted and select indexed 256 colors.
This creates a c file which I include in my project.
Looking at the contents of the c-array I can see the alpha byte for each color defined in the indexed table.
I change the color I am interested in for transparency by setting the alpha byte from FF to 00.
When i load the image to the display, transparent color stills shows in the image (opaque).

In lv_conf.h i have 
/*Image (dependencies: lv_label*/
#define USE_LV_IMG      1
#if USE_LV_IMG != 0
#define LV_IMG_CF_INDEXED   1       /*Enable indexed (palette) images*/
#define LV_IMG_CF_ALPHA     1       /*Enable alpha indexed images*/
#endif

 lv_img_set_src(obj, &img_indexed);

Is the alpha channel respected by lvgl? 

Am I missing something on how to get the alpha channel to be enabled using an indexed 256 color image?

Thanks for your help.

Looking forward to 5.3.

mheranco, 2019-01-18:
Hello, 
afaics using alpha values is not supported with indexed color images.

> "InLittelvGL these images will be always Chroma keyed."

 is written about indexed color images in the description below the image converter.

fishermr, 2019-01-18:
Thanks for the quick response.  On me.  Missed the description statement.

kisvegabor, 2019-01-18:
As @mheranco mentioned you can use Chrome keying with indexed imaged to get transparent pixels.

closed 2019-01-18 by kisvegabor