lvgl/lvgl #737: Regarding LED object in Littlevgl.

TejaswiGunti, 2019-01-19:
Can we increase the size of the LED or is it constant? If we can increase the size of the LED, where can I find it? I also want to know if I can change the shape of the LED to Oval? Could you please help me with this?

embeddedt, 2019-01-19:
@TejaswiGunti You change the size of `lv_led` with `lv_obj_set_size`, like pretty much any LittlevGL object.

I don't think there is a way to draw oval objects at this time though.

kisvegabor, 2019-01-21:
As @embeddedt `lv_obj_set_size` should work. You can set different width and height to get rounded rectangle shape (no elliptic):

```c
 lv_obj_set_size(led2, 60, 30);
```
[image]

TejaswiGunti, 2019-01-22:
Thank you so much for the help.. I got the shape rounded rectangle...

TejaswiGunti, 2019-01-22:
I have one more question if you can help I can complete my task.. I want the display(output) screen same as the size of my desktop.. I have successfully changed the resolution and it worked fine but the part of the screen is hiding behind the launcher(menu bar).. What should I do to get the output screen?