lvgl/lvgl #738: why lv_mem_alloc function can cause touch screen doesn't work?

xiaoloveyao, 2019-01-19:
lv_config.h setting:

/* Memory size which will be used by the library
 * to store the graphical objects and other data */
#define LV_MEM_CUSTOM      0                /*1: use custom malloc/free, 0: use the built-in lv_mem_alloc/lv_mem_free*/
#if LV_MEM_CUSTOM == 0
#define LV_MEM_SIZE    (80U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#define LV_MEM_ATTR                         /*Complier prefix for big array declaration*/
#define LV_MEM_AUTO_DEFRAG  1               /*Automatically defrag on free*/
#else       /*LV_MEM_CUSTOM*/
#define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
#define LV_MEM_CUSTOM_ALLOC   malloc       /*Wrapper to malloc*/
#define LV_MEM_CUSTOM_FREE    free         /*Wrapper to free*/
#endif     /*LV_MEM_CUSTOM*/

/*===================
   Graphical settings
 *===================*/

/* Horizontal and vertical resolution of the library.*/
#define LV_HOR_RES          (128)
#define LV_VER_RES          (128)
#define LV_DPI              32

/* Size of VDB (Virtual Display Buffer: the internal graphics buffer).
 * Required for buffered drawing, opacity and anti-aliasing
 * VDB makes the double buffering, you don't need to deal with it!
 * Typical size: ~1/10 screen */
#define LV_VDB_SIZE         (10 * LV_HOR_RES)  /*Size of VDB in pixel count (1/10 screen size is good for first)*/
#define LV_VDB_PX_BPP       LV_COLOR_SIZE      /*Bit-per-pixel of VDB. Useful for monochrome or non-standard color format displays. (Special formats are handled with `disp_drv->vdb_wr`)*/
#define LV_VDB_ADR          0                  /*Place VDB to a specific address (e.g. in external RAM) (0: allocate automatically into RAM; LV_VDB_ADR_INV: to replace it later with `lv_vdb_set_adr()`)*/

/* Use two Virtual Display buffers (VDB) parallelize rendering and flushing (optional)
 * The flushing should use DMA to write the frame buffer in the background*/
#define LV_VDB_DOUBLE       0       /*1: Enable the use of 2 VDBs*/
#define LV_VDB2_ADR         0       /*Place VDB2 to a specific address (e.g. in external RAM) (0: allocate automatically into RAM; LV_VDB_ADR_INV: to replace it later with `lv_vdb_set_adr()`)*/

/* Enable anti-aliasing (lines, and radiuses will be smoothed) */
#define LV_ANTIALIAS        1       /*1: Enable anti-aliasing*/

/*Screen refresh settings*/
#define LV_REFR_PERIOD      50    /*Screen refresh period in milliseconds*/
#define LV_INV_FIFO_SIZE    32    /*The average count of objects on a screen */

#if LV_MEM_CUSTOM == 0
static LV_MEM_ATTR MEM_UNIT work_mem_int[LV_MEM_SIZE / sizeof(MEM_UNIT)];    /*Work memory for allocations*/
static uint8_t * work_mem;
#endif

/**
 * Initialize your input devices here
 */
void touchpad_init(void)
{
	touch_ft6236_Configuration(); // touch screen initialize
  lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.read = touchpad_read;
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  lv_indev_drv_register(&indev_drv);
}

there are lv_mem_alloc function in the lv_indev_drv_register(&indev_drv) 
node = lv_mem_alloc(sizeof(lv_indev_t));
when I comment this function, touch screen works fine, what's the matter with this function?

xiaoloveyao, 2019-01-19:
Its the problem of lv_mem function, I should calculate needed RAM first then apply UI.

xiaoloveyao, 2019-01-19:
put mine related code here
///////////////////////////////////////////////////////////////////////
stack and heap size
///////////////////////////////////////////////////////////////////////
Stack_Size      EQU     0x00000400
Heap_Size       EQU     0x00000200

////////////////////////////////////////////////////////////////////////
lv_conf.h
////////////////////////////////////////////////////////////////////////
/**
 * @file lv_conf.h
 *
 */

/*
 * COPY THIS FILE AS lv_conf.h
 */

#if 1 /*Set it to "1" to enable content*/

#ifndef LV_CONF_H
#define LV_CONF_H

/*===================
   Dynamic memory
 *===================*/

/* Memory size which will be used by the library
 * to store the graphical objects and other data */
#define LV_MEM_CUSTOM      0                /*1: use custom malloc/free, 0: use the built-in lv_mem_alloc/lv_mem_free*/
#if LV_MEM_CUSTOM == 0
#define LV_MEM_SIZE    (80U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#define LV_MEM_ATTR                         /*Complier prefix for big array declaration*/
#define LV_MEM_AUTO_DEFRAG  1               /*Automatically defrag on free*/
#else       /*LV_MEM_CUSTOM*/
#define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
#define LV_MEM_CUSTOM_ALLOC   malloc       /*Wrapper to malloc*/
#define LV_MEM_CUSTOM_FREE    free         /*Wrapper to free*/
#endif     /*LV_MEM_CUSTOM*/

/*===================
   Graphical settings
 *===================*/

/* Horizontal and vertical resolution of the library.*/
#define LV_HOR_RES          (128)
#define LV_VER_RES          (128)
#define LV_DPI              60

/* Size of VDB (Virtual Display Buffer: the internal graphics buffer).
 * Required for buffered drawing, opacity and anti-aliasing
 * VDB makes the double buffering, you don't need to deal with it!
 * Typical size: ~1/10 screen */
#define LV_VDB_SIZE         (40 * LV_HOR_RES)  /*Size of VDB in pixel count (1/10 screen size is good for first)*/
#define LV_VDB_PX_BPP       LV_COLOR_SIZE      /*Bit-per-pixel of VDB. Useful for monochrome or non-standard color format displays. (Special formats are handled with `disp_drv->vdb_wr`)*/
#define LV_VDB_ADR          0                  /*Place VDB to a specific address (e.g. in external RAM) (0: allocate automatically into RAM; LV_VDB_ADR_INV: to replace it later with `lv_vdb_set_adr()`)*/

/* Use two Virtual Display buffers (VDB) parallelize rendering and flushing (optional)
 * The flushing should use DMA to write the frame buffer in the background*/
#define LV_VDB_DOUBLE       0       /*1: Enable the use of 2 VDBs*/
#define LV_VDB2_ADR         0       /*Place VDB2 to a specific address (e.g. in external RAM) (0: allocate automatically into RAM; LV_VDB_ADR_INV: to replace it later with `lv_vdb_set_adr()`)*/

/* Enable anti-aliasing (lines, and radiuses will be smoothed) */
#define LV_ANTIALIAS        1       /*1: Enable anti-aliasing*/

/*Screen refresh settings*/
#define LV_REFR_PERIOD      50    /*Screen refresh period in milliseconds*/
#define LV_INV_FIFO_SIZE    32    /*The average count of objects on a screen */

[... 734 lines cut ...]
			//touch: error rd touch data
	} 
	else 
	{
		// fine, touch data arrived, process
		// process clicks:
		uint32_t touch_count = buf.touches & 0xf;
		if (touch_count > 0) 
		{
			// always use first touch point
//			uint8_t ev = buf.points[0].event >> 6;
			// clear touch status
//			touch_event.event_id = 0;
			// calculate lcd pixel coords
			touch_event.x = (buf.points[0].xhi & 0x0F) << 8  | (buf.points[0].xlo);
			touch_event.y = (buf.points[0].yhi & 0x0F) << 8 | (buf.points[0].ylo);
//			touch_event.y = 128 - touch_event.y;		
			// touch screen test
//			OLED_ShowNum(1,30,5,touch_event.x,3,WHITE,BLACK);		
//			OLED_ShowNum(1,60,5,touch_event.y,3,WHITE,BLACK);		
		}
	}
}

//////////////////////////////////////////////////////////////////////////////////////////////

embeddedt, 2019-01-19:
@xiaoloveyao Maybe you're running out of heap space.

xiaoloveyao, 2019-01-20:
Program Size: Code=89218 RO-data=125054 RW-data=5236 ZI-data=118044

xiaoloveyao, 2019-01-20:
> @xiaoloveyao Maybe you're running out of heap space.

finally, I found the vital problem, when I read touch coordinate from ft6236(touch screen IC) by stm32 IIC bus, 
////////////////////////////////////////////////////////////////////////////////////////////////////
uint8_t Sensors_I2C_ReadRegister(unsigned char Address, unsigned char RegisterAddr, unsigned short RegisterLen, unsigned char *RegisterValue)
{
	HAL_I2C_Master_Transmit(&I2cHandle, Address, RegisterValue, 1, 1000);
	
	return HAL_I2C_Master_Receive(&I2cHandle, Address, RegisterValue, RegisterLen, 1000);
}
/////////////////////////////////////////////////////////////////////////////////////////////////////////
Input:
Address = 0x70
RegisterValue = 0x00

the schedule of IIC read data from register is
send write addr -- RegisterValue -- read addr -- data
send 0x70 -- 0x00 -- 0x71 -- data
I found that when I send RegisterValue (0x00) after finish sending address(0x70), the actual RegisterValue change from 0x00 to 0xE9,  the RegisterValue is always changed by littlevGL function, I don't know how to fix this problem, any some advise?
///////////////////////////////////////////////////////////////////////////////////////
change happens when I send data to DR
////////////////////////////////////////////////////////////////////////////////////////
HAL_StatusTypeDef HAL_I2C_Master_Transmit(I2C_HandleTypeDef *hi2c, uint16_t DevAddress, uint8_t *pData, uint16_t Size, uint32_t Timeout)
{
...
    /* Prepare transfer parameters */
    hi2c->pBuffPtr    = pData;
    hi2c->XferCount   = Size;
    hi2c->XferOptions = I2C_NO_OPTION_FRAME;
    hi2c->XferSize    = hi2c->XferCount;
...
      /* Write data to DR */
      hi2c->Instance->DR = (*hi2c->pBuffPtr++);
      hi2c->XferCount--;
      hi2c->XferSize--;
...
}
picture is get from Oscilloscope
[image]
[image]

xiaoloveyao, 2019-01-20:
I made a mistake in this function:
////////////////////////////////////////////////////////////////////////////////////////////////////
uint8_t Sensors_I2C_ReadRegister(unsigned char Address, unsigned char RegisterAddr, unsigned short RegisterLen, unsigned char *RegisterValue)
{
HAL_I2C_Master_Transmit(&I2cHandle, Address, RegisterValue, 1, 1000);

return HAL_I2C_Master_Receive(&I2cHandle, Address, RegisterValue, RegisterLen, 1000);
}
/////////////////////////////////////////////////////////////////////////////////////////////////////////
replace RegisterValue to RegisterAddr here, problem fixed.
HAL_I2C_Master_Transmit(&I2cHandle, Address, RegisterAddr , 1, 1000);

closed 2019-01-20 by xiaoloveyao