lvgl/lvgl #741: Is there a simpler way to do

3038922, 2019-01-21:
```
#include "ncrapi/system/systemData.hpp"
#include "ncrapi/userDisplay/userDisplay.hpp"
static double upDownFlag = 0;

std::vector<lv_obj_t *> updownBtn;
static lv_res_t saveAction(lv_obj_t *btn)
{
    (void)btn;
    size_t index = lv_tabview_get_tab_act(userDisplay->displayObj[OBJ_BTNM_SON]);
    //获取当前活动页面的索引
    std::string name = ((lv_tabview_ext_t *)userDisplay->displayObj[OBJ_BTNM_SON])->tab_name_ptr[index];
    bool flag = sysData->saveData(sysData->configData[name], name); //TODO 这里有BUG 记得修复
    if (!flag)
        sysData->addDebugData({"数据保存失败"});
    return LV_RES_INV;
}
static lv_res_t upDownAction(lv_obj_t *btnm, const char *txt)
{
    (void)btnm;                                                         /*Unused*/
    lv_obj_t *lab = static_cast<lv_obj_t *>(lv_obj_get_free_ptr(btnm)); //获取按钮栏关键的项目栏指针
    lv_obj_t *val = static_cast<lv_obj_t *>(lv_obj_get_free_ptr(lab));  //获取项目栏关键的值栏目的指针
    std::string text = lv_label_get_text(val);                          //搜索值栏目
    if (text.find(".") != std::string::npos)                            //如果找到.
        upDownFlag = 0.01;
    else
        upDownFlag = 1;
    size_t index = lv_tabview_get_tab_act(userDisplay->displayObj[OBJ_BTNM_SON]);
    std::string name = ((lv_tabview_ext_t *)userDisplay->displayObj[OBJ_BTNM_SON])->tab_name_ptr[index];
    auto tempData = sysData->configData[name];                                                  //获取当前索引
    double temp = ncrapi::stringToNum<double>(sysData->read(tempData, lv_label_get_text(lab))); //从动态数据库里读取值
    if (!strcmp(txt, "U"))
        temp += upDownFlag;
    if (!strcmp(txt, "D"))
        temp -= upDownFlag;
    std::ostringstream oss; //主要为了去掉小数点后多余的0 默认3位
    oss << temp;
    sysData->setData(tempData, lv_label_get_text(lab), oss.str());
    lv_label_set_text(val, oss.str().c_str());
    return LV_RES_INV;
}

namespace ncrapi {

void UserDisplay::createConfig(lv_obj_t *parent)
{

    if (displayObj[OBJ_BTNM_SON] == nullptr)
        displayObj[OBJ_BTNM_SON] = lv_tabview_create(parent, nullptr);
    std::vector<lv_obj_t *> tabs;
    std::vector<lv_obj_t *> key;
    std::vector<lv_obj_t *> values;
    static const char *updown[] = {"U", "D", ""};
    for (auto &it : sysData->obj) //遍历项目名字
    {
        size_t posX = 0, posY = 0;

        /*创建TABS*/
        tabs.push_back(lv_tabview_add_tab(displayObj[OBJ_BTNM_SON], it->showName().c_str())); //添加项目
        lv_page_set_scrl_fit(tabs.back(), false, false);
        lv_page_set_scrl_height(tabs.back(), 400); //设置页面可滚动部分的高度

        //创建文本页面
        DataConfigList tempData = sysData->configData[it->showName()]; //使用无序哈希查找名字匹配 这啥鸟可能会每次都搜索...
        for (auto &it1 : tempData)
        {                                                               //在当前TABS上创建项目
            key.push_back(lv_label_create(tabs.back(), nullptr));       //KEY就一个
            lv_obj_set_pos(key.back(), posX, posY);                     //设置key的位置
            lv_label_set_text(key.back(), it1.at(0).c_str());           //设置KEY的文本 //静态文本有问题
            for (auto it2 = (it1.begin() + 1); it2 != it1.end(); it2++) //遍历内容VALUE有些可能是2 3个的比如马达 it.beg()是key                                                                 //遍历内容VALUE有些可能是2 3个的比如马达 it.beg()是key
            {
                values.push_back(lv_label_create(tabs.back(), nullptr));   //新建vals Lab
                updownBtn.push_back(lv_btnm_create(tabs.back(), nullptr)); //VAL和UPDOWN可以好几个
                lv_obj_set_free_ptr(updownBtn.back(), key.back());         //创建关键指针.KEY关联UPDOWN
                lv_obj_set_free_ptr(key.back(), values.back());            //创建关键指针.值栏关联KEY
                lv_btnm_set_map(updownBtn.back(), updown);
                lv_obj_set_size(updownBtn.back(), 60, 25);

                lv_obj_set_pos(values.back(), posX + 90, posY);      //设置value位置
                lv_obj_set_pos(updownBtn.back(), posX + 110, posY);  //设置btn位置
                lv_btnm_set_action(updownBtn.back(), upDownAction);  //设置updown按钮动作
                std::ostringstream oss;                              //主要为了去掉小数点后多余的0 默认3位
                oss << *it2;                                         //获取值
                lv_label_set_text(values.back(), oss.str().c_str()); //设置按钮的值
                posX += 100;
            }
            if (key.size() % 2 == 0 && it1.size() == 2) //key表示偶数换行 it.szie()表示每行总共有2个数一个KEY 一个VAL
                posX = 230;
            else
            {
                posX = 0;
                posY += 20;
            }
        }
    }

    //创建保存按钮
    lv_obj_t *saveBtn = lv_btn_create(displayObj[OBJ_BTNM_SON], nullptr);
    lv_obj_set_pos(saveBtn, LV_HOR_RES - 140, LV_VER_RES - 50);
    lv_obj_set_size(saveBtn, 50, 25);
    lv_obj_t *saveLab = lv_label_create(saveBtn, nullptr);
    lv_label_set_text(saveLab, "save");
    //lv_btn_set_action(saveBtn, LV_BTN_ACTION_CLICK, saveAction);

    //退出按钮
    createExitBtn(OBJ_BTNM_SON);
} // namespace ncrapi
} // namespace ncrapi
```

turoksama, 2019-01-23:
I think the variable free_num may help.

kisvegabor, 2019-01-24:
In `keyboard_open_close` should use the `text_area` paramter and call `lv_kb_set_ta(kb, text_area)` to assign the keyboard to the clicked Text area.

3038922, 2019-01-24:
i'm using  `lv_obj_set_free_num_` solve this problem temporarily. But it seems that your method is better.

closed 2019-01-24 by 3038922