lvgl/lvgl #742: Output Display screen..

TejaswiGunti, 2019-01-22:


TejaswiGunti, 2019-01-22:
My questions is I want the display(output) screen same as the size of my desktop.. I have successfully changed the resolution and it worked fine but the part of the screen is hiding behind the launcher(menu bar).. What should I do to get the output screen?

embeddedt, 2019-01-22:
@TejaswiGunti You'll need to provide more details. Are you using the PC simulator, or an embedded display?

TejaswiGunti, 2019-01-23:
I am using PC simulator.. When I run my code the output screen comes up showing my application, I have changed the resolution in lv_conf.h and I am getting the size of output screen same as my desktop but some part of the screen is hiding behind the menu bar on the left side..

kisvegabor, 2019-01-23:
Then it should be an SDL setting in monitor.c
https://wiki.libsdl.org/SDL_WindowFlags
Try to add `SDL_WINDOW_ALWAYS_ON_TOP ` instead of the last `0` [here](https://github.com/littlevgl/lv_drivers/blob/master/display/monitor.c#L291).

TejaswiGunti, 2019-01-23:
Please do not close the issue, I am working on it.. I will update you the status..

TejaswiGunti, 2019-01-23:
Thanks for letting me know about SDL_WindowFlags.. This helped me.. I used SDL_WINDOW_FULLSCREEN in my application...

kisvegabor, 2019-01-23:
Great! :)

closed 2019-01-23 by kisvegabor