lvgl/lvgl #749: Monochrome theme with orange colors

MichaelVLV, 2019-01-23:
Hi,

I have initialized monochrome theme and create couple screeenes (with buttons and lv_list).

But I see that there are orange border present and non-monochrome backgroung.

LV_LIST:
[image]

Buttons:
[image]

Could you please advice which parts of style I should check? Why it is orange and brown/grey in mono theme?

embeddedt, 2019-01-23:
@MichaelVLV Are you using any group features?

MichaelVLV, 2019-01-23:
yes, all buttons in one group and elements of list in another

kisvegabor, 2019-01-23:
You should set LV_COLOR_DEPTH to 1 in lv_conf.h. This way littlevgl will set a monochrome style instead of the orange by default.

MichaelVLV, 2019-01-23:
@kis

> You should set LV_COLOR_DEPTH to 1 in lv_conf.h. This way littlevgl will set a monochrome style instead of the orange by default.

I suppuse I cant show colored images it this case? 
I changed LV_COLOR_DEPTH to 1 and now I got crash when trying to load picture.

My operete order: load and show colored image, then load screen#1 with buttons in monochrome style and usethis syle futher.

Is it possible? Can I change orange checkbox and list background to some another colors (white/black at least) manually?

kisvegabor, 2019-01-24:
I don't know what caused the crash but you can change the orange color. You need set a new `style mod callback` with `lv_group_set_style_mod_cb`.

Here is default orange style mod. function as  reference: https://github.com/littlevgl/lvgl/blob/master/lv_core/lv_group.c#L419

MichaelVLV, 2019-01-24:
@kisvegabor Thanks, thats hepled 👍

closed 2019-01-24 by MichaelVLV