lvgl/lvgl #751: What is the best way to organize large GUI

harteware07, 2019-01-24:
Hello All,

I am currently in the process of designing a large GUI with complex menue structure. 
My question is, what would be the best way to organize the `lv_obj_t *objects`?
would it be better to create them all in the beginning and hide them with `lv_obj_set_hidden` when they are not needed?
Or is it possible to create them, draw them and free them afterwards?
thank you for your input!

MiSimon, 2019-01-24:
I organize them in "views", creating the objects that i need to display for the given context and free them before i switch to the next "view".
This works quite well and allows a hirachy with an overall layout.

embeddedt, 2019-01-24:
@harteware07 I think it depends on how much memory you have available at one time. If you don't have constrained resources, creating them all and hiding them would definitely be faster than deleting them. Keep in mind that if all your menus are similar, you might be able to re-use objects to save space.

kisvegabor, 2019-01-25:
If I work on systems with a lot of memory and a few screens I usually create all the screens and load them if needed. This way the state of the screens are saved by LittlevGL and you don't need to save/restore them manually.

However on smaller devices or if there are a lot of screens I create the screen on the fly when they are required to save RAM.

harteware07, 2019-01-25:
I now create everything on beginning and work with `lv_obj_set_hidden` to manage the screen. This works well. If I have to save RAM later on I can easily change this to delete/create on the fly. 

one other related question:

in my init code i created a screen with my default color
`	mainScreen = lv_obj_create(NULL, NULL);
	lv_scr_load(mainScreen);
	lv_obj_set_style(mainScreen, &mainPageStyle);
`
but now all my icons/labels/etc are covered by mainScreen. I tried to put `mainScreen` instead of `lv_scr_act()` when i create the element with `lv_imgbtn_create(lv_scr_act(), NULL);`
but this crashes the simulator. 

- What would be the correct way to do it? 

- Can i put objects in the foreground?

thanks to you all!

kisvegabor, 2019-01-25:
Only one screen can be loaded at one time. So if you load an other screen the old won't be visible.

You can create screen sized obejcts instead of screen and hide them:
```c
lv_obj_t * holder1 = lv_obj_create(lv_scr_act(), NULL);
lv_obj_set_size(holder1, LV_HOR_RES, LV_VER_RES);

lv_imgbtn_create(holder1, NUL);
....
```

closed 2019-02-26 by harteware07