lvgl/lvgl #754: Notification from text area when the Enter key is hit

manison, 2019-01-24:
I have a one line text area and need to perform an action when a user hits the Enter key (i.e. confirms the text in the area). Is there a way to be notified of such an event? I can't find something like `lv_ta_set_action(o, LV_TA_ACTION_CONFIRM, cb)`.

embeddedt, 2019-01-24:
You need to do the callback on the `lv_kb` object, not the text area. There is no API for that right now but you should be able to do it this way:

1. The keyboard is really a `btnm` object. You need to override the current callback with `lv_btnm_set_action` but still call the old callback (retrievable with `lv_btnm_get_action`).
2. In your callback, you can check whether the `txt` argument is `Enter`, that means the user pressed Enter.

Here's an example that should get you started. Note that it's not tested.

```c
static lv_btnm_action_t old_kb_action;

static lv_res_t my_kb_action(lv_obj_t * kb, const char * txt)
{
    /* Call the old action (this will send the text to the text area) */
    lv_res_t res = old_kb_action(kb, txt);
    if(res == LV_RES_INV)
        return LV_RES_INV;
    
    if(!strcmp(txt, "Enter"))
    {
        /* Do something */
    }
}

. . .

/* 'kb' refers to your keyboard object */

old_kb_action = lv_btnm_get_action(kb);
lv_btnm_set_action(kb, my_kb_action);

```

manison, 2019-01-24:
Thanks. I din't mention the problem is that I don't have an onscreen keyboard (`lv_kb` object). I only have physical keyboard (keypad driver).
So it seems there is no out of the box solution to this problem right now. Am I correct?

kisvegabor, 2019-01-25:
There is no out of the box solution right now but it's still possible. Have a look at this article: https://blog.littlevgl.com/2018-12-13/extend-lvgl-objects

What you need is to change the signal function of the Text area. If `LV_SIGNAL_CONTROLL` and `\n` character arrives do not call the original Text area signal but do what you need.

manison, 2019-01-28:
Works like a charm, thanks.

closed 2019-01-28 by manison

rrroonn, 2019-02-13:
I think that it would be useful to tweek the keyboard code a bit so that the "enter" key functions as an OK  if the textarea is a single line.  At the moment it inserts a "\n" which I guess is silently ignored.

BTW, I used the workaround above - works as expected, thanks.

kisvegabor, 2019-02-14:
Good idea! Can you send a Pull request?