lvgl/lvgl #755: Issue in with lv_task

Masi2017, 2019-01-24:
Hi All,
I have an issue that sometime happens, since it doesn't happen all the time it's hard to find it . It happens mostly once I am trying to switch between  the menus ( using touch) . I'd appreciate if you can help me with that, here is some screenshot when problem arises and CPU crashes  ( I am using ESP32) .
[image]
[image]
[image]

Any idea what can cause this?

BrianPugh, 2019-01-24:
could be a multitude of things. Try updating to dev-5.3 if possible first to see if that solves your problem.

kisvegabor, 2019-01-24:
There were some bugfixes related to `lv_task` in `dev-5.3` so it definitely worth a try. 

I suggest trying the same UI in PC simulator because there you will have better debug opportunities. E.g. you will see where the code halts exactly.

Masi2017, 2019-01-28:
Hi all,
Thanks, I'll try dev-5.3 and see if problem is solved.

closed 2019-02-28 by stale[bot]