lvgl/lvgl #757: About the running refresh speed of this UI library

longdadouzi, 2019-01-25:
I have successfully run this UI on STM32F407VET6, but it seems to be a bit slower. Can make people feel slow, whether it is caused by animation. Where else can I improve its speed?

kisvegabor, 2019-01-25:
Have you enabled Complier optimization?

longdadouzi, 2019-01-26:
After listening to what you said, I turned the compiler optimization to the highest level but it didn't seem to have any effect. Whether it is configuration can be faster. It only keeps refreshing the screen, or it will only brush the star when it changes.

longdadouzi, 2019-01-29:
I don't know how to upload a video into Git, it seems that it does not support these formats, but I can send it to you by email.
How to use DMA to increase its speed?
/* Use two Virtual Display buffers (VDB) parallelize rendering and flushing (optional)
  * The flushing should use DMA to write the frame buffer in the background*/
I don't seem to understand

kisvegabor, 2019-01-30:
You can upload the video to youtube as a private video.

DMA will send the pixels to the display while you CPU can do other things (e.g. rendering the next part). But to do this 2 buffers are required:
1. Where the current frame is rendered
2. Form where the previous frame is being sent

longdadouzi, 2019-02-19:
There was a holiday in front, and there was no timely reply.

longdadouzi, 2019-02-20:
@kisvegabor

DevriesL, 2019-02-20:
@longdadouzi I thinkd you need to improve the transmission speed from STM32 to display(like DMA etc.), do it by your self :smile:

kisvegabor, 2019-02-20:
It seems related to this issue: https://github.com/littlevgl/lvgl/issues/847#issuecomment-464589949

longdadouzi, 2019-02-21:
#define LV_HOR_RES          (320)
#define LV_VER_RES          (240)
When I use VDB_SIZE to 320, I get the driver to be normal, but increase VDB_SIZE to 7680, why the original driver is not available.

And I don't understand the ypical size: ~1/10 screen function, is it necessary to set VDB_SIZE to one tenth of the pixel?

longdadouzi, 2019-02-21:
Thank you for your suggestion, I am using FSMC now, I think it may be the buffer size as he said.
@DevriesL

longdadouzi, 2019-02-21:
@kisvegabor 
I have to have this problem, and it is very embarrassing to run into this state by chance. As shown：
[image]
[image]

kisvegabor, 2019-02-22:
The shifted image seems to be an issue of your driver.

What happens if you try to change only the gib number (text) in the middle?

kisvegabor, 2019-03-02:
In the past few days [LittlevGL's Forum](https://forum.littlevgl.com) has launched to discuss questions there. See this [Announcment](https://forum.littlevgl.com/viewtopic.php?f=3&t=5).

If the issue still persists please open a topic for it on the Forum.

closed 2019-03-02 by kisvegabor