lvgl/lvgl #760: ESP8266 crash

defconhaya, 2019-01-25:
Hello,

I am using Platformio with Arduino core for ESP8266.
The settings in lv_conf.h are:
```
#define LV_MEM_SIZE    (16U * 1024U) 
#define LV_VDB_SIZE         (10 * LV_HOR_RES) 
```
The code I am running is the basic example:
```#include <Arduino.h>
#include <lvgl.h>

#include "SPI.h"
#include "TFT_eSPI.h"

#include <XPT2046_Touchscreen.h> 

XPT2046_Touchscreen ts(TOUCH_CS); 

TFT_eSPI tft = TFT_eSPI(); 

const float togglePeriod = 0.05; //seconds

//Ticker function to keep the GUI running
static void lv_tick_handler(void) {
  lv_tick_inc(togglePeriod * 1000);
}

void disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_p) {
  
  lv_flush_ready();
}

bool my_input_read(lv_indev_data_t *data)
{
  static uint16_t prev_x, prev_y;
  TS_Point p = ts.getPoint();

  if (p.z > 0) 
  {
    data->point.x = p.x;
    data->point.y = p.y;
    data->state = LV_INDEV_STATE_PR;
    prev_x = data->point.x;
    prev_y = data->point.y;
  } else {
    data->point.x = prev_x;
    data->point.y = prev_y;
    data->state = LV_INDEV_STATE_REL;
  }

  return false;        /*No buffering so no more data read*/
}

void setup() {
  Serial.begin(115200);

  tft.begin();
  tft.setRotation(1);

  lv_init();

  /*Initialize the display*/
  lv_disp_drv_t disp_drv;
  disp_drv.disp_flush = disp_flush;
  lv_disp_drv_register(&disp_drv);

  /* Initialize input device touch */
  lv_indev_drv_t indev_drv;
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read = my_input_read;
  lv_indev_drv_register(&indev_drv);

  lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
  lv_label_set_text(label, "Hello Arduino!");
  lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);

}

void loop() {

  
  lv_task_handler();
  delay(10);
}
```

And I'm getting the following exception:

```
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
Fatal exception 0(IllegalInstructionCause):
epc1=0xfeefeffe, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (0):
epc1=0xfeefeffe epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3ffffb30 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffcd0:  feefeffe feefeffe feefeffe feefeffe
3ffffce0:  feefeffe feefeffe feefeffe feefeffe
3ffffcf0:  feefeffe feefeffe feefeffe feefeffe
3ffffd00:  feefeffe feefeffe feefeffe feefeffe
3ffffd10:  feefeffe feefeffe feefeffe feefeffe
3ffffd20:  feefeffe feefeffe feefeffe 40207dae
3ffffd30:  00000000 0009013f feefeffe feefeffe
3ffffd40:  3fff3b18 0000013f 00000140 00000009
3ffffd50:  3fff55e4 00000000 00000000 3ffe84ec
3ffffd60:  00000000 3ffe84fc 3fff56ac 3ffffd80
3ffffd70:  00000000 000000ff 0000ffff 40203058
3ffffd80:  00000000 00ef013f 00000002 00000000
3ffffd90:  00000000 3ffe84fc 3fff56ac 4020855e
3ffffda0:  00000000 0000000e 00000079 4020865d
3ffffdb0:  3fff54fc 3ffffea8 3fff9d1c 0000ffff
3ffffdc0:  00000002 00000000 00000000 00ef013f
3ffffdd0:  3fff9d94 00000010 00000000 0000000e
3ffffde0:  000000ff 3ffe84fc 3ffffe64 40208721
3ffffdf0:  00000000 0013007a 00000000 00ef013f
3ffffe00:  3fff9d1c 00000010 00000000 4020111d
3ffffe10:  00000000 0013007a 00000026 3ffe84fc
3ffffe20:  3fff9d1c 00000000 0000013f 00000000
3ffffe30:  000000ef 000000f0 00000026 4020865d
3ffffe40:  00000004 00000000 00000002 3fff9d1c
3ffffe50:  3ffffea8 00000001 3fff54ec 40201c51
3ffffe60:  00000000 0014007b 00000079 4020865d
3ffffe70:  0000000e 00000000 00000002 00000000
3ffffe80:  3fff9d1c 3fffff10 3fff54ec 40201d8a
3ffffe90:  00000000 0013007a 00000000 00ef013f
3ffffea0:  00000000 00ef013f 00000000 0009013f
3ffffeb0:  3fffff10 000000ff 3fff54ec 40201bd6
3ffffec0:  00000000 00ef013f 00000000 3fff39c8
3ffffed0:  3fffff10 000000ef 3fff54ec 40201e45
3ffffee0:  3fffff10 000000ff 3fff54ec 40201cf8
3ffffef0:  3fffff40 00000002 3fff5644 3fff39c8
3fffff00:  0000000a 000000ef 3fff39d8 40201eef
3fffff10:  00000000 0009013f 00000000 4020111d
3fffff20:  006e0063 008100dd 3fff39d8 40202125
3fffff30:  00000081 3fff54ec 3fff5644 402011d0
3fffff40:  00000000 00000000 3fff5644 00000009
3fffff50:  00000000 0000000a 3fff568c 40206925
3fffff60:  0000000f 00000000 3ffe8a48 3fff54ac
3fffff70:  3fff946d 00000000 3fff54cc 402059af
3fffff80:  00000000 3fff9468 00000000 40205a97
3fffff90:  4020f3c4 feefeffe feefeffe 3fff94dc
3fffffa0:  3fffdad0 00000000 3fff94d4 4020f511
3fffffb0:  feefeffe feefeffe 3fff94d4 40211c94
3fffffc0:  feefeffe feefeffe 3ffe8660 40100a31
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
```
Here is the decoded exeception:

```
Exception Cause: 0  [Illegal instruction]

0x40207dae: indev_search_obj at lv_indev.c:?
0x40203058: lv_draw_rect at ??:?
0x4020855e: indev_proc_task at lv_indev.c:?
0x4020865d: lv_draw_aa_get_opa at ??:?
0x40208721: lv_draw_label at ??:?
0x4020111d: delete_children at lv_obj.c:?
0x4020865d: lv_draw_aa_get_opa at ??:?
0x40201c51: lv_init at ??:?
0x4020865d: lv_draw_aa_get_opa at ??:?
0x40201d8a: lv_obj_signal at lv_obj.c:?
0x40201bd6: refresh_childen_style at lv_obj.c:?
0x40201e45: lv_obj_design at lv_obj.c:?
0x40201cf8: lv_obj_get_coords at ??:?
0x40201eef: lv_refr_get_top_obj at lv_refr.c:?
0x4020111d: delete_children at lv_obj.c:?
0x40202125: lv_refr_area_part_vdb at lv_refr.c:?
0x402011d0: lv_obj_del at ??:?
0x40206925: lv_ta_set_cursor_pos at ??:?
0x402059af: lv_mem_monitor at ??:?
0x40205a97: lv_task_create at ??:?
0x4020f3c4: Adafruit_SPITFT::writePixel(unsigned short) at ??:?
0x4020f511: Adafruit_SPITFT::invertDisplay(unsigned char) at ??:?
0x40211c94: _printf_common at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf_i.c:90
0x40100a31: wdt_feed at ??:?
0x4010f000: ?? ??:0
```

kisvegabor, 2019-01-25:
Hi,
`lv_indev_drv_init(&indev_drv);` and `lv_disp_drv_init(&disp_drv);` is missing.

See this a reference: https://github.com/littlevgl/pc_simulator_sdl_platformio/blob/master/lv_pio_sdl/src/main.c#L39

By the way, if you are on Linux you can easily try LittlevGL with Platformio in a simulator. See https://github.com/littlevgl/pc_simulator_sdl_platformio

embeddedt, 2019-01-25:
@defconhaya Great!

closed 2019-01-25 by embeddedt