lvgl/lvgl #763: Align object to lv_list element

BrianPugh, 2019-01-26:
Is there a way to align an object to a lv_list element and have it scroll with it? I'm using `lv_list` as a menu, and I want a `lv_sw` along side one of the options.

turoksama, 2019-01-26:
Let's try to get that element object and create your object in it? It is a button actually.
I want a lv_ta along side of it, too.

embeddedt, 2019-01-26:
@BrianPugh At the moment for customized lists it can be better to use the page object and create your own containers inside. You can use styles to make it look very similar or identical to a list.

turoksama, 2019-01-26:
Unfortunately, we can not align object to lv_list element.

BrianPugh, 2019-01-26:
I solved my problem by doing the following. The key was figuring out that the objects within a list element button were organized via `LV_LAYOUT_ROW_M`:

1. Create the list element, getting the `lv_btn` object
2. Get the `lv_label` in the `lv_btn` object via `lv_list_get_btn_label`
3. Get the label width via `lv_obj_get_width`
4. Set the width to be n pixels shorter via `lv_obj_set_width` to make space for my n pixel wide `lv_switch`.
5. Create the switch object with the btn as its parent.

closed 2019-01-26 by BrianPugh

turoksama, 2019-01-26:
Hmm...That's a tricky one. Well done!

turoksama, 2019-01-26:
@BrianPugh 
As it is organized via LV_LAYOUT_ROW_M, will it(list element) not align left when shorter & look ugly?

BrianPugh, 2019-01-26:
So each element was laid out like
```
+-----------------+
|img|label        |
+-----------------+
```
With the label filling up the entire remaining width (image optional). Since i make the label `sw` shorter, everything is laid out as it should be.
```
+-----------------+
|img|label    |sw |
+-----------------+
```

kisvegabor, 2019-01-27:
I'd do it this way:
```c
    lv_obj_t * list = lv_list_create(lv_scr_act(), NULL);
    lv_obj_t * btn = lv_list_add(list, SYMBOL_HOME, "A very very long list element text comes here", NULL);
    lv_obj_set_width(lv_list_get_btn_label(btn), 50);   /*Make the label shorter*/

    lv_obj_t * sw = lv_sw_create(btn, NULL);
    lv_obj_set_protect(sw, LV_PROTECT_POS);     /*Do not affect is by  the layout or other automatic positioning*/
    lv_obj_align(sw, NULL, LV_ALIGN_IN_RIGHT_MID, -5, 0);
```

[image]