lvgl/lvgl #768: CI to verify PR automatically

amirgon, 2019-01-27:
@kisvegabor - Did you consider adding CI such as [Travis](https://travis-ci.org/)?
How do you ensure today that changes to lvgl don't break the build? I noticed that `lv_mpy` got broken due to some lvgl commits. CI could warn us that something like that won't happen.

A very popular CI is [Travis](https://travis-ci.org/). It has the following advantages:

- Well integrated with GitHub Pull Request build flow. When someone opens a PR, Travis checks it automatically and tells us (after some time) if all builds passed, or what failed.
- It supports C builds on Linux.
- It can even cross-compile to another architecture (ARM for example) and run it (emulated) on QEMU!  
[This](https://www.tomaz.me/2013/12/02/running-travis-ci-tests-on-arm.html) and [this](https://medium.com/@wolfv/cross-compiling-arm-on-travis-using-clang-and-qemu-2b9702d7c6f3) are examples of how to do it. Probably more effort to set this up vs. simple Linux build, though.
- It's free for open source projects!

In my opinion, when lvgl scales up and used by more architectures and projects (such as Micropython, for example), it would be very helpful to have some automatic tool for CI which is integrated with GitHub.

BrianPugh, 2019-01-27:
to be clear, Travis would just check if the build is passing, not if a PR has some sort of negative interaction with a component of LVGL, right?

kisvegabor, 2019-01-28:
I don't have experience with CI tools so far. 

I'm working with LittelvGL every day so I see if something is not working. But I don't use the Makefiles so didn't notice the missing `lv_gc.c` file.

To work with Travis we should create a project with `lvgl` and `lv_examples` submodules and create a Makefile for it?

kisvegabor, 2019-01-28:
I found this: https://docs.travis-ci.com/user/languages/c/ I got the basic idea.

It seems to either a dedicated project is still required or a [pc_simulator_sdl_eclipse](https://github.com/littlevgl/pc_simulator_sdl_eclipse) should be modified to be compiled without SDL.

amirgon, 2019-01-28:
@kisvegabor it would be very helpful if Travis would also build `lv_mpy`.
There are couple of things that can only be checked there (like gc). After building it we could also run a small python script and make sure it passes with no errors.

amirgon, 2019-01-28:
> to be clear, Travis would just check if the build is passing, not if a PR has some sort of negative interaction with a component of LVGL, right?

@BrianPugh In principle Travis can also run the built result. What we run and what would it check is up to us.  
But the simplest thing to do is just to make sure build is passing, and not attempt to run anything.

embeddedt, 2019-01-28:
> LittlevGL works if you don't register any displays. Just does nothing when the disp_flush function should be called.

I can't verify it right now, but I tried doing that very recently and there was a `NULL` pointer error relating to the fact that there was no display.

kisvegabor, 2019-01-29:
@embeddedt You are right, it really wasn't working. I've just fixed it.

embeddedt, 2019-01-29:
@amirgon @kisvegabor The problem with CI on a highly configurable project like this one is that you need a way to test all possible configurations. For example, in #772, that error only shows up with multi-language support disabled.

amirgon, 2019-01-29:
> @amirgon @kisvegabor The problem with CI on a highly configurable project like this one is that you need a way to test all possible configurations. For example, in #772, that error only shows up with multi-language support disabled.

@embeddedt Actually this is an excellent example why we need CI.  
A CI could check **multiple** configurations for each PR. It's harder and time consuming to do manually, but quite easy automatically.  
We can configure a matrix of configurations to check. For example, "everything enabled", "everything enabled except feature X" for each feature X etc.

puzrin, 2019-01-30:
> To work with Travis we should create a project with lvgl and lv_examples submodules and create a Makefile for it?

Though this is technically possible, this will prevent collaborators to write tests because change 2 repos in parallel is not convenient. I would suggest to keep tests local.

Also, prior to run tests, it's common to execute linter & formatter to check obvious errors and follow coding style.

I'm not familiar with C tools, but this link seem to have actual info https://invisible-island.net/personal/lint-tools.html

> It seems either a dedicated project is still required or pc_simulator_sdl_eclipse should be modified to be compiled without SDL.

No. Try to formalize, what do you really need to test right here. If this repo is not about SDL driver, using it as test's dependency is wrong direction.

May be, you need a special dummy driver for tests (raw buffer, display mock), to check pixels after some operations. But this does not require anything related to SDL.

You may need to select testing framework, and mocking framework to simplify write/execute things.

Which tests (except lint/format/buld) can be done at start? Inspect issues history, and add tests if:

- some mistake happened twice
- some mistake was very annoying and you would not like it to repeat

---

Talk about testing can be very long. I tried to point how to get maximal profit with minimal efforts at first step. Feel free to ask any questions.

kisvegabor, 2019-01-30:
I think finally I understood the concept and why is it really good.

In summary:
1. We should test only lvgl not a whole project where lvgl is only a submodule
2. We should use different configs in `lv_conf.h`. Now `lv_conf.h` is included in the source files so lvgl can be compiled as a standalone project. However I can imagine a flag (-DLV_CONF_NOT_PRESENT) to include [lv_conf_checker.h](https://github.com/littlevgl/lvgl/blob/dev-5.3/lv_conf_checker.h) instead of `lv_conf.h`. This way the settings can come from the Makefile.
3. Knowing that lvgl can be compiled without warnings is already useful
4. To make better tests we should have an `lv_test` folder. I don't think pixel level testing is required. Testing against memory leaks and similar things would be more useful.

Please correct me if I still see it wrong.

puzrin, 2019-01-30:
> However I can imagine a flag (-DLV_CONF_NOT_PRESENT) to include lv_conf_checker.h instead of lv_conf.h. This way the settings can come from the Makefile.

IMHO that's too magical and still hardcoded. I'd suggest use explicit definition everywhere (pass to build option config path, search path and so on).

In general - yes, you may need multiple configs for different things.

> To make better tests we should have an lv_test folder.

Yes. Note, project tends to use too many `lv_` prefixes, and too much of content of root level. Consider change structure on next major release to more classic (something like `/src`, `inc`, `test`, `vendor`, ...)

> I don't think pixel level testing is required. Testing against memory leaks and similar things would be more useful.

I did not intended to attract your attention to pixel testing. I explained principle, what to test, and how to get things isolated from foreign libraries.

There are 2 major kinds of tests, used everywhere:

- Unit tests (test particular objects)
- Integration tests

Both are useful, but if you start to implement all, that will never ends. It's more rationale to make minimal steps i described + make sure it's easy to write unit tests in future.

amirgon, 2019-01-30:
@kisvegabor 
> 1. We should test only lvgl not a whole project where lvgl is only a submodule

That was not my thinking.  
I imagined `lvgl` tested as a submodule of `lv_micropython`. Building lvgl alone will not catch problems that are exposed only when building with Micropython compilation flags, headers included by Micropython etc.  
Building lvgl in `lv_micropython` would also allow running Micropython script. Such script can make sure `gc.collect()` works fine and there are no memory leaks, that all gc-root variables were defined correctly etc.

> 2\. We should use different configs in `lv_conf.h`.

Maybe a matrix of configurations that enables/disabled features. Can be done with a simple script.

puzrin, 2019-01-30:
IMHO lv_microphython can have it's own tests, not affiliated to lvgl tests anyhow.

kisvegabor, 2019-01-30:
@puzrin 

> IMHO that's too magical and still hardcoded. I'd suggest use explicit definition everywhere (pass to build option config path, search path and so on).

I'd like to keep the current relative path inclusion too because in some IDEs (e.g. Arduino IDE) you have only very limited build config options (or they are hidden very deeply). It works out of the box if you place `lv_conf.h` to the correct place. `LV_CONF_INCLUDE_SIMPLE` is added to enable placing `lv_conf.h` to any place. See: https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_arc.h#L17
But if you build `lvgl` alone `lv_conf.h` will be required.  That's why I suggested to replace it with `lv_conf_checker.h` which is in the lvgl folder.

> Sorry, i did not intended attract your attention to pixel testing. I tried to explain principle - what to test and how to keep things isolated and self-sufficient.

I highlighted the pixel level testing because we don't want this then display and input device drivers are not required at all.

> Both are useful, but if you start to implement all, that will never ends. It's more rationale to make minimal steps i described + make sure it's easy to write unit tests in future.

I agree, let's start with a simple case instead full featured tests.

@amirgon 

> That was not my thinking.
I imagined lvgl tested as a submodule of lv_micropython. Building lvgl alone will not catch problems that are exposed only when building with Micropython compilation flags, headers included by Micropython etc.
Building lvgl in lv_micropython would also allow running Micropython script. Such script can make sure gc.collect() works fine and there are no memory leaks, that all gc-root variables were defined correctly etc.

This way the `lvgl` submodule needs to be continuously updated in the micropython repo, right? If yes, then it would be better to get feedback "immeditelly" after a PR is merged.

In this regard, I agree with @puzrin. In general, the binding should have their own tests. Once we introduce CI for lvgl it shouldn't rely on a specific binding because:
- the binding can hide some bugs
- we are no testing the bare library
- it would be more difficult to localize an issue (e.g. memory leak)
- the project becomes fragile

So I agree and I'd like to have tests (even very detailed tests) in micropython but we should have lvgl's own tests in C.

puzrin, 2019-01-30:
> But if you build `lvgl` alone `lv_conf.h` will be required. That's why I suggested to replace it with `lv_conf_checker.h` which is in the lvgl folder.

I don't understand you. "Tests" are standalone programs. Why can't they use `LV_CONF_INCLUDE_SIMPLE` and pass directory path with config in makefile (or any-other-build-system-you-prefer)? Tests do not depend on IDE.

kisvegabor, 2019-01-30:
> I don't understand you. "Tests" are standalone programs. Why can't they use LV_CONF_INCLUDE_SIMPLE and pass directory path with config in makefile (or any-other-build-system-you-prefer)? Tests do not depend on IDE.

So as discussed above we should have an `lv_test` folder (or with another name) where there will be a `test_main.c` with a `main()` function and an `lv_conf.h`. Correct so far?
If yes, then it would be easier to pass the config option with Makefile, isn't it? In addition, having two 2 lv_conf.h (one for tests and for the user) might cause issues if the wrong lv_conf.h is chosen for the build. It can happen if all directories are added as include path.   

Sorry, if all these are obvious for you I try my best to understand it and find the best way. :)

For first I suggest making only a test which compiles lvgl without any tests but with various configs.

puzrin, 2019-01-30:
Hm... seems i see what you mean. You are right, via makefile only one value can be set without kludges. But workaround possible.

There can be many test files. Each test is independent and has it's own `main()` (that's not mandatory, but very common). For example, in platformio you should create subfolders for each independent test (`/test/test_arc/test_arc.c`).

Workaroung is to pass 2 additional include directories for each test build:

1. Possible local config location, `test/<test_name>/config/`.
2. Default location.

So, if you need specific config for some test - create an subfolder for. If it's missed, default one will be used.

amirgon, 2019-01-30:
> In this regard, I agree with @puzrin. In general, the binding should have their own tests. Once we introduce CI for lvgl it shouldn't rely on a specific binding because:

@kisvegabor Let's discuss a specific example:   
Someone opens a PR where he adds a new function declaration, but doesn't provide a function definition.  
As long as no one calls this function, **this will pass build and tests for "bare" lvgl**, but **lv_micropython would break** because the binding would generate a wrapper to an undefined function.   

- If lvgl PR would trigger a test for this, it will warn the comitter about this before this PR is accepted.
- If only `lv_micropython` would check it, as you suggest, the PR will be accepted with a change that breaks Micropython. Only when someone tries to update the lvgl submodule in `lv_bindings` and tries to build `lv_micropython` we would know that something is wrong.

My point is that changes in **lvgl** can break **micropython**. Therefore I suggested checking Micropython when lvgl PR is submitted. This is not a rare case, it happens again and again (#772, #761, #756). In each of these cases `lv_micropython` was broken because of an lvgl commit, but "pure" lvgl didn't break.

Doesn't it make sense to catch it right when it happens, on the PR?

> * the binding can hide some bugs
> * we are no testing the bare library

I'm not saying not to test the bare library, I think you should test it.  
I'm suggesting to check `lv_micropython` as well, in addition to it.

> * it would be more difficult to localize an issue (e.g. memory leak)
> * the project becomes fragile

You can still decide to accept the PR although `lv_micropython` fails, it's your decision. But at least we would know something is wrong right away, and not long time after it happened.

embeddedt, 2019-01-30:
@amirgon @kisvegabor Based on the discussion,it looks we need all the tests to be run off of the `lvgl` repository, not `lv_micropython`. The tests in this repo have to check the binding with every pull request. @amirgon brings up a good point about making sure that the binding is always compatible.

puzrin, 2019-01-30:
I see no problem to add MP bindings tests call. That's a bit strange, but does not breaks isolation.

As far as i remember, the only way to define custom testing pipeline with complicated dependencies is to setup independent CI server or specific hooks bot. Adding sub-project tests call is much more simple and has no serious negative side effects (i did such things in my projects).

kisvegabor, 2019-01-30:
**Micropython tests**
I see and agree. It's good to have tests in micropython (or other bindings) too because:
  - it can reveal issues which are not visible in C
  - we should know if it's still working
The technical side (using generic free tools) is questionable as @puzrin highlighted

**Shall we place the test folder inside lvgl or not?**
@puzrin said it's better to have them inside.  
@embeddedt said we should separate them.   
I see the advantage and disadvantage both of them.

**Get started**
What would be an absolute minimal environment to get started?

In the next few weeks, I'd like to add the most important features to `dev-6.0`. So I can't immerse in setting up a test environment right now. Somebody would like to manage it at least until we get started with it?

embeddedt, 2019-01-30:
One more thing I thought of.

@amirgon's example shows a function that is declared but not defined. IMHO we shouldn't rely on the Micropython binding to find things like that. There must be some type of test designed to detect those kinds of issues.

I'm not saying we shouldn't use the MicroPython binding as a test, just pointing out that it shouldn't be the only test for C language issues like @amirgon showed.

kisvegabor, 2019-01-30:
> This Makefile can run a git clone for lv_micropython and update the lvgl sub module to include that PR.

@amirgon It's fine if it works this way.

@embeddedt 
> @amirgon's example shows a function that is declared but not defined. IMHO we shouldn't rely on the Micropython binding to find things like that. There must be some type of test designed to detect those kinds of issues.

A script like this can do a lot of things.  For example, @rreilink's script highlighted some inconsistencies in paramter names and enum names: https://github.com/littlevgl/lvgl/issues/682 and https://github.com/littlevgl/lvgl/issues/681  
The question is: is better to have a dedicated fully customizable script to do this or rely on Micropython's script (which is more simple once it already checks a lot of things but probably not everything)? I think we can decide it later once we have a running environment.

embeddedt, 2019-01-30:
> The question is: is better to have a dedicated fully customizable script to do this or rely on Micropython's script (which is more simple once it already checks a lot of things but probably not everything)? I think we can decide it later once we have a running environment.

This is exactly what I was bringing up. Waiting for a running environment sounds good to me.

puzrin, 2019-01-30:
> Shall we place the test folder inside lvgl or not?
> @puzrin said it's better to have them inside.

I did not suggested to keep extra test sources inside lvgl. I suggested to call those. Small script, cloning repo and executing "make test" there. Dirty but cheap.

Of cause, `lv_micropython`'s tests should in it's repo - in the same repo where tested sources are located.

See [example](https://github.com/nodeca/nodeca.forum/blob/9460fbbe44347a4eba505d251f60a2ac0b955308/Makefile#L27-L39) from my project. When someone run submodule tests, makefile downloads full project and run tests for all submodules. Reason is exactly the same - one submodule can crash another one (and we do a lot of parallel changes).

@amirgon I suggested to do exactly as you said. There are no technical difficulties, only difficulties with my english :)

> What would be an absolute minimal environment to get started?

1. Src formatting checker (indents, spaces, ...) - helps to keep the same coding style when many people participate.
2. Code analyzer (linter) (unused vars, suspicious patterns, ...). Most advanced is Coverity, but it's free version available only via github hooks. Something local would be very useful, to know about obvious problems immediately (before you waste time for debug). https://invisible-island.net/personal/lint-tools.html
3. Testing framework. Scan `test` folder, automate tests build, provides advanced assertions, sugar to simplify tests writing. Also provides reporters for different presentation formats (for terminal console, for IDE and other automation tools). https://stackoverflow.com/questions/65820/unit-testing-c-code
4. Mocks/spies. Probably you may not need those immediately. Those may be a part of testing framework, but may be an independent libs too.
5. Coverage reporter (to see is specific code lines really tested or not, very useful at start, when you have no tests at all).
6. Build framework. I develop under linux only, and `make` was always enougth for all my simple needs. But i think a lot of lvgl collaborators will need to execute builds on Windows. I don't know is `make` available there or not. Seems MS finally did available bash shell on windows ("Windows System for Linux").

embeddedt, 2019-01-31:
@puzrin

> Src formatting checker (indents, spaces, ...) - helps to keep the same coding style when many people participate.

I think `astyle` is set up to format the code, but it has to be done by hand.

> Code analyzer (linter) (unused vars, suspicious patterns, ...)

In my own projects, I turn on a very large number of warnings in GCC. That takes care of a lot of issues.

For more complex errors, I've found `valgrind` to be very effective and useful. Maybe that can be integrated into the test system.

> Build framework.

If the tests are done on a hosted server, isn't that server already running Linux?

kisvegabor, 2019-01-31:
Thank you for all of you for sharing your thoughts about this topic. It seems we have clarified the goals and possibilities.  The next step is to get started! 

@amirgon 
> If I find some time for that I'll bring up an example of Travis building lv_micropython.

Awesome! Curiously waiting for it!

puzrin, 2019-02-06:
https://github.com/ThrowTheSwitch/Unity

After some dig of available testing frameworks, Unity seems a good choice

- Actively developed, easy to use, MIT and so on.
- Used in platformio - may be convenient in long-term.

Also, the same guys develop https://github.com/ThrowTheSwitch/Ceedling to automate tests and coverage reports. But this may be not needed at start (can be hardcoded in `test/Makefile`)

kisvegabor, 2019-02-07:
Seems like an easy to use framework! 

I was thinking about what can we test in LittlevGL and concluded there are not so many things. I mean the majority of the things happens on the display and it wouldn't worth to see if something is really drawn and the color/size/shape etc is really correct. Emulate a touchpad an action to see a button is really clicked, changes state etc. So it would be very difficult the achieve high coverage because it's difficult to test the GUI part. In addition, writing and maintaining tests is a huge extra burden. 

So I'm not sure we really need so many tests. Just a few for the basic (non-GUI) parts should be enough. What do you think?

puzrin, 2019-02-07:
Unit tests are low level. They don't test full event path "from user touch to button appearance change". They test multiple small reactions of each object. When every small object is valid, combination is valid (that's very simlified explanation, but correct in general).

For example, you can be safe about regressions, and so on.

> What do you think?

I'm not TDD fanboy :). If you don't feel need of tests - then you really don't need those. When fuckups frequency become high - you will know.

I can suggest only one constructive thing - when you write your code, try to imagine, how you could write unit tests for it. If that's possible - then your code is ok.

rreilink, 2019-02-07:
Matplotlib does, as a part of their unit tests, something which is IMHO quite nice: they compare the bitmap output to a predetermined reference. This could be useful to test regressions in e.g. the drawing routines. In lvgl one could capture the frame buffer and check equivalence.

kisvegabor, 2019-02-08:
@puzrin 
> I can suggest only one constructive thing - when you write your code, try to imagine, how you could write unit tests for it. If that's possible - then your code is ok.

It's really a useful takeaway,

@rreilink good to know, thank you. Now, we haven't got so many problems due to the lack of unit tests. Most of the issues come from the library can't be built in some configurations. So it can be important in a later stage.

puzrin, 2019-02-08:
- http://www.testingtoolsguide.net/listing-category/static-analysis/?licence_type=open_source - list of tools for static analyzis. 
  - Clang Static Analyzer & cppcheck seems most advanced. CSA should be available with clang without build.
- https://github.com/ThrowTheSwitch/Unity/blob/master/test/Makefile - example how to organize coverage check (but without upload reports to coverall.io)

For coding style i'd suggest [ClangFormat](https://clang.llvm.org/docs/ClangFormat.html). Mostly because availablie with clang without manual builds.
  - Note, it already supports some presets (LLVM, Google, Chromium, Mozilla, WebKit). May be worth to select some, and relax restrictions you don't like.
  - It also supports imlined dorectives like `// clang-format off` & `// clang-format on` to skip cheks. For example, when you wish to type too long line.

kisvegabor, 2019-02-10:
@puzrin Thank you for this collection!

puzrin, 2019-02-28:
https://github.com/littlevgl/lv_i18n is almost ready, and can be a live sample, how tests & coverage reports helps.

- For C code - 1 bug + some unreachable branches found (that's significant percent for small code size).
- Huge number of bugs in JS.

All this reduce support tail (when you get bug reports next 1-2 month and have to fix those). I don't think this can fix in advance 100% of bugs, but reduce those in tens of times, and protects from regressions.

---

I also played with code analyzer and formatter. IMHO clang tools are good enougth.

- `scan-build` did not showed anything interesting, because codebase is too small (and well tested), but it would be interesting to integrate.
- `clang-format` - perfect tool, but i don't know your preferences and can't push with my own. Ideally, someone of core devs should spend some hours and create `.clang-format` config for all projects of this org. This will be used then for automatic format (where possible) and validation in CI.

embeddedt, 2019-03-21:
Not stale.

kisvegabor, 2019-06-27:
There was no activity here for 3 months so I close this issue.

If you are interested in creating some tests for LittlevGL, please comment here.

closed 2019-06-27 by kisvegabor

embeddedt, 2019-12-18:
Discussion of this as a more generic release strategy issue continues in #1305.