lvgl/lvgl #773: NRF52 porting

wiesener, 2019-01-30:
Hi,

first of all great project. Great documentation.

Now we would like to use your library for an NRF52840 based GUI project with 240x320 2.4 inch TFT without touch. Do you think it is feasible? Any suggestions or tutorials to study besides your documentation and porting guide?
Cheers

kisvegabor, 2019-01-30:
Hi,

1 MB flash, 256 kB RAM and 64 MHz are definitely enough to make nice things. Adding LittlevGL should be simple if you have a display driver. 

It seems like an interesting project! Please keep us updated! :)

delcomp, 2019-02-01:
Hey wiesener, I'm currently working an an update for our XG8800 family. That's a combo of an LCD and cpu board. One of the LCD's is a 2.4" 240x320 and the cpu board can also accept an NRF52840 (other CPU's are ATXMega, ESP32, Arduino MKR, Particle). I just ported the ILI9341 & RA8875 to our hardware and I'm starting to add support for the littlevgl library. Although my focus is currently on the larger LCD's with the hardware accelerate RA8875 driver, maybe your project could also use our hardware. Just message me and I'll send you some info. Because all this is in the works, there is only preliminary info on our website www.x-graph.be, but currently nothing mentioned on the NRF82840, only the MKR version.

kisvegabor, littlevgl is a gem. It lacks deeper hardware support for chips like the RA8875. I'm currently trying to integrate the hardware acceleration functions of the RA8875 in littlevgl, keep you updated.

wiesener, 2019-02-01:
Yes sounds interesting. I will check the updates.

kisvegabor, 2019-02-01:
@delcomp "External GPUs" are not directly supported by LittlevGL right now, but there is a workaround.  
 Have look at this discussion: https://github.com/littlevgl/lvgl/issues/512

closed 2019-03-01 by stale[bot]