lvgl/lvgl #776: Input device driver registration has been broken

manison, 2019-01-30:
Commit a883f0b39a93a941ee21e988a43223e666c46d25 broke registration of multiple input device drivers. The particular change is here: https://github.com/littlevgl/lvgl/commit/a883f0b39a93a941ee21e988a43223e666c46d25#diff-a53b4f200ec1da6041f5757b0ea47fe9R52

The line added to `lv_indev_drv_init` in this commit resets the head of driver linked list everytime a new driver is being initialized. This results in previous drivers to be lost and when a new driver is registered by `lv_indev_drv_register` it is added to the head.

cc @amirgon

kisvegabor, 2019-01-30:
Good catch! As `indev_list` is global variable it is initialized to `NULL` automatically. So I think that line can be deleted. It's the same for `lv_disp_drv_init`. @amirgon, do you agree?

manison, 2019-01-30:
> So I think that line can be deleted.

I think so however I don't know the relation to MicroPython garbage collector.

kisvegabor, 2019-01-30:
Just to be sure lat's wait @amirgon's answer.

amirgon, 2019-01-30:
@manison @kisvegabor 

### First - regarding the relation to Micropython gc: `indev_list` is a *gc root*.  

A *gc root* is a global variable that **nothing is referring to it**. That means that there is no dynamically allocated memory block or stack frame which contains a pointer to it.  
gc must be aware of all gc roots to work correctly. For example, if it doesn't know about `indev_list`, it will not identify anything pointing to the first element of the linked list and will deallocate it on the next gc collection cycle.  

I tried to identify all gc roots in lvgl project (I hope I caught them all).  
This is required because lvgl on Micropython uses the Micropython dynamic memory allocation functions,  which means that every block of memory allocated with `malloc` is subject to garbage collection.  

I also wanted to give an opportunity for Micropython users to call functions that refer to gc roots. In case of `lv_indev_drv_register`, Micropython doesn't support registering input device driver **yet**, but there is no reason why not to add such support in the future. (the only thing missing is the ability to define a callback which is not related to `lv_obj`. @kisvegabor this is related to [our discussion](https://github.com/littlevgl/lvgl/issues/557#issuecomment-443455949) about callbacks, and the missing canonic `user_data` or `context` argument. If we had that, registering input device driver in Micropython would have been supported today).  

The method Micropython provides to declare a *gc root* and register it to Micropython gc is to define it in specific Micropython struct inside Micropython and access it there instead of as a global variable. There could have been better ways to do it but they chose this one to save memory probably.  
To achieve this in lvgl I replaced the global `indev_list` by `LV_GC_ROOT(indev_list)` and added `indev_list` to `LV_GC_ROOTS` which is [added to Micropython gc](https://github.com/littlevgl/lv_micropython/blob/38d1e3cbace3bd2526139dbcf0aaa2c7f4ada4f6/ports/unix/mpconfigport.h#L301) when building Micropython.

### Now about initializing `indev_list` to `NULL`

Probably every gc root needs to be explicitly initialized. After adding gc support to lvgl, gc roots are no longer global variables that can be initialized during compilation time, and Micropython does not attempt to initialize gc roots.  
@manison says correctly that `indev_list` cannot be initialized in `lv_indev_drv_init` so we need to find another way to initialize it.    
Today I initialize most of the gc_roots [on `lv_init`](https://github.com/littlevgl/lvgl/commit/a883f0b39a93a941ee21e988a43223e666c46d25#diff-a770841b193605fd4cc6254b586607a3R65).  

@kisvegabor is it ok to initialize `indev_list` there as well?  
Can we assume drivers are only added **after** `lv_init` was called?

Another option is to still initialize it in `lv_indev_drv_init` **but only once**.  
We can do something like this:

```c
static bool indev_list_initialized = false

void lv_indev_drv_init(lv_indev_drv_t * driver)
{
    if (!indev_list_initialized){
        LV_GC_ROOT(indev_list) = NULL;
        indev_list_initialized = true;
    }
    ...
```

closed 2019-01-30 by kisvegabor