lvgl/lvgl #779: Odd behavior while dragging window

embeddedt, 2019-01-31:
When dragging a window that is a child of the page object, the page seems to be picking up the drag events as well.

To reproduce, add `lv_win_set_drag(win, true)` below [line 267](https://github.com/littlevgl/lv_examples/blob/06b952e564af4c114846d7ff048a2a279efdec00/lv_tests/lv_test_theme/lv_test_theme_1.c#L267) in `lv_test_theme_1.c`. When you try to drag the window you'll notice the page scrolling as well.

I'll try to find the root cause.

kisvegabor, 2019-01-31:
I tried it and it seems it's because of `fit` property of the page.

Try to add:
```c
    lv_page_set_scrl_fit(parent, false, false);
    lv_page_set_scrl_height(parent, LV_VER_RES);
```

embeddedt, 2019-01-31:
@kisvegabor That appears to have worked. Is this caused by the container fit issue mentioned in the documentation?

> You **can't set the child position with hor/ver fit enabled**. Let's imagine what happens. If you change the position of the only child when fit is enabled the the container will move/fit "around" the new position. If you have more objects on a container then you can align them relative to each other. As a workaround you can create a small transparent object on the container. It will fix the container to not "follow" the children.

kisvegabor, 2019-01-31:
Exactly!

embeddedt, 2019-01-31:
@kisvegabor Alright, that makes sense. It's not really a bug so I'll close this issue now.

closed 2019-01-31 by embeddedt