lvgl/lvgl #781: Suggestion: reduce readme's top buttons

puzrin, 2019-01-31:
IMHO readme's top is overloaded with buttons. I understand desire to show "awesomeness" of project, but this looks like tons of animated smiles :). You already has webassembly builds, that's much better to make impression. Also, some button's links go to image. That's useless.

Buttons are usually expected to contain some super-important info. For example:

- CI tests status
- Coverage %
- Version number
- Link to gitter chat (to keep tracker more clean from simple question)

I do not insist, only share personal opinion/experience. Feel free to close if it's not useful for you.

kisvegabor, 2019-01-31:
Thank you for the feedback!

About the badges:
- it's really useless to make them links but I didn't find a  way to remove the link
- I think this info is also important to immediately know if this project is for you or not (embedded gui c/c++)
- Licence, version are also disaplyed
- Of course CI tests results, coverage etc. are also important. I'd like to add them once we launch CI

About the animated demo:
- I think it's also important for a user to immediately  what he can do with this library
- As it's a GUI library and image tells more than 1000 of words.

I made some research about how the "modern" repo's Readme look like and tried my best to make something eye-catching and informative. 
I've looked around here to get some idea: https://github.com/matiassingers/awesome-readme
Platformio's main repo has a similar readme: https://github.com/platformio/platformio-core

Looking at it again maybe it's really a little bit crowded.

I'd be happy to hear your feedback and, of course, I'll change it based on the feedbacks.

embeddedt, 2019-01-31:
Have a look at #782.

kisvegabor, 2019-01-31:
@embeddedt 
Merged, thank you!

What do you think about the new design?

Meanwhile, I made more light-weighted.

puzrin, 2019-01-31:
> What do you think about the new design?

Is this question about readme or about themes?

- Readme structure looks perfect - short enougth to be not boring, bit interesting to motivate reading and clicking links.
- Themes are much better than everything i could find in other embedded gui libs. But still a bit "not modern" if apply highest expectations. Let me explain. Interface design is very difficult work. So, many companies/communities write documents "interface design guide" (use this words to google) with detailed description how to place every object type on panels, popup dialogs, ... It's rationale to select one you like and clarify current look&feel. For example, "Metro" from Miscrosoft is very nice and unobtrusive for small displays IMHO. If you don't like metro - there are other ones from gnome, kde, mac os, and so on...

C47D, 2019-01-31:
Offtopic: There's a typo on the README (master branch) _And share with your firends_ i guess it should be _And share with your friends_.

kisvegabor, 2019-01-31:
@puzrin 
> Proposals: ...

It is really redundant this way. I'm thinking about your proposal. 

> Readme structure looks perfect - short enougth to be not boring, bit interesting to motivate reading and clicking links.

Great!

> Themes are much better than everything i could find in other embedded gui libs. But still a bit "not modern" if apply highest expectations. Let me explain. Interface design is very difficult work. So, many companies/communities write documents "interface design guide" (use this words to google) with detailed description how to place every object type on panels, popup dialogs, ... It's rationale to select one you like and clarify current look&feel. For example, "Metro" from Miscrosoft is very nice and unobtrusive for small displays IMHO. If you don't like metro - there are other ones from gnome, kde, mac os, google, and so on...

It's really difficult to do it well. I learned the basics of graphic design but I know it's far from perfect. Somebody with more experience in design could update the themes. However, I think if your expectations are high you need to design you UI yourself to meet your exact needs. For example, in my work, I work with a designer who sends screenshots and I implement them.

@C47D 
> Offtopic: There's a typo on the README (master branch) And share with your firends i guess it should be And share with your friends.

Fixed thank you.

puzrin, 2019-01-31:
> Somebody with more experience in design could update the themes. However, I think if your expectations are high you need to design you UI yourself to meet your exact needs. For example, in my work, I work with a designer who sends screenshots and I implement them.

I work actively with hired designers, and understand very well cost of such work, and count of required iteration. My comment is not to criticize current look, but how to get maximal effect with limited human resources. I think this is possible.

Suggestion is: switch to ready existing modern "interface design guide" instead of inventing custom one from scratch. If such approach is acceptable, i can try to prepare new issue with list of available choices. Then we can vote for preferable one, and check if it's technically possible with current lvgl skinning features. Then me or someone else can prepare commit with necessary changes & recommendations.

embeddedt, 2019-01-31:
@puzrin @kisvegabor Possibly we could update the material theme to match Material Design more closely. I don't think it would be an insurmountable task.

kisvegabor, 2019-01-31:
I agree the themes can be updated and even new ones can be created based on existing interface design guides. Let's open a new issue for it and continue there.

@puzrin 
I updated readme and removed some badges. It really looks better now. I'll rename licence.txt -> LICENCE.txt and add links to the badges.

puzrin, 2019-01-31:
- IMHO centering looks annoing (difficult to read). I'd return alignement back.
- Consider remove line `Star the project if you like it! And share with your friends`. Looks a bit "old-fashioned" (IMHO). Or, at least, remove yellow star. This steel attention from more importang line above (with bold links).

puzrin, 2019-01-31:
> centering is visually better because the "top" elements have a very different width

This looks like "layout design done by professional programmer" :). At least, i had no intent to ask making things worse :)

> I still like to encourage people to star the repo, but I agree it didn't look so good. Any idea?

People should post because they wish to tell everybody about awesome product. Not because you ask. Indirect motivation. Webassembly demo, available in one click, is very serious reason to impress.

You can see my org's repos. I have some very "starred", and this did not helped me anyhow. People selected projects and starred those, because they have no alternatives of comparable level. And lvgl has no comparable alteratives in it's area, i've checked very carefully :)

> I can make them equal only if I add extra space at the top. Maybe a smaller shadow would be better.

Just drop this shadow at all and keep standard space. Common look, left-aligned is more easy to consume.

kisvegabor, 2019-02-01:
Some updates:
- I changed the display frame to a more realistic one.
- Reduced a the shadow

I'd like to keep the center alignment. I asked my designer colleague who also thinks it really looks better.

@puzrin 
I suggest forking lvgl and update the Readme as you think. This way we can compare them.

puzrin, 2019-02-01:
No problem. See https://github.com/puzrin/lvgl. A did a bit more polish. Use any part you wish if you like it.

kisvegabor, 2019-02-01:
Thank you! To be honest it looks better than I thought! :sweat_smile: 
I still need to think about it.

> ... Looks like collection of buzzwords :). 

That was the goal :) Make it as informative and impressive as possible.

> Header + animated image already contain this info in more descriptive form.

Having a description is crucial because it will be displayed in a Google search:

[image]

puzrin, 2019-02-02:
Hm...

Search "lvgl":

[image]

Search "embedded gui":

[image]

puzrin, 2019-02-02:
Updated readme:

https://github.com/puzrin/lvgl

Moved text below image, try to visually hide without remove.

puzrin, 2019-02-02:
\+links order +header

I suggest use anything you prefer and close this. Issue is too primitive to spending more time for it.

kisvegabor, 2019-02-02:
> I suggest use anything you prefer and close this. Issue is too primitive to spending more time for it.

You are right. The Reame was really too crowded. I think it's okay now. From now it's only a matter of taste.

Thank your for the suggestions!

closed 2019-02-02 by kisvegabor