lvgl/lvgl #784: v5.3 is released

kisvegabor, 2019-02-01:
Hi,

v5.3 is released today! https://github.com/littlevgl/lvgl/releases/tag/v5.3

Thank you very much for all the contribution! :+1: 

There are a few things to do now:
1. Update the documentation
2. Update the projects

If your planning to work on the documentation or the projects, please comment here to clearly see the progress.

I already updated [pv_simulator_sdl_eclipse](https://github.com/littlevgl/pc_simulator_sdl_eclipse) project and will update [stm32f429_disco_no_os_sw4stm32](https://github.com/littlevgl/stm32f429_disco_no_os_sw4stm32).

seyyah, 2019-02-01:
Congratulations :tada: 

> There are a few things to do now:

3. Update the examples: lv_spinbox, lv_lang

amirgon, 2019-02-01:
> If your planning to work on the documentation or the projects, please comment here to clearly see the progress.

I would like to add some Micropython Binding documentation, but first I'm planning to [make some changes](https://github.com/littlevgl/lv_binding_micropython/issues/5#issue-405579627) in `lv_micropython_binding`  in order to make it more scalable and portable.

kisvegabor, 2019-02-01:
@seyyah Updated, thanks!

@amirgon @embeddedt 
Awesome!

kisvegabor, 2019-02-03:
I have updated the [stm32f429_disco_sw4stm32](https://github.com/littlevgl/stm32f429_disco_no_os_sw4stm32) and the [esp32_ili9341](https://github.com/littlevgl/esp32_ili9431) projects.

MCF, 2019-02-06:
I have update the [pc_simulator_sdl_visual_studio](https://github.com/littlevgl/pc_simulator_sdl_visual_studio) project.

embeddedt, 2019-02-06:
@MCF Great! Thanks!

amirgon, 2019-02-08:
I updated `lv_binding_micropython` and `lv_micropython` to reflect lvgl head.
I also updated the documentation on `lv_binding_micropython` [README file](https://github.com/littlevgl/lv_binding_micropython/blob/master/README.md), it contains explanations about the design and python examples.  

@kisvegabor Do you think we should have something on the formal LittlevGL docs, or is the README enough? If you do think we should have something on the docs, where should we put it and how should we structure it?

kisvegabor, 2019-02-09:
@amirgon 
I will write a Release post on the blog as I can where I will list the news (like Mpy binding) too. I think it would be great to have a sperate post about micropython binding because it is
1. sharable
2. linkable
3. searchable

I mean something simple about:
- what is micropython
- why is it good to use micropython
- quick set up guide for the unix build 
- run some test code + screenshots

I'm planning to advertise the binding in the feature list in the readme and on the website.

kisvegabor, 2019-03-02:
I think who want to participate in the version update already wrote here.
Thank you for your contribution! :+1:

Closing

closed 2019-03-02 by kisvegabor