lvgl/lvgl #785: v6.0 general discussion

kisvegabor, 2019-02-01:
**Schedule**  2019 summer

**Development branch** `dev-6.0`

**Main new features**
- [ ] Generic action handling #316 
- [ ] Multi-display support with run time display parameter settings feature #329 #109
- [ ]  Better GPU handling #87
- more come soon

BrianPugh, 2019-02-01:
what incompatibilities will there be between 6.x and 5.x ?

For bug fixes, what branch should we open PR's to?

kisvegabor, 2019-02-02:
@BrianPugh 
I updated the post above to answer your questions.
The biggest API change will be the introduction of  generic actions `lv_obj_set_action`.

puzrin, 2019-02-02:
Suggestion:

- Directories structure change ( -> `src`, remove `lv_` prefix, add `test` etc)

Let's break everything :)

amirgon, 2019-02-09:
@kisvegabor I would like to bring up again the need for a `context` parameter in all callback functions, as discussed [here](https://github.com/littlevgl/lvgl/issues/557#issuecomment-443455949).  

What I think is missing is a parameter (`void *user_data`) that is received by every callback registration function, saved by lvgl, and passed over every time lvgl calls the callback function. This is common practice when defining a callback mechanism to pass such a variable.

Without `user_data` parameter we have a problem which is not limited to Micropython.
To illustrate the problem think of the following example:  

---

A user has two displays of the same type he wants to use (either simultaneously, or select one on runtime). The user can create a driver **instance** for each display driver, but the driver **callback function** is the same function for both driver instances. What makes the driver instances different from each other is some instance specific data such as which pins they are connected to etc.  
Today when the user registers the driver he only passes a callback function, and when the callback function is called he cannot know for which driver-instance it was called.  

---

In Micropython the problem is harder.  
While in C the user could create two C callback functions one for each driver instance (not so pretty, but possible), on Micropython every callback should be attached to a Micropython object, and a C function pointer alone is not enough for this since we generate the Micropython objects on runtime.  
Today, when creating the Micropython module from lvgl headers, we only have limited callback support  (Only for objects, one action handler per object, and no support for callbacks unrelated to objects).  
I think it would be better to provide a generic way to convert C callbacks to Micropython callbacks on other cases, such as display/input drivers and others.  

Today, you register only the callback function:`lv_disp_drv_register(&disp_drv)` where `disp_drv` is a struct of C callbacks.  
What I suggest is something like `lv_disp_drv_register(void *user_data, &disp_drv)`.  
Later on when the callback should be called, lvgl will call it with the `user_data` parameter: `void disp_flush(void *user_data, int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)`. The driver-instance specific parameters could now be accessed through `user_data`.  

I think every callback should adhere to a coding convention that includes a `user_data` parameter (received when registering the callback and passed over when lvgl calls the callback).  
More Examples where we miss this today: `get_bitmap` on `lv_font_t` and `lv_log_register_print`.

kisvegabor, 2019-02-11:
@amirgon 

I see the problem.

**Display driver**  
It's possible to add a `user_data` element to the driver which can be set along with the flush function. In v6.0 `lv_disp_drv_register` will return a `lv_disp_t` variable similarly to `lv_indev_drv_register`. I suggest passing this `lv_disp_t` pointer to the flush callback because it contains other useful data too (e.g. resolution or other settings). I'm planning to change the `x1, y1, x2, y2` parameters to one `lv_area_t`, so will have only 3 parameters.

**Action callbacks**
A pointer to the object is passed to the callback which means the contexts. As discussed earlier, in v6.0 you can define a custom `free_data` (or `user_data`) `struct` in `lv_conf.h`. In the action function you can get it with `lv_obj_get_free_data(obj)`.  Is it working well with Micropython?
 
**Current state/Plans**
I'm currently working on the multi-display and runtime display config support. I hope it will be merged to `dev-6.0` in a few weeks . Once it's ready I'll add the generic action handling. With these, the two most important and most API breaking changes will be added.

amirgon, 2019-02-11:
@kisvegabor My hope was for a simple convention that would allow **generic** and **automatic** conversion of **every** callback function in lvgl to Micropython.  

Your suggestion gives a specific (non generic) solution only for disp-driver and action-callbacks, but not to a future callback that you might add, and not even for some existing callbacks (such as `get_bitmap` on `lv_font_t` and `lv_log_register_print`).  
With your suggestion we'd need ad-hoc handling of every callback type, and we are not *Forward Compatible* (we cannot support callbacks added to lvgl in the future without fixing the script).  

My suggestion is to define a convention to any callback, no matter what it is: 

- `register_callback(void* user_data, ...)`
- `void (*callback)(void* user_data, ...);`

With such convention, the script will identify the `user_data` parameter and use it to pass along the Micropython object when registering the callback, and invoke the callback as a Micropython function using `user_data` parameter passed to the callback.  
This would work for any callback, no matter what "type" of callback it is, even callbacks you might add in the future.

kisvegabor, 2019-02-11:
@amirgon You are right, it'd be inconsistent. So I'm open add a `context` as the first parameter to every callback but this `context` shouldn't be the `user_data` but its "parent" data structure because you can get the `user_data` from a `lv_obj_t`, `lv_indev_t`, `lv_font_t` but it does not work in the opposite way. So having the "parent" structure you have lot of additional data including the `user_data`.

puzrin, 2019-02-11:
Can metadata join be implemented via proxy layer? It's not very clear, does it need deep integration with lvgl core.

amirgon, 2019-02-11:
> So I'm open add a `context` as the first parameter to every callback but this `context` shouldn't be the `user_data` but its "parent" data structure

@kisvegabor, ok, it makes sense.

- Do we always have a "parent" structure? For example, what is the parent structure of `lv_log_register_print`?
- Is there a generic way to identify the parent structure and access the `user_data` there? We can define such convention. For example - parent structure will alway be the first parameter to both the callback function and the callback registration function, and the parent structure will always have a member called `void *user_data`.

embeddedt, 2019-02-11:
@amirgon @kisvegabor 

> * For example - parent structure will always be the first parameter to both the callback function and the callback registration function, and the parent structure will always have a member called `void *user_data`.

For simplicity, it would also be handy to have `user_data` as the first member of that parent structure.

embeddedt, 2019-02-11:
Sorry, somehow I keep closing issues when I comment. Reopened.

puzrin, 2019-02-11:
Ah, seems i understood initial idea. I that's about equivalent of `this` in OOP, that's a certainly useful pattern. In plain C this looks like `method(context, params...)`. Where context = current object.

amirgon, 2019-02-11:
@puzrin Yes. More generally, a function pointer is defined per **type** while context is defined per **instance**. Without context we would not have access to instance data.

puzrin, 2019-02-11:
I have not enougth C experience, to say exactly "right implementation must be like this", but such feature is really mandatory. With high probability `*ctx` (equivalent of pointer to current object's instance) should be first param, but i'm not 100% sure.

kisvegabor, 2019-02-12:
@embeddedt "Close and comment" button instead of "Comment"? :)

> For simplicity, it would also be handy to have `user_data` as the first member of that parent structure.

Why not, I agree. It might be useful once.

I updated the TODO list in the first post.

kisvegabor, 2019-02-12:
@amirgon 
> Do we always have a "parent" structure? For example, what is the parent structure of `lv_log_register_print`?

I didn't see the purpose of `context` in case of `lv_log_register_print`. It should just register a `printf`-like function.

> Is there a generic way to identify the parent structure and access the `user_data` there? We can define such convention. For example - parent structure will alway be the first parameter to both the [...]

Parent structure can be the first parameter of the callback but the callback are not always registered with API functions. For example, in the display driver, you just assign the flush function to structure. However, if it's registered with API function then the parent structure can be the first element.

puzrin, 2019-02-12:
> Parent structure can be the first parameter of the callback but the callback are not always registered with API functions. For example, in the display driver, you just assign the flush function to [...]

We speak about convention like "we need emulate classes somehow", because it can be suddenly required anywhere. If you don't need context - just ignore first param of function, that's not a problem.

embeddedt, 2019-02-12:
@kisvegabor

> @embeddedt "Close and comment" button instead of "Comment"? :)

Yes, I wasn't trying to click it but somehow I managed to.

@puzrin

> If you don't need context - just ignore first param of function, that's not a problem.

I agree.

puzrin, 2019-02-12:
Let me add some buzz. Design of API, stable for long time ahead is not trivial thing. Sometime it's impossible to describe all factors via direct statements like "we should avoid floats". In this case it's common to do "one step back" and impress requirements via "probabilities":

- "probability of this is high enougth"
- "probability of this is about zero"

In this scope "probability of context need in GUI lib is 100%". If accepted as base of API signatures principle (first param) of methods, it can also help to not reinvent similar things in different parts of lvgl internals. Note, i use more wide term "method" instead of "callback", because issue is more generic.

amirgon, 2019-02-12:
@kisvegabor 

> I didn't see the purpose of `context` in case of `lv_log_register_print`. It should just register a `printf`-like function. Can you explain it?

If you assume there is a single global "logger" which is registered once, I guess there is no need for a "parent" struct, or a "context".  
But if you ever want to allow multiple independent registrations of loggers (for example, one to a file and other to UART), or log to two files with different log levels, then I think a context could be useful.  

If we decide on a canonical way to define callback functions, I think it would be a good practice to apply it to **all** callbacks, `lv_log_register_print` included (although I agree this is not a must)

> Parent structure can be the first parameter of the callback but the callback are not always registered with API functions. For example, in the display driver, you just assign the flush function to [...]

I'm trying to think of a way to automatically convert any callback function to Micropython.  
In case of a registration function that  doesn't receive callback function as a parameter, (only a struct in this case), it is considered a regular function that needs no special handling (actually `lv_disp_drv_register` is already converted to Micropython today).  
The problem arises only when I need to assign a function pointer as a callback function, in this case, assign the struct fields which are pointers to callback functions.  

So, if we agree that:

- The callback function always receives a pointer to the parent struct as the first parameter
- The parent struct first field is always `void *user_data` (or if you want to call it `void *context` I don't care,  as long as I can use it for storing whatever data I need, and lvgl only passes it to the callback function and doesn't use it in any other way)

then, I think I can create a Micorpython wrapper in a generic and automatic way, no matter how the registration function is defined.

kisvegabor, 2019-02-13:
> If you assume there is a single global "logger" which is registered once, I guess there is no need for a "parent" struct, or a "context".
But if you ever want to allow multiple independent registrations of loggers (for example, one to a file and other to UART), or log to two files with different log levels, then I think a context could be useful.

Clear now, thank you for the explanation. `lv_log` don't have any data structure or user data to pass right now but it can be updated if we decide it's required.

> I need a way to **update** `user_data` when a callback is registered and **read** `user_data` when callback is called. This is why I suggested `user_data` as a parameter to both the registration function and the callback function. ....

I think we are talking about the same thing. There are two cases:
1. There are API functions to set/get data of something, e.g. `lv_obj_t`. Use case:
```c
lv_obj_t  * btn = lv_btn_create(lv_scr_act(), NULL);
lv_obj_set_user_data(btn, pointer_to_something);
lv_obj_set_action(btn, my_btn_action);

lv_res_t my_btn_action(lv_obj_t * btn, lv_action_t action) 
{
   void * user_data = lv_obj_get_user_data(btn)
   ...
   return LV_RES_OK;
}
```
2. You are initializing structures and assign callbacks there. E.g. `lv_disp_drv_t` will work this way:
```c
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);       
    disp_drv.disp_flush = my_disp_flush;
    disp_drv.user_data = pointer_to_something;   
    lv_disp_drv_register(&disp_drv);

...
void my_disp_flush(lv_disp_t * disp, lv_area_t * area, lv_color_t * color_p)
{
  void * user_data = lv_disp_get_user_data(disp);
  ...
}
```

amirgon, 2019-02-13:
> I think we are talking about the same thing

@kisvegabor, I'm not sure. There is an open issue here and no agreed solution yet.

In the [previous post](https://github.com/littlevgl/lvgl/issues/785#issuecomment-462974689) I presented a problem, please let me try to explain it again:

If I understand correctly, you suggested passing a "parent struct" to the callback, and register the parent struct instead of the callback function. You also assume the same parent struct could be used by more than one callback.  
For example you suggest having these callback functions: 

```c
// callbacks
typedef void (*disp_flush)(lv_disp_t * disp, lv_area_t * area, lv_color_t * color_p);
typedef void (*disp_fill)(lv_disp_t * disp, lv_area_t * area, lv_color_t color);

// registration
void lv_disp_drv_register(lv_disp_t *disp)'
```

The problems I see with this are:

- **both** callbacks (flush and fill) refer to the **same** parent structure (`disp`). This means we would have **a single** `user_data` for both. However, I need a separate `user_data` for each callback, since each could be a member of a different object.
- An automatic script could have trouble identifying `lv_disp_drv_register` as a registration function. `lv_disp_drv_register` looks like a regular function that receives a struct, not as a callback registration function that needs to be handled in a special way.  
A registration function is special since it needs to automatically initialize `user_data`. As I said above there may be a way to handle this by looking into the struct that the function receives, but there are several problems with that (see my [previous post](https://github.com/littlevgl/lvgl/issues/785#issuecomment-462974689))

---

### My suggestion:

We could register each callback function separately, with a different registration function. Each would receive its own with `user_data`.  
So instead of a single `lv_disp_drv_register(&disp_drv)` we could have:

```c
// callbacks
typedef void (*disp_flush_t)(void *user_data, lv_disp_t * disp, lv_area_t * area, lv_color_t * color_p);
typedef void (*disp_fill_t)(void *user_data, lv_disp_t * disp, lv_area_t * area, lv_color_t color_p);

// registration
void lv_disp_drv_register_flush(disp_flush_t disp_flush, void *user_data, lv_disp_t *disp_drv);
void lv_disp_drv_register_fill(disp_fill_t disp_fill, void *user_data, lv_disp_t *disp_drv);
```

This is generic - When the binding script sees a function that receives a function pointer and `user_data` it will know it's a registration function, it will know which callback is registered and what to pass as `user_data` for this callback registration.  
This ensures that we have 1-to-1 relation between each callback and its `user_data`.
To summarize, the following needs to be true:

- Every callback function should have a separate registration function, and each registration function should register exactly one callback.
- The registration function should receive both the function pointer and `void *user_data` (and can receive other parameters)
- The callback function should receive `user_data` (and can receive other parameters)

kisvegabor, 2019-02-15:
@amirgon I got it. Let's continue here: https://github.com/littlevgl/lvgl/issues/316

kisvegabor, 2019-02-21:
I implemented the first version of multi-display support. 

You can use the `multi-disp` branch of  [Eclipse PC simulator](https://github.com/littlevgl/pc_simulator_sdl_eclipse/tree/multi-disp) project to try it. [Here](https://github.com/littlevgl/lvgl/blob/multi-disp/lv_porting/lv_port_disp_templ.c#L56) you can see how to register a display. `lvgl`, `lv_drivers` and `lv_examples` also have a `multi-disp` branch.

Main API changes:
- VDB is removed from `lv_conf.h`. You need to allocate the buffer yourself and assigned to the displays in runtime. See the simulator project or the example above.
- Unbuffered drawing is removed because it has no real advantage compared to buffered drawing. 
- `lv_scr_act()` is renamed to `lv_disp_get_scr_act()`.
- `LV_HOR_RES` and `LV_VER_RES` is changed to `LV_HOR_RES_MAX` and `LV_VER_RES_MAX` to specify the maximal supported resolution.

Feedback is welcome :)

manison, 2019-02-21:
> I implemented the first version of multi-display support.

I tested this and it seems it works fine. Good work 👍 

A few comments:
Does not build with `LV_COMPILER_VLA_SUPPORTED == 0` (fixed in #881).

I would recommend keeping `lv_scr_act` (defined as `lv_disp_get_scr_act(NULL)`) to ease the upgrade process.

I don't like the idea of every callback in `lv_disp_drv_t` (and presumably also in `lv_indev_drv_t`) having its separate user data. (Not talking about object's generic action handlers here.) I watched the discussion in #316 and wasn't keen to this idea and IMHO it does not make sense here and it's overkill. It would make a little sense if the user data would be passed to the callback directly, but `lv_disp_t` is being passed instead. So if anybody *really* needs distinct user data for each callback, she can define

```c
struct lv_disp_user_data_t {
    void *data_for_flush_cb;
    void *data_for_render_cb;
    ...you get the idea...
};
```

and then access specific user data in the callback through `lv_disp_t::driver`. But I think the usual scenario is that the driver needs the same user data (typically some low-level handle to underlying graphical context, framebuffer address etc) in all callbacks. Now I need to fill 4 user data for 4 callbacks with the same data. It also means 4 times larger memory footprint. I just think it should be the other way around.

kisvegabor, 2019-02-21:
@manison Thank you for testing and the PR!

> I would recommend keeping `lv_scr_act` (defined as `lv_disp_get_scr_act(NULL)`) to ease the upgrade process.

I'll consider it!

> I don't like the idea of every callback in `lv_disp_drv_t` (and presumably also in `lv_indev_drv_t`) having its separate user data...

I suggested the same but @amirgon highlighted some advantages of directly having user data for each callback. The last few posts of #316 summarize the advantages of this approach. If in light of them you see other aspects which should be taken into account please comment there.  (If you are watching the conversation and have some ideas don't hesitate to share them! )

embeddedt, 2019-02-27:
@kisvegabor I tested it just now. It appears that input is constrained to one display at the moment. I assume that is because the input system hasn't been updated yet.

embeddedt, 2019-02-28:
I've noticed that graphical performance has increased substantially. Not sure exactly what changes you made have caused that, but great work! :+1:

kisvegabor, 2019-02-28:
> I tested it just now. It appears that input is constrained to one display at the moment. I assume that is because the input system hasn't been updated yet.

Yes. I didn't find a simple way yet to solve it in SDL.

> I've noticed that graphical performance has increased substantially. Not sure exactly what changes you made to cause that, but great work!

I haven't expected improvement in performance but good to hear it :)

embeddedt, 2019-03-04:
Some news: I updated the Windows project to `dev-6.0`: https://github.com/littlevgl/pc_simulator_win_codeblocks/tree/dev-6.0

It seems to work pretty well except there's an initial drawing issue. Will investigate further soon.

[image]

kisvegabor, 2019-03-05:
@embeddedt 
1. Does it disappear when you move to an other tab? 
2. Is there an issue if you only create only a button instead of the demo?

embeddedt, 2019-03-05:
> Does it disappear when you move to an other tab?

Once I click another tab everything works properly for the remainder of the application's lifetime.

> Is there an issue if you only create only a button instead of the demo?

I hadn't tested that yet. I'll test it soon and tell you the result.

kisvegabor, 2019-03-08:
A huge update is merged to `dev-6.0`:
- General "events" instead of object-specific actions. Set by `lv_obj_set_event_cb`. The possible events are [here](https://github.com/littlevgl/lvgl/blob/dev-6.0/lv_core/lv_obj.h#L76). The most important events are already working. E.g. DRAG is still not sent. The affected the API of lot of objects. Near all of them is updated but `btnm` and `mbox` is still not updated.
- Multi-display support. See the [simulator](https://github.com/littlevgl/pc_simulator_sdl_eclipse/blob/dev-6.0/main.c#L127) for an example.
- `user_data` added for `lv_obj_t`, `lv_disp_drv_t`, `lv_indev_drv_t`, `lv_group_drv_t`. The user can be `typedef`ed in `lv_conf.h`. `LV_USER_DATA_SINGLE` can enable the `user_data` and `LV_USER_DATA_MULTI` adds `user_data` for every `..._cb` 
- `SYMBOL_...` is renamed to `LV_SYMBOL_...`
- `USE_LV_...` is renamed to `LV_USE_...`
- Several smaller API changes

manison, 2019-03-08:
> Near all of them is updated but **btnm** and **mbox** is still not updated.

I'm curious about these, since now the `lv_btnm_action_t` receives the extra argument identifying the button clicked. I can see the new `lv_event_cb_t` which is supposed to replace the _actions_ does not have any extra argument to carry such information.

manison, 2019-03-08:
I'm testing those changes and I'm hitting an issue which may be related to btnm/mbox still not updated or may be not, I'm not sure.

When the signal callback
https://github.com/littlevgl/lvgl/blob/bff65c217dccdd87aa71255120ad8a4b864b592e/lv_core/lv_indev.c#L809
returns `LV_RES_INV` then the `lv_obj_send_event` on the following lines crashes because the `act_obj` is invalid.

In my case it's mbox (respectively btnm placed inside the mbox) signal callback (`lv_btnm_action_t`) that deletes the message box and returns `LV_RES_INV`,

embeddedt, 2019-03-08:
@kisvegabor Great to hear that! I'll take a look at it as soon as possible.

kisvegabor, 2019-03-09:
@manison 
> I'm curious about these, since now the lv_btnm_action_t receives the extra argument identifying the button clicked. I can see the new lv_event_cb_t which is supposed to replace the actions does not have any extra argument to carry such information.

I'm planning to do it with API functions like `lv_mbox_get_last_active()` (or something similar) which you can call in `LV_EVENT_CLICKED`.

> I'm testing those changes and I'm hitting an issue which may be related to btnm/mbox still not updated or may be not, I'm not sure.

Thank you for pointing that out. I've pushed a fix.

manison, 2019-03-11:
@kisvegabor 
> I'm planning to do it with API functions like lv_mbox_get_last_active() (or something similar) which you can call in LV_EVENT_CLICKED.

I have a scenario where I'm generating bntm clicks programmatically in response to keyboard events. (E.g. ENTER simulates press to _Yes_ button, ESC binds to _No_). Right now the code looks (roughly) like this:

```c
lv_action_t action = lv_btnm_get_action(btnm);
if (key == ENTER) action(btnm, "Yes")
else if (key == ESC) action(btnm, "No");
```

How well does this play with `lv_send_event` and `lv_mbox_get_last_active`?

Suggestion: Wouldn't it be better if events also carried an extra (e.g. `void *`) argument so the generic actions are truly generic? I believe it's quite common pattern in other GUI frameworks that the event handler receives an extra data depending on the type of the event (e.g. [vxWidgets](https://docs.wxwidgets.org/3.1/classwx_command_event.html), [Windows API](https://docs.microsoft.com/en-us/windows/desktop/controls/bn-clicked), [.NET WinForms](https://docs.microsoft.com/en-us/dotnet/api/system.windows.forms.control.mouseclick), …).

> Thank you for pointing that out. I've pushed a fix.

Tested, works OK. Thanks.

kisvegabor, 2019-03-12:
I cleaned up the event interface for `lv_btnm`, `lv_mbox`, `lv_kb` and `lv_tabview`. It still not deeply tested.

I added `lv_btnm/mbox_get_active_btn` and `lv_btnm/mbox_get_active_btn_text` which van be used in the `event_cb` to get the active button. This way not only `CLICK` but `PRESSED`, `LONG_RESSED` and `LV_PRESS_REPEAT` events also can be used.

@manison 
Your scenario is very clear this way. I'm still thinking about how to support it in `v6.0`.   
I'd like to try the event interface as clears as possible and I have two concerns with adding an extra `void *` argument:
1. It might be required only by a few  object types
2. It's not intuitive because it can be different for every object type or can have no meaning at all.

manison, 2019-03-12:
@kisvegabor 
> Your scenario is very clear this way. I'm still thinking about how to support it in v6.0.

I think this scenario is legit and we should support it. I'm curious about the solution you come up with :)

> I'd like to try the event interface as clears as possible and I have two concerns with adding an extra void * argument

I understand. What about introducing 2nd variant of `lv_obj_send_event` which would send an extra data and would be useful for those few cases? Still generic enough.

```c
/** Like lv_obj_send_event, but sends an extra data to be associated with the event. */
lv_res_t lv_obj_send_event_ex(lv_obj_t * obj, lv_event_t event, void *extra);

/**
    For btnm LV_EVENT_CLICKED the extra data represents ID of the clicked button.
    This information is accessible through the lv_btnm_get_active_btn in the event handler.
*/
```

kisvegabor, 2019-03-13:
@manison 
Based on your suggestion I got an idea:
- Change `lv_obj_send_event(obj, event)` to `lv_obj_send_event(obj, event, param)`
-  To make the `event_cb` simple keep the curent interface without `param`
- Add an `lv_event_get_param()` function to get the parameter
- LittlevGL will always set `param` to `NULL`
- This way all the API are still clean and flexible. If you want to trigger an event manually you can pass and get an arbitrary paramter.

manison, 2019-03-13:
@kisvegabor 
I understand your idea, however I don't get how would my btnm event handler determine when to call `lv_event_get_param` and when to use `lv_btnm_get_active_btn`. This does not seem generic at all. Am I missing something? The only possibility would be to track the _active btn_ in btnm internally as the event param.

I can see also another potential problem with this: once you start chaining events (firing an event from another event), one event could rewrite data from the other event which could be source of hard to find issues.

I understand your concerns about purity of the interface. On the other hand the majority of UI frameworks (some of them I mentioned above) ended up with the solution of event data and probably for a reason. (And yes, it's much easier in OO languages :) In principle I'm OK with the suggested `lv_event_get_param` if we solve the dichotomy between `get_param` and `get_active_btn`.

kisvegabor, 2019-03-14:
> I understand your idea, however I don't get how would my btnm event handler determine when to call lv_event_get_param and when to use lv_btnm_get_active_btn. This does not seem generic at all. Am I [...]

If `lv_event_get_param` returns not NULL you can know that it was called by the user and `param` should be taken into account.

> I can see also another potential problem with this: once you start chaining events (firing an event from another event), one event could rewrite data from the other event which could be source of hard to find issues.

I keep track with the current "event calling" object too to see if it was deleted in the event. I solved the chaining like [this](https://github.com/littlevgl/lvgl/blob/dev-6.0/lv_core/lv_obj.c#L1182). The same can be used with `param` too.

> On the other hand the majority of UI frameworks (some of them I mentioned above) ended up with the solution of event data and probably for a reason. 

- Windows API: it seems to me only the object and the event is passed
- .NET: It seems quite the same prototype as in LittlevGL `Control1_MouseClick(Object sender, MouseEventArgs e) `
- wxWidget: it wasn't clear to me :sweat_smile:

manison, 2019-03-14:
> If lv_event_get_param returns not NULL you can know that it was called by the user and param should be taken into account.

Frankly I really don't like this solution. It's not generic at all. Consider an application that can be controlled by the pointer input device only. All btnm event handlers are using `lv_btnm_get_active_btn` to determine which button was clicked. Now I decide that keyboard control will be added and the event will need to be send programmatically. Then I need to go through all the event handlers and modify them to be aware of the other source of the data. I think that either:
* btnm should send events with `param` set (then I can use `lv_event_get_param` in all cases)

or
* `lv_btnm_get_active_btn` should be alias to `lv_event_get_param`

#### About other frameworks

> Windows API: it seems to me only the object and the event is passed

Additional event (message) data are passed through the _wParam_ and _lParam_ arguments ([SendMessage](https://docs.microsoft.com/en-us/windows/desktop/api/winuser/nf-winuser-sendmessage)). Meaning of the arguments depends on the type of the message.

> .NET: It seems quite the same prototype as in LittlevGL Control1_MouseClick(Object sender, MouseEventArgs e)

Not really:
* name of the event (e.g. _MouseClick_) → `lv_event_t`
* _sender_ → `lv_obj_t`
* _EventArgs_ → N/A

> wxWidget: it wasn't clear to me

Similar to .NET: `void handlerFuncName(wxCommandEvent& event)`
* _wxEventType_ → `lv_event_t`
* _this_ → `lv_obj_t`
* additional data in _wxEvent_ → N/A

kisvegabor, 2019-03-14:
I'm not against to make something more flexible even in the cost of an extra parameter in the event function if there are some objects which will use it. I can imagine only the `lv_btnm` related objects like 
- `lv_kb`
- `lv_mbox`
- `lv_tabview`

In these cases, the pressed button's text can be passed.

Can you imagine other objects where we can use the extra arguemnt?

manison, 2019-03-14:
To be honest I don't know all the objects in LittlevGL so I can't tell. But this would allow us to extend any object/event in the future and even objects we can't imagine today :)

kisvegabor, 2019-03-15:
@embeddedt Thank you for pinning, good idea!

@manison I've just realized that it won't work this way. Some events are really generic and called by `lv_indev.c`. If you click on an object `LV_EVENT_CLICKED` will be sent. So the object itself can't set `param` because it event doesn't know the event is sent. It's because CLICK is a generic think which means nothing else then you clicked the object. In contrary `LV_EVENT_VALUE_CHANGED` is sent by the object itself because only it knows when its value has changed.
So I can imagine two solutions:
1. Add a new EVENT (similarly to VALUE_CHANGED). Need to find a general name/concept for this event.
2. Add an `lv_btnm_set_btn_active()` to manually activate a button of the button matrix to simulate a click

manison, 2019-03-15:
> Some events are really generic and called by lv_indev.c. If you click on an object LV_EVENT_CLICKED will be sent. … In contrary LV_EVENT_VALUE_CHANGED is sent by the object itself because only it knows when its value has changed.

This is similar how things works on Windows. Some messages are sent by system ([WM_MOUSEMOVE](https://docs.microsoft.com/en-us/windows/desktop/inputdev/wm-mousemove), [WM_KEYDOWN](https://docs.microsoft.com/en-us/windows/desktop/inputdev/wm-keydown), …). The control based on these messages generates its own notifications messages. E.g. button control generates [BN_CLICKED](https://docs.microsoft.com/en-us/windows/desktop/controls/bn-clicked) notification when it receives _mouse click_ or _Enter key pressed_ messages.

> So the object itself can't set param because it event doesn't know the event is sent.

Maybe the event should be routed to the object first, like in Windows. E.g. in the edit control (~ text area object) the WM_KEYDOWN generates [WM_CHAR](https://docs.microsoft.com/en-us/windows/desktop/inputdev/wm-char) which in turn can lead to [EN_CHANGE](https://docs.microsoft.com/en-us/windows/desktop/controls/en-change) notification.

> Add a new EVENT (similarly to VALUE_CHANGED).

I like this approach. But we need the event data anyway.

> Add an lv_btnm_set_btn_active() to manually activate a button of the button matrix to simulate a click

Don't like this. It's a mere workaround for single specific scenario.

> Need to find a general name/concept for this event.

I don't think that there have to be a single set of general events shared by all objects. It's common that specific objects have their own specific events (take the [button](https://docs.microsoft.com/en-us/windows/desktop/controls/bumper-button-control-reference-notifications) and [edit box](https://docs.microsoft.com/en-us/windows/desktop/controls/bumper-edit-control-reference-notifications) examples mentioned above). I believe there can be a basic set of general events (like `LV_EVENT_CLICKED` etc) and specific events for particular objects.

_lv_obj.h_
```c
enum {
  LV_EVENT_PRESSED,
  ...
  LV_EVENT_OBJ_FIRST
};
typedef uint8_t lv_event_t;
```

_lv_btnm.h_
```c
enum {
  LV_EVENT_BNTM_PRESSED = LV_EVENT_OBJ_FIRST,
  ...
}
```

embeddedt, 2019-03-17:
FYI: https://github.com/littlevgl/pc_simulator_win_codeblocks has been updated to reflect the latest changes in `dev-6.0`. Everything is working properly again.

kisvegabor, 2019-03-17:
@manison 
Thank you for the detailed explanation!

> I don't think that there have to be a single set of general events shared by all objects. It's common that specific objects have their own specific events (take the button and edit box examples [...]

By "general name" I meant a general name for an event to handle selecting a button or option etc. For example `LV_EVENT_SELECT` which can be called when a message box button is clicked, or a tabview tab is selected.

I still feel that the use of a `param` is too narrow.   
In `LV_EVENT_SELECT` you can decide:
- to call `lv_event_get_param()` because in this case it will be provided by the library or by you when the event is triggered manually.
- or get the corresponding function the see the current state (e.g. `lv_btnm_get_btn_active`) 

Introducing object-level events (like `WM_CHAR`) is a conceptual thing. To be honest I don't want to add it for some special use case of a few objects. I think the above prosed `LV_EVENT_SELECT` would be clear an usable, and still not disturbing a usual use-cases. What do you think?

By the way, `LV_EVENT_REFRESH` is directly for the user to trigger the refresh of an object (e.g. a label) manually. `param` will come in handy here too.

kisvegabor, 2019-03-17:
@embeddedt Awesome news! I will also try `dev-6.0` on STM32F4 Discovery soon.

manison, 2019-03-18:
> I think the above prosed LV_EVENT_SELECT would be clear an usable, and still not disturbing a usual use-cases. What do you think?

I'm OK with that.

To sum things up:
* the mbox will fire `LV_EVENT_SELECT` when the button is clicked
* lvgl itself will provide extra data for this event (accessible through `lv_event_get_param` in the event handler)
* the extra data will contain ID of the clicked button
* the event can be sent programmatically with `lv_obj_send_event(obj, LV_EVENT_SELECT, data)`

Is this summary correct?

kisvegabor, 2019-03-19:
Exactly! 

I've just updated `dev-6.0` with `LV_EVENT_SELECTED` and `lv_event:get_data`. Can you try it?

manison, 2019-03-19:
It seems to work fine, although I would rather pass integer button ID instead of the button text for several reasons:
* it's easier (and more efficient) to compare integers than strings (`==` vs `strcmp`)
* if anyone really needs the text, it's a matter of single additional call (`lv_btnm_get_btn_text`), there is not the opposite API
* keeping track of the current button text can get messy once you start changing texts at runtime (because of [localization](https://github.com/littlevgl/lv_i18n))
* internally the active button is already integer anyway

kisvegabor, 2019-03-20:
@manison I agree with all the points. Thanks for pointing them out! 
I updated `dev-6.0` accordingly.

fallstool, 2019-03-20:
Hi 
can the font system support feature of Kerning?
this problem has discussed in  isssue [#168](https://github.com/littlevgl/lvgl/issues/168)

manison, 2019-03-20:
@kisvegabor 
> I agree with all the points. Thanks for pointing them out!
I updated dev-6.0 accordingly.

Great, thanks! I noticed there's inconsistency in pointer types. In btnm the event data is `uint32_t *`, whereas in mbox and tabview it's `uint16_t *`. This should be unified. Or the event data does not need to be dereferenced and the pointer can be used as integral type directly:

```c
lv_event_send(obj, LV_EVENT_SELECTED, (const void *)(uintptr_t)btn_id);
uintptr_t btn_id = (uintptr_t)lv_event_get_data();
```

kisvegabor, 2019-03-20:
@manison I think handling pointers as integer are error prone and hacky. I'm planning to introduce `lv_int_t` for integer types where just a "big number" (>= 16 bit) is required. It will be used in lot of places so it should make this case clearer.

embeddedt, 2019-03-20:
> I'm planning to introduce `lv_int_t` for integer types where just a "big number" (>= 16 bit) is required.

Is it necessary to introduce a new type? What's wrong with just using `(u)int16_t` or `(u)int32_t`?

manison, 2019-03-20:
> I think handling pointers as integer are error prone and hacky.

IHMO it's error prone the same way the casting `void *` to anything else is. But I'm not strongly opposed to using integral pointers. Just use the same type. And I agree with @embeddedt – please do not introduce another integral data type. There is plenty in _<stdint.h>_, just pick one and use it consistently in all places.

[272 more replies]

closed 2019-07-15 by kisvegabor