lvgl/lvgl #786: [OFF] Welcome here!

kisvegabor, 2019-02-01:
Hi all,

We are talking so much here and we know only little about each other. 

I opened this issue to have a place where you can write a few lines about your background.

I'm sure it will help to make the forum more personal and also to see what skills you have. It's not only for LittlevGL but to make new relationships as well.

So a few lines about me:
I'm from Hungary and was born in 1990. I was 17 years old when I started to work with electronics. I have some experience with hardware design but most of the time I'm writing embedded software in C.  
I created the first, experimental version of the LittlevGL for my hobby projects in 2009. Since then it was rewritten and published on GitHub in 2016.   
Besides coding, I needed to learn a lot about SEO (Search Engine Optimization), marketing, web programming and design.  
And finally a little bit of personal thing: since December 2018 we are happy parents of our first baby, called Zakariás (Zachary in English) :)

BrianPugh, 2019-02-01:
Hey guys, my name is Brian Pugh. I'm a deep-learning scientist for a startup called Geomagical in Silicon Valley; we specialize in developing novel algorithms for indoor scene perception. When I'm sick of Python for the day, I go home and work on my open-source cryptocurrency hardware wallet utilizing the LVGL library called Jolt. Jolt's GUI is displayed via a 0.96" 128x64 OLED, which is while you'll often see me make pull requests nitpicking about pixel perfect objects ;) .

amirgon, 2019-02-01:
Hi everyone!
My name is Amir, I'm from Israel. I'm married and have 3 sweet little children (ages 2.5, 5 and 7 years old). I also like swimming (on a nearby pool) and skiing (usually in France). On my day job I work on a large corporation which develops ASIC, where I had the chance of doing lots of interesting stuff mainly related to software tools development, including porting gcc to a new core, developing hardware simulation framework, and even some .NET and web development.  
As a hobby I'm making something cool :wink: from ESP32+display+some other stuff, when it's ready I'll let you know, or if I'm lucky you'll read about it in the newspapers...  
I'm already using Micropython on my project and wanted to add LittlevGL. I really need them to work well together so I'm making an effort to [bind LittlevGL and Micropython](https://github.com/littlevgl/lvgl/issues/557) and I feel there is good progress! I might be getting too involved with this instead of making progress with my own project, but I don't care, I'm really enjoying it!

pete-pjb, 2019-02-02:
Hi All,

I am Pete Bone from the UK, I live and work out in the countryside with my lovely partner Debbie and her two daughters.  I have two grown up daughters 22 and 26 years old and a granddaughter who is six months old.  I have been working with electronics since I was about 5 years old, building various circuits etc. I aquired my first computer (Sinclair ZX81 (Z80)with 1K of RAM!) in 1981 at age 12 and progressed through BBC micros(6502), Atari STs(Motorola 68000) and eventually onto PCs. I am a hardware and software development engineer working in C, Assembler and Verilog mainly.  I also do CAD work Schematic Capture and PCB design.  I have designed for work, embedded systems with various 8051 micros, STM32s, Texas Instruments DSPs, Motorola (now Freescale/NXP) Network processing SOCs, 386/486/Pentium/PCI based systems and Xilinx FPGA/SOC and CPLD devices.  I have also worked quite a lot with Analog Devices RF and synthesis chips.
I am currently working on a personal project for my own home to provide automation based on the Zedboard.  I initially looked at Linux but it's far to big and bloated these days hence I have gone with FreeRTOS & LittleVGL instead.  I am intending to share my work in the form of a home automation system. I have Verilog based controllers on board; a VGA controller, a control for WS2812B LED strips, multi-channel PWM dimmers, security sensors, temperature and humidity sensors(DHT22), PIR sensors, solenoid valve control and sound detection. I have various electronic circuits to interface everything together. I am also looking at trying to incorporate a VNC server for remote device control in the longer term to enable tablet/mobile phone and PC access to the system. The Verilog controllers are instantiated in the programmable logic fabric of the Zynq 7020 device on the Zedboard and interfaced to the on-board dual ARM Cortex-A9 cores via memory mapped AXI interfaces.  I am running two instantiations of FreeRTOS, one on each core, using one core to run the GUI, management and IP stack and the other core to do the real-time processing of system events etc. with a block of shared memory for the cores to exchange information. Once I have the system completed I will share it with the open source community.

turoksama, 2019-02-02:
Hi guys, my name is Samuel Wong from china. I started to work with electronics as a tinker, when I was 14 years old, soldering here and there, loved to collect any kinds of electric boards since then. I was really fascinated with embed programing when I went to college. 
I created an experimental GUI for my hobby projects around 2010. At the same time, uCGUI is the most popular one. I gave up trying when I noticed my limited C skills and started to work with various 8051s, RFs, STM32s, LCDs.
All these years, I am experienced with hardware design(all the way from design to production) while customizing embedded software in C.
I am currently working on a personal project for industry automation based on CORTEX-M4/M7, which is while you'll often see me make pull requests when it is needed.

seyyah, 2019-02-04:
Hi, I'm Nurettin from Turkiye, an all-around website developer with over 10 years of experience in Linux systems. I'm working as a freelancer. We're developing medical device based on ARM. I love C coding. I am professional experienced programmer working on Ruby on Rails, PHP, HTML5, Bootstrap, Wordpress, CSS3, RoR. I've worked on several Unix infrastructures and internal application systems including web-servers (Apache, Nginx), architecture and core technologies (SSH etc). I have good server deployment knowledge into various server hosting Platform i.e. Digital Ocean, Heroku. I use docker, vagrant etc. I worked on multiple web technologies & Restful APIs for cross platform communication. I also code review & refactoring and am using tools for high quaity code like Rubocop, Rails best practices. I'd like Image Processing & Computer Vision with Matlab. I've some experience about mobile programming, deep/neural networking. Thanks

C47D, 2019-02-04:
Hi, my name is Carlos and i'm from México, I'm new in the embedded development world (1 year formally working, 2 years by my own), i really like to program with C, but I'm also interested in new langs like Rust. I feel more comfortable writing software than designing hardware but I'm always trying to get better in both.

My goal here is to learn how to use lvgl and write about it in Spanish.

tjstyle, 2019-02-08:
Hi, I'm Teguh from Indonesia, but just call me TJ.
Started to hacking the Handphone operating system about 15 Years ago, then go to Smartphone era when Android project is released to the public, active in the Android open source community and also work a professional job for smartphone manufacturer as a system engineer.
Started to play with an embedded world about 10 Years ago then become a professional job till now.
Have some work that needs to use GUI display in the projects, and I use LittleVGL for that.

kisvegabor, 2019-03-02:
Thank you for the introductions! It was amazing to read how versatile the community here! 

To reduce the number of opened issue I close it now.

If anyone finds it later feel free to write a few lines about yourself.

closed 2019-03-02 by kisvegabor

FFFFFF12ABB17, 2020-11-24:
Hi all!
I'm a little late to the party but I'm an software developer/electrical engineer from Portland, Oregon with a focus on DSP and audio applications.  I'm a 90's kid with a love for all things future past.  Currently, I run my own small electronics company here in Eugene, Oregon, where I live with my beautiful wife, Lexi and our 3 kids Malachi, Arya, and Zoey.

kisvegabor, 2020-11-25:
@FFFFFF12ABB17 
Thank you for the introduction! :slightly_smiling_face: