lvgl/lvgl #791: i18n next

puzrin, 2019-02-02:
In continue of https://github.com/littlevgl/lvgl/issues/562#issuecomment-459437016

Current i18n support is basic. It's much better than nothig, but i'd prefer to have some advanced features out of box:

- Plurals support.
- Easy way to integrate with web translators ([example](https://www.transifex.com/nodebb/nodebb/)).
- Easy phrase add + fallbacks when translation missed.

## Things to disuss

### General

- Plurals are essential, but are ordinals needed?
  - comment: i use plurals very wide on web, but never used ordinals.
- `gettext()` or [special syntax like this](https://github.com/nodeca/babelfish)?
  - `gettext()` is needed anyway as de-facto standard. But it's use/translation is very boring. In many cases having all in one phrase will be much more convenient.
- What kind of additional tools needed?
  - example: it's convenient to write code "as id goes", and then run tool to update all inconsistencies of translations.
- Fallbacks to default language? Current implementation expects to fill all translations when new phrase added. That's not realistic. If one fill all translations with english - it will be impossible to control real translation progress.

### Technical

- Revisit desired C code structure. Current one is good for small manualy editable files only.
  - Probably just need 2 formats, with examples
  - Probably better look can be acheived with macros
- If embedded syntax for plurals is interesting, need to polish it + API
  - Do we need some additional sugar? May be some formatter for numbers, but i'm not sure.
- What about memory allocations? We can not know in advance exact string length, when numbers injected. But it would not be cool to reallocate memory on every call. How is this solved in embedded (i don't know)?

---

What can i do:

- write all necessary standalone tools in JS, if this is acceptable.
- clarify spec for plurals syntax.

puzrin, 2019-02-02:
I argee, with current resources it's not realistic to care about text shaping.

> It seems to me from the list above, the handling of the plurals is most difficult.

What do you mean? Plural rules decoders can be [generated automatically](https://github.com/nodeca/plurals-cldr/blob/master/support/generate.js) from unicode cldr data.

If we stay that sources are stored in yaml/json, then it's not a problem to transform syntax into gettext-style phrases (1-4 phrases for each plural) by javascript. No need to implement parser in C.

> To format the texts we should use `printf` syntax like `gettext()` does.

I don't know. `printf` is fat and complicated. I'd suggest to keep things separate for begining. That will allow to stabilize architecture fast, but still reserve extendability.

```cpp
result = printf(_('I have %02d ((nail|nails))', count), count);
```

Looks a bit ugly, but not fatal.

> I suggest letting the current text ID based handling because it's much faster than dealing with string compares (like in the case of `gettext`).

Both can be supported. It's not a big difference how to generate code. And since lang files are generated, those may include binary tree index for fast lookup by string. Need to check existing `.PO` format, may be it's already ok for our needs.

I would like to use enums, as discussed before, but those are not human friendly. When you add new phrase, you have to:

1. Go to file with enums list and add new one
2. Add default translation
3. Return baack to continue programming

With `gettext` you just write phase. Big difference :). 

Can enum's machinery be hidden via marcos? (it should look like `gettext`, but have  enums inside after preprocessing). Any other ideas?

> There should be only one temporal buffer (buf for sprintf) because in LittelvGL the texts of the labels are allocated in the dynamic memory.

Do you mean buffer will be in stack? That's nice. But how do we know minimal possible size of this buffer?

> I've just realized that with the above described `lv_lang_refr_var_int` we can have only 1 variable in the text. For example:  having a text like `"#name has %02d#count((nail\|nails))"` if we call `lv_lang_refr_var_int("count", nail_cnt, )` the value of #name will be unkown.

Why can't we stay without such function? Only with text load to label - `lv_label_set_text(printf(_('I have %02d ((nail|nails))', count), count))`. May be i missed some point.

kisvegabor, 2019-02-03:
> What do you mean? Plural rules decoders can be generated automatically from unicode cldr data.
If we stay that sources are stored in yaml/json, then it's not a problem to transform syntax into gettext-style phrases (1-4 phrases for each plural) by javascript. No need to implement parser in C.

I plural handling requires the mentioned "variable linking". I meant that part.

> Both can be supported. It's not a big difference how to generate code. And since lang files are generated, those may include binary tree index for fast lookup by string. Need to check existing .PO format, may be it's already ok for our needs.

I never worked with so advanced localization but - in addition to the fastest computing - IDs seems more robust. Let's say you fix a typo in the default (English) string:   
`Teh best thing ...` --> ` The best thing ...`. After that, the translations of this text won't be found. Am I correct?

> Do you mean buffer will be in stack? That's nice. But how do we know minimal possible size of this buffer?

Speaking about systems with limited resource a define to set the buffer's size can work.

> Why can't we stay without such function? Only with text load to label - lv_label_set_text(printf(_('I have %02d ((nail|nails))', count), count)). May be i missed some point.

I assume you meant `sprintf` instead of `printf`.  
Your suggestion can work if the user keep track of the labels where `count` is used. In your suggestion updating `count` looks something similar:
```c
count = 10;
lv_label_set_text(label1, sprintf(_("I have %02d ((nail|nails))", count), count));
lv_label_set_text(label2, sprintf(_("All my %02d ((nail|nails)) are beautiful", count), count));
lv_label_set_text(label3, sprintf(_('She painted  her %02d ((nail|nails))', count), count));
```

It doesn't seem user-friendly. In my suggestion:
```c
count = 10;
lv_lang_refr_var_int("count", count);
```
But as I mentioned it would support only 1 variable/text. To enable the use of more varaibles a function like `lv_lang_get_var("name", buf)` would be required to collect all the required variables. IN this case the variables need to be registered with e.g. `lv_lang_add_var_str("name", user_name)` or `lv_lang_add_var_int("count", &count)`.

puzrin, 2019-02-03:
> in addition to the fastest computing - IDs seems more robust. Let's say you fix a typo in the default (English) string:
`Teh best thing ...` --> `The best thing ...`. After that, the translations of this text won't be found. Am I correct?

Yes, such disadvantage exists. But that's not fatal, and you will be reported by CI tests about floating translations - not too difficult to fix. I'm trying to find if any automated "merge" available (via external tools).

Enums are still not human friendly, and this should be solved prior to consider this seriously as main way. Their current use breaks 2 principles at once:

- diverge from "mainstream" translation way
- ass pain level of usability for code write

> Your suggestion can work if the user keep track of the labels where `count` is used.

Now understand. You wish to have reactive bindings (easy way to synchronise values of multiple objects). That's a common approach in interface-related programming. But you suggest very specific implementation + for narrow use. That's almost guaranteed to become architecture problem in future.

I'd suggest to google for basic libs (how to create objects and to link their properties to be updated in sync). Then decide how one of such lib could be applied to lvgl everywere in universal way. And should this be done in lvgl or in userspace 

In general - for good design "building blocks" should isolated. That means i18n should work independent on do you have reactive bindings or not

puzrin, 2019-02-03:
See short example in https://github.com/schlangster/cpp.react, about reactive programming. I don't know what is available for pure C.

kisvegabor, 2019-02-04:
I educated myself a little bit about reactive programming. To implemented or use a complete reactive interface in lvgl is clearly not a subject of an embedded GUI library written in C. Due to the limitations of the language itself a full-featured reactive interface doesn't seem reasonable. What I can imagine are:
1. A very minimalistic reactive support what I mentioned before.
2. Move the whole thing outside of lvgl i.e. the user should solve it as he can/want (not sure if it's possible and nor a good idea)
3. Rethink plural support from scratch. We should keep in mind that we won't be able to reach the same flexibility than in JavaScript. (or it will be too "expensive") 

I vote for the 3rd option. I'll think about what other options we have.

puzrin, 2019-02-04:
I'm against (1), because this will become a problem very fast, according to my experience. In JS you have unlimited possibilites to shout yourself in leg and in head :). That's very strong motivation to improve architecture design skills and detect problems on early stages.

I'm for (3), intent of this issue is to understand what can be done, and how to do that well. May be, after investigation we will go to (2). But as far as i know, there are no ready `gettext` solutions for simple embedded. I see no problem, if this will be a separate repo like `lvgl_gettext`, `lvgl_i18n` or something similar.

puzrin, 2019-02-05:
Technical thoughts to (3). It's possible to solve phrases lookup difficulties via macros:

`_('phrase to translate')` => `gettext(crc32('phrase to translate'), 'phrase to translate')`

First param is index helper - any predictable hash of 16-32 bits long. It may be not unique, but will narrow scans significantly (instead of many strings compare you can do half division integer search, if translations are properly ordered). In most 99.99% of cases, result will be unique. For the rest - additional checks will be needed, but that's fast.

Price to pay - 4 more bytes for each translatable phrase. Looks acceptable.

Can enums be generated in similar "lazy" way (without manual centralized definition, filled by macros in compile time)?

puzrin, 2019-02-05:
I've prepared mind map. That's very convenient to sort out & share toughts on early stages, when everything is not clear. You can copy source and change as you wish.

https://coggle.it/diagram/XFmcuZ2sQEyc4M8J/t/i18n-for-embedded

[image]

kisvegabor, 2019-02-05:
Based on our goals, the suggested solutions and the possibilities I think I put together a viable concept. Here is how I was thinking:  
(Sorry for being so long. It's a compex feature and I can't explain it shorter):

### How variable binding (reactive) is handled with gettext?
I found that you simply need to refresh the labels when a variable changes. This approach is already widely used even on large systems so we should be fine with that too.

### How to select the correct plural form with gettext?
Handling plurals in English is easy because the nouns have only two forms. But in some languages (it seems in Russian too) it's more complicated because there can be 3, 4, 5 or even 6 forms.  
So I've checked how `gettext` supports plurals and found [this](https://www.gnu.org/software/gettext/manual/html_node/Plural-forms.html). In short:
- The PO (language) file has all possible plural versions of the texts. 
- Some rules can be defined in the header of the PO files to decide which plural form to use.
- `ngettext` gets `"singular"`, `"plural"` and `number` as arguments and based on the rule in the PO file decides which plural version to return. It a little bit obscure why it needs the singular and plural form too when it might return a third version.

It gives the opportunity to change `1` to `one` in the translation. E.g. 
- `("I have %d dog", 1)` --> `"I have one dog"`
- `("I have %d dog", 3)` --> `"I have 3 dogs"`

### How to use all this in our case?
We can't use a PO file as it's a text file which needs to be read and parsed. We should store the translated texts in the flash as strings.

We can have a function (call it `t` for translator) like this `t("msg_id", cnt)` which returns the translated text with the correct plural form based on `cnt`. E.g. `t("I have %d dog", 3)` --> `"I have %d dogs"`. The return value is a pointer to a string in the flash. Let's have an other function (called `r` for replace) which works like `sprintf` but it can be defined by the user if he needs something more lightweighted. So to construct the final string: `r(t("I have %d dog", 3), 3)` (similar to what you mentioned earlier). A macro can be used to make more readable: `rt("I have %d dog", 3)` It has the disadvantage of we still can have only one number in one text but it can work: 
```c
r("%s %s", rt("I have %d dog", 3)), rt("and %d cat", 2);`
```

### How to store the translated strings of a language in an effective way?
It's not trivial how to store the plural forms because one `"msg_id"` can be associated with more plural forms and the number of plural forms is different in languages. So having 2 strings for "%d dog"` (`"One dog"` and  `"%d dogs"`) would make searching difficult.
To solve this I suggest using `\0` delimiters in the string separate the plural versions: `"One dog\0%d dogs\0\0"`. This way a substring can be returned and we can see if there are more sub-strings from the two `\0`.
 
So we have a 1:1 relation which easy to associate.
If we store the `"msg_id"`s in ascending order in a string array then it would be easy to find the real ID (index in the translations arrays): 
1. Search the first occurrence of the first letter
2. Form here search the first occurrence of the second letter
3. and so on

### So shat are the steps?
**Initialization**
1. Store the translation of each languages in a string array.
2. Collect the pointer of the translation arrays in `cons char ** languages[]` array.  

**Usage**
1. Somewhere the user's code `t` gets a text (`"msg_id"`) and a number (`cnt`).
2. Get the selected language as ` languages[lang_sel]`
3. Get the numeric ID of the `msg_id` by searcing for it in the ordered list of message IDs. (Done by lvgl)
4. Get the IDth long string (with all pulars) on the given language. (Done by lvgl)
5. Convert the `cnt` to plural ID (with an algorithm declared in the given language's file)
6. Return the `plural ID`th sub-string from the long string. 

### Compatibility with simple usage (no pulars)
If no plurals are used the `cnt` to `plural ID` function in the languages file should always return 0 to get the first (and only) sub-string.

### Now strings are used as IDs. Will it make more difficult to reuse texts?
A macro like `#define MSG_ID_MAIN_TITLE "Main title"` can work instead of typing a text.  
To save RAM a list of IDs can exist whose value can be [stringified](https://gcc.gnu.org/onlinedocs/cpp/Stringizing.html) by the preprocessor. It will result in `"msg_id"`s like `"12"`.

### Automation
In the default case, where texts are used as `"msg_id"`s a scrip can collect all the texts from the source code. It just needs to search for `t(".*", .+)` (simplified regexp)

puzrin, 2019-02-06:
MEMO. Technical details. It's e bit early to discuss, before top architecture become stable. Consider this just a notes to remember. May be those will be useless later...

1. gettext fast lookup. When argument is something `const* char`, we can use it's address as key, to translate into translation struct address. This will need to define some memory for hash map. Don't know, will it help or not.
2. Instead of setting text to label directly, we can have alternate method "on_get_text()", driven by some event. For most of cases, such function can be defined inline as `auto`, this should look simple & convenient. This can simplify definition of reactive bindings on user's side. You can just call `.refresh()` to force update. Just an idea, may be useless.

puzrin, 2019-02-06:
Domains (namespaces) support. This may be required for very big phrase sets, to split omonyms (very rare case, but if happens, should have resolution).

BUT, if we use texts as IDs, it will be more simple to have convention like this:

```
_('[domain1_name]phrase');
_('[domain2_name]phrase');

phrase - the same text, which should have different translations for different contexts.
```

Advantage:

- Requires nothing to do, simple to use
- Transparent to future alternative implementations

Disadvantage:

- Can not be used (looks not good) without translation (when default `_()` param value is displayed)

kisvegabor, 2019-02-06:
In general, I agree with your comments. 

> Do you agree to use gettext's approach with text-based id's instead of enums? (because it's more user-friendly)

I agree. In addition, instead of a sorted list of translated strings, we can use something like:
```c
{"msgid1", "translated text 1"},
{"msgid23", "translated text 23"},
{"msgid7", "translated text 7"},
```

It should be still fast to search among short "msgid"s but it's more robust and easier to edit if no external tools are used.

> I did not understand your position about reactive bindings.

I also suggest not to deal with binding.  Your idea about  `.refresh()` fits very well too a planned new feature. In v6.0 `action`s will be basic object features. So every object can have an action callback. If something happens the callback will be called with an `even_id`. In our case (`LV_ACTION_REFRESH`) This action can be sent manually by the user too. 

> We need to discuss `sprintf`-like formatting, with emdedded systems specifics:
  
It can be independent of the library. Once the user gets back the translated "plural-correct" string he can format it as he wants. He can use sprintf or implement more simple function depending on his requirements.

> Domains (namespaces) support. This may be required for very big phrase sets, to split omonyms (very rare case, but if happens, should have resolution).

[This document](https://refspecs.linuxfoundation.org/LSB_1.3.0/gLSB/gLSB/baselib-dcngettext.html) defines the following terminology for `dcngettext`:
- domainname: temporarily change the selected language
- category: specify the context

Adding some special part to the `"msg_id"` to show the category is good idea. It will consume more space but only in very rare cases.
I'd use `@` like `"continue@button"` means the text is located on a button.

Changing the domain (language) temproarily can happen similarly to `dgettext`: `d_(LANG_ID, "some text")`

puzrin, 2019-02-06:
You are right. I mixed up domain with category. Categories are needed very rare, but can be emulated by prefix. Domains - i could not imagine when this can be needed at all, but can be solved by direct language switch call.

So, to keep api simple and easy to develop:

- Domains, categories - now postpone (probably, forever)
- `sprintf` - now postpone, but validate gettext() signatures for possible extendability. Probably, external wrapper is optimal, because plurals are rare in embedded.

> It should be still fast to search among short "msgid"s but it's more robust and easier to edit if no external tools are used.

That's important question to discuss.

I understand desire to not use "all that cryptic tools in strange JS" and do all manually. BUT keeping file easy to write/modify for humans can add very serious unwanted restrictions. I see 2 ways:

- Say "just don't use this package and switch to hand-made i18n".
- Try to support 2 formats - one for script-generated, second for manually-edited.

I feel, attempts to invent one universal thing for all may fall to infinite loop. I would prefer to spend time for script-based solution only, and to it fast. But i don't like to push anyone with my personal preferences. So, need to decide how to be useful for all with minimal efforts.

My suggestion is to "forget about manual edits" on first iterations, and see resulting format. Then decide if we can do it human-friendly or separate format required.

Also, i suggest to review spec draft first, and check how much do user lose without scripts (checker, merger). Remember, user still may need script to build fonts. But, again, i understand how difficult is to change habbits. Will be glad to discuss any suggestions.

puzrin, 2019-02-06:
https://github.com/littlevgl/lvgl/wiki/i18n-draft

Created very basic spec draft to start API/scripts stabilization.

Seems that's too much for single issue. And wiki is not very convenient for such work. IMHO it worth to discuss repo name + global names, and create a separate repo.

Note, i try to design things as much flexible as possible. That means, those should work later for Rust with minimal changes.

- I don't know, does it worth to "lock" names with `lv_` prefix. Things seems to be universal (except minor adapters for code & font generation)
- I certaintly wish this repo to be under littlevgl org umbrella
  - That's initially done for lvgl :).
  - May be need help with C coding (don't know yet)
  - I will do everything to keep good code quality & maintainability, but will not maintain it forever myself.

kisvegabor, 2019-02-07:
>> It should be still fast to search among short "msgid"s but it's more robust and easier to edit if no external tools are used.
>
> That's important question to discuss.

As the tranlated text's need to fetched only when the screen is created and when a label is refreshed I don't think it will have a notable effect on performance. Let's say we have 500 "msgid"s each with length of 10 and 10 label on a screen --> 50,000 chahracters to compare. On 100 Mhz (let"s say 1 instruction 1 check) it is 5 us extra time on screen load.

> I understand desire to not use "all that cryptic tools in strange JS" and do all manually. BUT keeping file easy to write/modify for humans can add very serious unwanted restrictions...

Not "do all manually" but I think it's important to have manual way too. Scrips are for professional use manual editing for experimental usage or small projects. 

IMO a well-visible comment like "GENERATED FILE DO NOT EDIT" should be fine.

If the result C file is easy to understand/read by a human then I think we don't need two formats. The above-proposed format seems fulfilling good for both cases:
```c
{"msgid1", "translated text 1"},
{"msgid23", "translated text 23"},
{"msgid7", "translated text 7"},
```

> Seems that's too much for single issue. And wiki is not very convenient for such work. IMHO it worth to discuss repo name + global names, and create a separate repo.

OK, I can craete a new repo for it and give write access to you. Let"s call it simply `i18n`. We can change it anytime without any issues.

I don't stich with the `lv_` prefix. 

I can put together an example C language file to show my view about it.

puzrin, 2019-02-07:
> Scrips are for professional use manual editing for experimental usage or small projects.

What do you mean? Professional does not means complicated/difficult. If you can do high quality code with the same efforts, why not?

Also, gettext works out of box without translation (because mgs_id is already valid english phrase). What experimental usage to you mean? Just write code without translations, and it will be good enougth.

> IMO a well-visible comment like "GENERATED FILE DO NOT EDIT" should be fine.

That's not a problem at all. I think we will find acceptable solution. Let's discuss low level format later, i'd like to think about yaml/json files first.

> OK, I can create a new repo for it and give write access to you. Let"s call it simply `i18n`. We can change it anytime without any issues.

Agree. Please give me also temporary permissions to add collaborators to that repo, if possible.

seyyah, 2019-02-07:
It maybe contribute to the discussion [1]

```c
const char mesg1[] __attribute__((section (".USER_FLASH"))) 
             = "Ciao a tutti";
const char mesg2[] __attribute__((section (".USER_FLASH"))) 
             = "Riesco a sentire la mia mente va Dave";
```

## :clipboard: Reference
1. https://stackoverflow.com/questions/9820333/efficiency-of-gettext-in-memory-translation

kisvegabor, 2019-02-08:
@puzrin 

> What do you mean? Professional does not means complicated/difficult. If you can do high-quality code with the same efforts, why not?

I mean the user should be able to play with it without tools. Create 2 languages with 2 phrases and see how it works, what happens. Anyway, it seems possible with our current approach.

> Agree. Please give me also temporary permissions to add collaborators to that repo, if possible.

Done: https://github.com/littlevgl/i18n

@seyyah 
Interesting reading. It seems we are on the right way. :)

puzrin, 2019-02-08:
Close in favour of https://github.com/littlevgl/i18n/issues (moved to separate repo)

**All interested persons are encouraged to visit tracker and leave comments.** That's important. Talks in tracker are not difficult and less expensive than multiple code rewrites.

Note, i try to design the best possible solution for lvgl, with focus on usability and simplicity. But i have weak experience with C/C++ development process and can suggest stupid things sometime :). Kick me if you feel something goes wrong :).

closed 2019-02-08 by puzrin

puzrin, 2019-03-20:
> For advanced i18n an other thing still would be required: "text shaping". HarfBuzz is very common. It is used to render special scripts like Arabic. It handles the change of glyphs and adjusts letter spacing based on the special rules of the language's type system. Unfortunately, HarfBuzz can't be used in our case because it parses the ttf files directly and supposes the font renderer uses the ttf files. It's not for small embedded systems. It's would be a huge (even not reasonable) task implement it from scratch. What we can do is to add at least Right-To-Left and Arabic support.

Just for your info: https://github.com/foliojs/fontkit.

This guys implemented correct font rendering via JS with all unicode features support, and output via SVG backend (second-order bezier curves). I'd recommend to inspect if you ever decide to improve text output. Things are not trivial, but not very difficult too.

Fontkit is part of https://github.com/foliojs/pdfkit (battle-tested).

kisvegabor, 2019-03-21:
I doubt that we can use anything else than a library written in C directly for an embedded system. Anyway, I'll take look, thank you!

puzrin, 2019-03-21:
> I doubt that we can use anything else than a library written in C directly for an embedded system.

Sure, but you can check algorythms and total complexity.