lvgl/lvgl #792: Convert Micropython to JavaScript in order to create an online simulator

kisvegabor, 2019-02-02:
In continue of #687 

Discuss how to concert micropython to javascript to have an online simulator where people can easily try LittlevGL in their browser without downloading anything.

embeddedt, 2019-02-02:
@amirgon That port of Micropython builds properly but I ran into errors while trying to run it. I think that Emscripten has evolved since it was made.

amirgon, 2019-02-02:
@embeddedt Did you see [micropython-ports-wasm](https://github.com/pmp-p/micropython-ports-wasm)?  
It looks rather new and active (initial commit 3 days ago), @pmp-p claims he got Micropython working in WASM.

embeddedt, 2019-02-24:
I haven't had a lot of time to work on this lately, but I hope to get to it at some point.

kisvegabor, 2019-02-24:
Shall we "pin" this issue to leave it opened as a reminder?

embeddedt, 2019-02-24:
@kisvegabor Sure.

amirgon, 2019-03-13:
micropython/micropython#3575
Micropython now officially has an Emscripten port! micropython/micropython@7d675f3a17b435c4dc7880ff597ab81274721ae9 

lvgl+micropython could make a nice interactive lvgl demo on lvgl web page.

embeddedt, 2019-03-13:
@amirgon I'll see what I can do, but I have limited time right now.

kisvegabor, 2019-03-13:
@amirgon Awesome! It should be much easier with this.

kisvegabor, 2019-04-14:
@embeddedt Did you have a chance to make some experiments with Emscripten and Micropython?

embeddedt, 2019-04-14:
@kisvegabor I tried to work with it but I found that I don't understand Micropython well enough to be able to make it work with LittlevGL.

kisvegabor, 2019-04-15:
Would you like to warm it up? Probably @amirgon could help to understand the Mpy specific parts.
Or shall we postpone it? (It could be a great thing though but I understand if there are more important things now)

embeddedt, 2019-04-15:
I will try to experiment with it again this week.

kisvegabor, 2019-04-15:
Awesome!

embeddedt, 2019-04-19:
@amirgon I made some more progress on this, but it looks like Emscripten doesn't support SDL threads properly at the moment... could you figure out if the binding can be modified to avoid the need for another thread?

amirgon, 2019-04-20:
@embeddedt The micropython binding was designed to be single threaded. You can run everything (micropython+lvgl+display driver) on the same thread.  
I don't think you should use SDL at all. You probably need a special display driver (and input device driver) for Emscripten.  
How does Emscripten write to the screen and read mouse input? If you want I can help creating the drivers for it.

embeddedt, 2019-04-20:
@amirgon Emscripten uses SDL2 for graphics.

> The micropython binding was designed to be single threaded. You can run everything (micropython+lvgl+display driver) on the same thread.

Is `SDL_CreateThread` supposed to be called here then?

https://github.com/littlevgl/lv_binding_micropython/blob/da8588210df7e34644f94fe3bf605d70cb01058a/driver/SDL/modSDL.c#L32

amirgon, 2019-04-20:
> Is `SDL_CreateThread` supposed to be called here then?

@embeddedt The only purpose of the SDL thread over there is to call `lv_tick_inc` and `lv_task_handler` indirectly, using `mp_sched_schedule`.  
This will schedule `lv_task_handler` to run next time micropython is ready, on the same thread micropython is running.  
If you have other ways to schedule things in Emscripten (something like javascript's `setTimeout`), that would be good enough instead of SDL thread for calling these functions periodically.

embeddedt, 2019-04-20:
@amirgon Okay, I'll see what I can do.

amirgon, 2019-04-20:
@embeddedt How do you call SDL main loop (`monitor_sdl_refr_core`) if you don't use SDL threads?

embeddedt, 2019-04-20:
@amirgon I'm now using an Emscripten callback function to call the `lv_tick_inc`, `lv_task_handler`, and `monitor_sdl_refr_core` functions. I no longer use `mp_sched_schedule`.

amirgon, 2019-04-20:
> I no longer use `mp_sched_schedule`

@embeddedt `mp_sched_schedule` ensures that a function is called when micropython is ready, and not in the middle of the execution of some micropython statement.
Depending on how you schedule micropython main loop and how you schedule `lv_task_handler`, this might be ok (or not).

embeddedt, 2019-04-20:
I got it to work with `mp_sched_schedule`. Now I have to figure out how to get the REPL working so that people can put in their own code.

[image]

amirgon, 2019-04-20:
Nice progress @embeddedt ! :+1:

embeddedt, 2019-04-20:
@amirgon @kisvegabor Here's a very early prototype:

https://embeddedt.github.io/lv_micropython/ports/javascript/lvgl.html

amirgon, 2019-04-20:
> @amirgon @kisvegabor Here's a very early prototype:
> 
> https://embeddedt.github.io/lv_micropython/ports/javascript/lvgl.html

Very nice!

I think the REPL still has some work, to enable auto completions, history, and also print object representations (try `btn` vs `print(btn)`). These are very important for the user experience.
Maybe you could have a look at the REPL of the unix port (or any other port actually).

embeddedt, 2019-04-21:
@amirgon I did some work on the REPL - now it uses the standard Micropython REPL interface instead of a custom JavaScript one. Object representations are now shown correctly.

amirgon, 2019-04-21:
> @amirgon I did some work on the REPL - now it uses the standard Micropython REPL interface instead of a custom JavaScript one. Object representations are now shown correctly.

I can now see Object Representations, nice work @embeddedt !  

A few problems I still see:
 
- History is not working well (up key goes back in history, but down key doesn't go forward)
- Completions are not working. For example, In micropython REPL, writing `lv.btn.set_` and clicking <kbd>TAB</kbd> would show all possible completions to `lv.btn.set_` such as `lv.btn.set_pos`, `lv.btn.set_size` etc. This is really useful for the user!
- While you added an HTML button for <kbd>CTRL</kbd>+<kbd>E</kbd>, it's really useful only when you can paste multiple lines, not a single line. On your REPL it only works for a single line, not for pasting a code snippet which consists of multiple lines.
- It's a bit slow. I'm not sure how bad it is yet, but just try to run `dir(lv)`. On my old machine the browser froze for about 8 seconds before showing the result. (On the unix port, for example, the result is shown immediately). I'm not sure it's really a problem, since micropython is really inefficient in the way it implements `dir` (it actually tries to match every possible string it's aware of through string interning), but is more efficient on other things. Need to try Emscripten with a real micropython program.
- Is it possible to support the keyboard shortcuts (such as <kbd>CTRL</kbd>+<kbd>E</kbd>) instead of (or in addition to) HTML buttons?
- Is it possible to upload a script? It's not really practical to type into the input box more than a few lines...

embeddedt, 2019-04-21:
> A few problems I still see:
> 
>     * History is not working well (up key goes back in history, but down key doesn't go forward)

That's supposed to work. I will fix that.

>     * Completions are not working. For example, In micropython REPL, writing `lv.btn.set_` and clicking TAB would show all possible completions to `lv.btn.set_` such as `lv.btn.set_pos`, `lv.btn.set_size` etc.

I'm not sure if the browser lets me capture the TAB key.. if it does I will implement that.

>     * While you added an HTML button for CTRL+E, it's really useful only when you can paste multiple lines, not a single line. On your REPL it only works for a single line, not for pasting a code snippet which consists of multiple lines.

Yes; that's not particularly useful in my implementation. I will probably get rid of that.

>     * It's a bit slow. I'm not sure how bad it is yet, but just try to run `dir(lv)`. On my old machine the browser froze for about 8 seconds before showing the result.

I noticed that as well. I'm suspecting that the high rate of SDL refresh calls might be the culprit.

>     * Is it possible to support the keyboard shortcuts (such as CTRL+E) instead of (or in addition to) HTML buttons?

They are supported, *on some browsers*. It seems that there's no consistency across which shortcuts can be intercepted by JavaScript, so for compatibility reasons I provided the buttons as well. I found that in Firefox `Ctrl+E` and `Ctrl+C` would work but `Ctrl+D` couldn't be intercepted (it bookmarks the site instead).

>     * Is it possible to upload a script? It's not really practical to type into the input box more than a few lines...

That's the next thing I was planning to implement.

amirgon, 2019-04-21:
> I'm not sure if the browser lets me capture the TAB key.. if it does I will implement that.

@embeddedt I think that autocompletion is very important. If you can't capture TAB, then by some other keyboard shortcut (although it's best to keep the same conventions as the standard micropython REPL and use TAB, if possible).  

Maybe you could even show a list of possible completions interactively while the user types! This could be very cool.

embeddedt, 2019-04-21:
> Maybe you could even show a list of possible completions interactively while the user types! This could be very cool.

That would require piping some type of structure from Micropython back out to JavaScript - something I'd like to avoid for simplicity reasons.

But I will get autocompletion working in some form.

kisvegabor, 2019-04-23:
@embeddedt 
Awesome progress! COngratulation!

To execute `help(lv)` to make 5-10 seconds as well. However `lv.btn(lv.scr_act())` and `lv.obj_set_width()` was pretty fast.

embeddedt, 2019-04-23:
Thanks. I plan to get some more work done on it this weekend.

embeddedt, 2019-05-09:
@amirgon @kisvegabor I will be quite busy until the beginning of June. I have not had and don't think I will have any time to work on this until then.

kisvegabor, 2019-05-10:
@embeddedt 
No problem. It's fine to deal with it after v6.0 as well.

embeddedt, 2019-06-01:
@amirgon @kisvegabor The latest update removes the three buttons, and implements standard MicroPython pasting and autocompletion.

https://embeddedt.github.io/lv_micropython/ports/javascript/lvgl.html

At this point I think the main things left to do are to clean up the code, switch to LittlevGL version 6.0, and possibly adjust the page layout.

amirgon, 2019-06-01:
Very nice @embeddedt! :+1:  
Now it really feels like a micropython console, with syntax completions, keyboard shortcuts etc.  

I tried running the [`advanced_demo.py`](https://gist.github.com/amirgon/7f2a0702932a8d08611241d10c1f0e57) (a modified version of [this](https://github.com/littlevgl/lv_binding_micropython/blob/c910795ac3a32991793187c22e3325b4c27dc80b/gen/advanced_demo.py) after removing driver registration code).  
If you want to try it, click <kbd>CTRL</kdb>+<kdb>E</kbd>, paste the script and click <kbd>CTRL</kdb>+<kdb>D</kbd> 
(could be useful to add file upload, btw, instead of pasting scripts)

It is a bit slow.
You can feel it in the animations and the event responsiveness, but still very useful even as it is!  

I wonder if it's possible to improve performance. I noticed there's a warning on the js console saying:
```
Looks like you are rendering without using requestAnimationFrame for the main loop.
You should use 0 for the frame rate in emscripten_set_main_loop in order to use requestAnimationFrame, as that can greatly improve your frame rates!
Browser_mainLoop_runner @ lvgl_mp.js:78
```

Could it help to either use `requestAnimationFrame` or set frame rate to 0?

embeddedt, 2019-06-01:
> Could it help to either use `requestAnimationFrame` or set frame rate to 0?

I don't think so. This is relevant for games, where they are constantly redrawing the screen, but I don't think it matters as much for our GUI library.

I will have a look at the performance tools in the browser and see if I can find anything.

embeddedt, 2019-06-01:
I just found that there was a discrepancy between the tick rate expected on the C end and the actual tick rate being provided by JavaScript. This was causing animations to be severely delayed.

amirgon, 2019-06-01:
@embeddedt That's better!
Now the animation is faster, but there is still a small issue with event responsiveness.  

When clicking on the sytle selection dropdown, for example, if the click is too short, it is ignored. You need to hold the mouse button for a short moment (instead of just "click" it) in order to make it work.  

It may be related to the way the SDL driver is implemented. If it just checks N times a second it the mouse button is down, it might miss a short click.

embeddedt, 2019-06-01:
> It may be related to the way the SDL driver is implemented. If it just checks N times a second it the mouse button is down, it might miss a short click.

It uses SDL events, which are queued. Thus, regardless of the frequency at which events are checked for, I believe that the mouse event would still be processed at some point.

I think this might be an Emscripten issue, and not a LittlevGL (or MicroPython) one.

I did try increasing the frequency at which events are checked for from 50ms to 20ms but that made no visible difference (to me).

amirgon, 2019-06-01:
> I think this might be an Emscripten issue, and not a LittlevGL (or MicroPython) one.

@embeddedt, that's weird. Emscripten is used for games and losing a mouse click in a game is a serious problem. So I'm very skeptic it's an Emscripten issue.

In fact - I can see the same problem on the unix simulator! (when trying very short mouse "clicks") Are you sure events are queued on the SDL driver?

kisvegabor, 2019-06-02:
@embeddedt 
Amazing! The framerate is still a little bit low at me but it's absolutely usable. The autocompletion is working well too.

Actually, the slow input device is a LittelvGL (or rather a drive implementation) issue. Although SDL buffers the events and passes all of them to [mouse_handler](https://github.com/littlevgl/lv_drivers/blob/master/indev/mouse.c#L68) LittlevGL will read the last state only in every [LV_INDEV_READ_PERIOD](https://github.com/littlevgl/lvgl/blob/master/lv_conf_templ.h#L107). So it possible that a mouse click is overwritten by the release in `mouse_handler` until LittlevGL reads it.
To solve it
1. The events can be buffered in `mouse.c`. Possible and supported but will increase complexity, therefore, reduce readability (which would be important in an example driver)
2. Lower `LV_INDEV_READ_PERIOD` and [SDL_REFR_PERIOD](https://github.com/littlevgl/lv_drivers/blob/master/display/monitor.c#L28) and increase the call rate of `lv_task_handler`. It should also help a lot and worth to try first IMO.

amirgon, 2019-06-02:
> 1. The events can be buffered in `mouse.c`. Possible and supported but will increase complexity, therefore, reduce readability (which would be important in an example driver)

@kisvegabor  what if the driver, instead of buffering events, would just count missed "clicks" (press followed by release)? When `mouse_read` is called it will check how many clicks it missed and replay them. This will only require managing a "clicks" counter.

kisvegabor, 2019-06-02:
@amirgon 
> what if the driver, instead of buffering events, would just count missed "clicks" (press followed by release)? When mouse_read is called it will check how many clicks it missed and replay them. This will only require managing a "clicks" counter.

What about the coordinates? Use the first coordinate for the whole period?

amirgon, 2019-06-02:
> What about the coordinates? Use the first coordinate for the whole period?

@kisvegabor Yes. Or the last coordinate.
If the `mouse_read` is called frequently, the user might double or tripple click fast enough in between these calls, but not move the mouse to an exact position, so I think using the last coordinates could be good enough.

kisvegabor, 2019-06-02:
It sounds to like an acceptable limitation. However, I'm still trying to add extra complexity only if required. 
It might be platform dependent (so might be worse on your machine/browser/OS) but I noticed the issue only if I clicked very quickly and it wasn't disturbing for me.
 
Anyway, @embeddedt Could you test with higher call rates as described [here](https://github.com/littlevgl/lvgl/issues/792#issuecomment-498018097)?

embeddedt, 2019-06-02:
@kisvegabor I am currently calling `lv_task_handler` every 20ms. I can try lowering `LV_INDEV_READ_PERIOD`, but right now I think it's either 30 or 50 ms, so there isn't much room to lower it further.

amirgon, 2019-06-03:
@kisvegabor we can change the driver only on `lv_binding_micropython/driver/SDL/`. It is not used as a reference to the user, only internally for micropython, so I see no harm in putting a little more complexity into it.    

What do you think?

kisvegabor, 2019-06-03:
> we can change the driver only on lv_binding_micropython/driver/SDL/. It is not used as a reference to the user, only internally for micropython, so I see no harm in putting a little more complexity into it.

Perfect!

amirgon, 2019-06-03:
@embeddedt I've pushed [a fix](https://github.com/littlevgl/lv_binding_micropython/commit/fee418992f96f9e9a431fe2a98bb616cf18d9162) to the SDL driver of Micropython.  
It solves the "lost mouse clicks" problem on the unix port.

Could you add this to the Emscripten port? I expect it to solve the problem there as well.

embeddedt, 2019-06-04:
@amirgon It appears to have fixed the problem: https://embeddedt.github.io/lv_micropython/ports/javascript/lvgl.html

amirgon, 2019-06-04:
Thanks @embeddedt, looks good!

kisvegabor, 2019-06-04:
As I see the only missing thing is to switch to `v6.0`, right?

embeddedt, 2019-06-04:
@kisvegabor Yes. I am planning to switch it very soon.

amirgon, 2019-06-04:
I have two new ideas for the Emscripten port:

- Accept a URL parameter which will be passed as a **path to a script to execute**.  
  For example (something like): 

  `https://embeddedt.github.io/lv_micropython/ports/javascript/lvgl.html?https://gist.githubusercontent.com/amirgon/7f2a0702932a8d08611241d10c1f0e57/raw/24c5b5308af500128664d65abb45fdf4ac5c6ddf/advanced_demo.py` 

  In this example the URL would make the Emscripten port execute `advanced_demo.py` when the web page is loaded.  
This is useful in a forum or a github issue when someone wants to illustrate some problem, for example. He could publish a URL that in a single click would load micropython and execute his script on the web for anyone to see.

- GitHub is well integrated with Travis for CI (for free), such that every commit would make Travis build the project.  
But Travis could be used to do more than that - **it could build a micropython Emscripten port and deploy it on GH pages automatically on every commit!**  
So, without doing anything manually (such as downloading, building or deploying), we could run our code on the latest version of lvgl just by clicking a URL.  
Here are examples of running Emscripten by Travis automatically: 
  - https://emscripten.org/docs/compiling/Travis.html
  - https://github.com/jirutka/emscripten-travis-example  

  It's also possible to make Travis automatically deploy gh-pages without checking-in the credentials to the repo! Here is how it's done:
  - https://gist.github.com/domenic/ec8b0fc8ab45f39403dd

What do you think?

embeddedt, 2019-06-04:
> Accept a URL parameter which will be passed as a **path to a script to execute**.

This one should be fairly easy to implement. I will try it today.

embeddedt, 2019-06-04:
I have implemented that support. Here is a link to test it:

https://embeddedt.github.io/lv_micropython/ports/javascript/lvgl.html?script=https://gist.githubusercontent.com/amirgon/7f2a0702932a8d08611241d10c1f0e57/raw/24c5b5308af500128664d65abb45fdf4ac5c6ddf/advanced_demo.py

It only supports running a single script at the moment. I can easily extend it if necessary.

embeddedt, 2019-06-04:
@amirgon I will look at Travis once I get dev-6.0 working. I am having issues with the binding generator.

I switched `lib/lv_bindings` to the `dev-6.0` branch, updated submodules, and tried to compile, but I got this error (verbosity enabled):

```c
python3 ../../lib/lv_bindings/gen/gen_mpy.py -X anim -X group -X task -E build/lvgl/lvgl.pp.c ../../lib/lv_bindings/lvgl/lvgl.h > build/lvgl/lv_mpy.c
usage: gen_mpy.py [-h] [-I <Include Path>] [-D <Macro Name>]
                  [-E <Preprocessed File>]
                  input [input ...]
gen_mpy.py: error: unrecognized arguments: -X -X group -X task ../../lib/lv_bindings/lvgl/lvgl.h
../../py/py.mk:162: recipe for target 'build/lvgl/lv_mpy.c' failed
```

amirgon, 2019-06-04:
@embeddedt You need to integrate some changes from lv_micropython.
Both lv_micropython_binding and lv_micropython have dev-6.0 branch.
The error you are getting is probably related to [this](https://github.com/littlevgl/lv_micropython/commit/b83a5c0ae8ff702d9a370acc2aec122b5c7e5102) change.

embeddedt, 2019-06-04:
Hmm. I tried that and now I get this error:

```c
Traceback (most recent call last):
  File "../../lib/lv_bindings/gen/gen_mpy.py", line 1412, in <module>
    gen_obj(obj_name)
  File "../../lib/lv_bindings/gen/gen_mpy.py", line 1334, in gen_obj
    gen_mp_func(method, obj_name)
  File "../../lib/lv_bindings/gen/gen_mpy.py", line 1259, in gen_mp_func
    try_generate_type(func.type.type)
  File "../../lib/lv_bindings/gen/gen_mpy.py", line 1077, in try_generate_type
    if isinstance(type_ast, basestring): raise SyntaxError('Internal error! try_generate_type argument is a string.')
NameError: name 'basestring' is not defined
```

I would try to fix it myself but I don't know anything about how your script works.

[390 more replies]

closed 2019-12-22 by embeddedt