lvgl/lvgl #794: suggestion: modify root folders structure

puzrin, 2019-02-03:
Currently root become too "fat", it mixes code-related things (`lv_`), and process-related things (docs, tests, maintenance scripts & configs...)

I'd suggest to reorganize folders to more "common" way, something like this:

```
/docs
/src
/test
/support (scripts like lv_conf_checker.py)
```

kisvegabor, 2019-02-04:
Sounds reasonable to me. 

The key thing is to make lvgl's setup as simple as possible i.e. make sure it works without any build/project settings. Now you have to only:
1. clone the repo
2. add lv_conf.h next to the lvgl folder

@embeddedt what do think about it?

I'm not sure when should we add this change. Now, at the beginning of `dev-6.0` or at the end of it.
Maybe now is better.

embeddedt, 2019-02-04:
@kisvegabor @puzrin I think the overall idea is a good one, but we should make sure that (as @kisvegabor said) LittlevGL is usable without needing to configure include directories.

puzrin, 2019-02-04:
Why something should stop working from such change? Only internal relative path will be affected.

For user everything should stay the same, if he did not tried to include some nested files directly.

embeddedt, 2019-02-04:
@puzrin In another issue you had suggested that we remove the `lv_` prefix from folder names as well. That could cause issues in cases like this:

https://github.com/littlevgl/lvgl/blob/fb1a7fc0fc4857cd3eae24f35d36bca5fa5fab46/lv_draw/lv_draw.h#L22-L23

This kind of include setup is used in header files throughout the library.

puzrin, 2019-02-04:
Of cause, paths INSIDE lvgl files should be updated, but those are exact, and shoud not fail.

Problem can be only if you include something via search path. In this case name can become less unique. But as far as i know, only "lv_conf.h" is included such way, and i suggested to remove `lv_` prefixes for directories only.

kisvegabor, 2019-02-06:
The proposed solution really won't break the current behavior. I mentioned it to show what are the limits where we can move.
 
So having an `src` folder is a good idea! 

However, we should leave the `lv_` prefix of folders because:
- it's clearer. (although the file name would make everything fully clearer)
- auto-complete can show the relevant folders after typing `lv_`
- the directory structure is very simple (no subdirs) so we would save only 3 letters

puzrin, 2019-02-06:
Ok. Let's split my proposal to parts:

1. Move `lv_` folders to `src`. Seems no objections.
2. Rename `lv_*` folders. I can not insist, because have not enougth experience in C programming. IMHO, this is not principle in long term and will not cause any maintenance issues.

Moving code folders to `src` is enougth.

kisvegabor, 2019-02-06:
Great 👍

kisvegabor, 2019-02-27:
I'll do it soon.

kisvegabor, 2019-03-17:
I updated the project structure: https://github.com/littlevgl/lvgl/tree/dev-6.0

I left `lvgl.h` and `lvgl.mk` in root for backward compatibility.

Any comments?

puzrin, 2019-03-17:
LGTM

kisvegabor, 2019-03-17:
Great, thank you! Then I close this issue.

closed 2019-03-17 by kisvegabor

puzrin, 2019-03-17:
Seems i missed one small thing. It would be more convenient to see [template](https://github.com/littlevgl/lvgl/blob/dev-6.0/src/lv_conf_templ.h) on root level. Also,  may be rename to `lv_conf_template.h`.

kisvegabor, 2019-03-18:
Good idea. Done here: https://github.com/littlevgl/lvgl/tree/dev-6.0/templates

puzrin, 2019-03-18:
I meaned a bit different things.

- `/lv_conf.template.h` - in root, because this file is mandatory.
- other files seems about ports, not mandatory. Probabbly beter in `/docs/drivers` or `docs/porting` or something like this.

kisvegabor, 2019-03-18:
It's really reasonable to move `lv_conf_template.h` to root.

I added the porting templates to `porting` folder to make them easier to find.

puzrin, 2019-03-18:
> I added the porting templates to porting folder to make them easier to find.

You just cleared root from previous "easier to find" folders :).

manison, 2019-03-19:
I just want to note that there is a typo in the root directory name _scrip**t**s_.

kisvegabor, 2019-03-20:
@manison Thank you I fixed them

@puzrin I think the current directory is clear in it's easy to find things in it. What are your suggestions to improve it further?

puzrin, 2019-03-20:
> What are your suggestions to improve it further

`/porting` => `/docs/porting`. Current root in not complete yet. It will be extended with `/test`, `.clang_format` and may be something other.

kisvegabor, 2019-03-20:
> /porting => /docs/porting. 

It's better to make `porting` more visible as they can help a lot to the user. Inside `docs` it's less probable that they will find it. 

>  It will be extended with /test

Yes, once we will have teste :)

> .clang_format and may be something other.

Now, we have`astlye`. Is`clang_format`  better?

puzrin, 2019-03-20:
> Now, we have `astlye`. I `sclang_format` better?

I'm not experienced enougth, to give qualified answer. At first glance, those seems similar. Seems both are available under ubuntu in packages.

Here is interesting discussion https://github.com/dealii/dealii/issues/6686 (big project migrated to `Clang-Format`)

---

Checked current astyle integration - seems not enougth

- Too few params to define coding style (and not based on existing CS).
- No makefile to automate things. Rely on manual run will not work.

C47D, 2019-03-21:
@puzrin some time ago I was interested in adding a git hook to run clang-format on the pre-commit phase, see (here) https://github.com/littlevgl/lvgl/issues/71].  Maybe you can make it work as I wasn't able to do so.

puzrin, 2019-03-21:
Hooks are local. Someone may forget to setup it. Here required something like makefile for manual run + travis-ci check that nothing needs modification.

C47D, 2019-03-21:
Maybe the project collaborators kisvegabor and embeddedt can run it before merge PRs, I think there's a hook for that too.

kisvegabor, 2019-03-22:
Personally, it wouldn't be a problem to periodically run the `formatter` becaue we don't have so much PRs. In addition, PRs are added to dev branch so it's not a problem to have unformatted code remporaly. 

My problem is that (and that's why I didn't run astlye for a while) I can't configure astlye well enough. E.g. 
```c
if(x == 0) {
  x++;
}
```
is fine, but I don't want `{` in the same line when:
```c
if(x == 0 &&
   y == 0)
{
  x++;
}
``` 

Anyway, if are interested in discussing the "formatters", please open a new topic it.