lvgl/lvgl #796: [lvgl v5.3] How to enable the bare minimum functionality? (or lvgl taking too much flash?)

C47D, 2019-02-04:
Hi,

I'm trying to make lvgl v5.3 run in my microcontroller (PSoC CY8C5888LTI-LP097, Flash: 256kB, SRAM: 64kB, Stack size: 0x600, Heap size: 0x80), i do want to use it with OLED SSD1306 128*64.

This is my lv_conf.h file:
```c
/**
 * @file lv_conf.h
 *
 */

/*
 * COPY THIS FILE AS lv_conf.h
 */

#if 1 /*Set it to "1" to enable content*/

#ifndef LV_CONF_H
#define LV_CONF_H
/*===================
   Dynamic memory
 *===================*/

/* Memory size which will be used by the library
 * to store the graphical objects and other data */
#define LV_MEM_CUSTOM      0                /*1: use custom malloc/free, 0: use the built-in lv_mem_alloc/lv_mem_free*/
#if LV_MEM_CUSTOM == 0
#  define LV_MEM_SIZE    (64U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#  define LV_MEM_ATTR                         /*Complier prefix for big array declaration*/
#  define LV_MEM_ADR          0               /*Set an address for memory pool instead of allocation it as an array. Can be in external SRAM too.*/
#  define LV_MEM_AUTO_DEFRAG  1               /*Automatically defrag on free*/
#else       /*LV_MEM_CUSTOM*/
#  define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
#  define LV_MEM_CUSTOM_ALLOC   malloc       /*Wrapper to malloc*/
#  define LV_MEM_CUSTOM_FREE    free         /*Wrapper to free*/
#endif     /*LV_MEM_CUSTOM*/

/* Garbage Collector settings
 * Used if lvgl is binded to higher language and the memory is managed by that language */
#define LV_ENABLE_GC 0
#if LV_ENABLE_GC != 0
#  define LV_MEM_CUSTOM_REALLOC   your_realloc           /*Wrapper to realloc*/
#  define LV_MEM_CUSTOM_GET_SIZE  your_mem_get_size      /*Wrapper to lv_mem_get_size*/
#  define LV_GC_INCLUDE "gc.h"                           /*Include Garbage Collector related things*/
#endif /* LV_ENABLE_GC */

/*===================
   Graphical settings
 *===================*/

/* Horizontal and vertical resolution of the library.*/
#define LV_HOR_RES          (128)
#define LV_VER_RES          (64)

/* Dot Per Inch: used to initialize default sizes. E.g. a button with width = LV_DPI / 2 -> half inch wide
 * (Not so important, you can adjust it to modify default sizes and spaces)*/
#define LV_DPI              100

/* Enable anti-aliasing (lines, and radiuses will be smoothed) */
#define LV_ANTIALIAS        1       /*1: Enable anti-aliasing*/

/*Screen refresh period in milliseconds*/
#define LV_REFR_PERIOD      30

/*-----------------
 *  VDB settings
 *----------------*/

/* VDB (Virtual Display Buffer) is an internal graphics buffer.
 * The GUI will be drawn into this buffer first and then
[... 315 lines cut ...]
 * Non-user section
 *************************/
#if defined(_MSC_VER) && !defined(_CRT_SECURE_NO_WARNINGS)    /* Disable warnings for Visual Studio*/
#  define _CRT_SECURE_NO_WARNINGS
#endif

/*--END OF LV_CONF_H--*/

/*Be sure every define has a default value*/
#include "lv_conf_checker.h"

#endif /*LV_CONF_H*/

#endif /*End of "Content enable"*/
```

I'm trying to enable the bare minimum lvgl functionality (monospace font and labels) but i'm getting the following compiler error:
[image]

It's the first time i get this kind of error and tried reducing the heap size to solve it without success, i must be doing something silly as i think lvgl can fit on my mcu flash without trouble, but it is not, does anybody sees any error on my conf file?

I haven't added any application code tho.

Regards,
- Carlos

C47D, 2019-02-04:
Well, i think i found the root cause, i changed the line
```c
#  define LV_MEM_SIZE    (64U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
```
to:
```c
#  define LV_MEM_SIZE    (32U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
```

Is this the 'right' way to solve it?

embeddedt, 2019-02-04:
@C47D That error is not referring to your flash, it's saying that there's too much RAM usage. Try making your VDB size `10 * LV_HOR_RES`.

C47D, 2019-02-04:
Thanks for the quick reply @embeddedt,
Thanks for the confirmation, reducing `LV_MEM_SIZE` to `(30 * 1024)` indeed reduce the Flash and SRAM consumption (Flash: 7.8%, SRAM: 56.3%). Setting `LV_VDB_SIZE` to `LV_HOR_RES * 10` increase it just by a little bit (Flash: 7.8%, SRAM: 56.9%).

I need to get more familiar to lvgl 👍 

Regards,
Carlos

embeddedt, 2019-02-04:
@C47D Once you've built your application you can adjust `LV_MEM_SIZE` to make it more accurate for your needs. Or if your platform already has a heap system you can use that by enabling `LV_MEM_CUSTOM`.

C47D, 2019-02-04:
@embeddedt Thanks, i was about to ask you about more ways to save memory within the library but you are right, that should be done at the end of the development. Should i close this issue? The main question was (quickly) solved :).

embeddedt, 2019-02-04:
@C47D If the main issue is solved usually we recommend that you close the issue and open a new one if you have another question.

One way to easily cut down on memory usage is to try and re-use your graphical objects instead of creating many. This reduces the amount of heap space you need in total.

C47D, 2019-02-04:
> One way to easily cut down on memory usage is to try and re-use your graphical objects instead of creating many. This reduces the amount of heap space you need in total.

Thanks for the tip, i will try it out.

closed 2019-02-04 by C47D