lvgl/lvgl #797: custom tick header

ruslanproject, 2019-02-05:
good day.

right now in 5.3:
#define LV_TICK_CUSTOM_INCLUDE  "sonething.h"         /*Header for the sys time function*/

it is not correct, bucase if I use system tick without need include any file - it is not work.

May be change to:
#define LV_TICK_CUSTOM_INCLUDE  #include "sonething.h"         /*Header for the sys time function*/

and in "lv_hal_tick.c":

#if LV_TICK_CUSTOM == 1
LV_TICK_CUSTOM_INCLUDE
#endif

This is a more universal solution.

embeddedt, 2019-02-05:
@ruslanproject Another way would be to just include a standard C header like `string.h` (which should have no effect).

kisvegabor, 2019-02-06:
Using a system header as @embeddedt suggested should be fine!

I think we should use a system header in the example too (but definitely not something with a typo like now 😅) I vote for `<stdint.h>` just because it's more common in the library then `<string.h>`

ruslanproject, 2019-02-06:
use any include file not good idia, it is not clean for understand
if I not need any special include for tick function, use any "standart" include is irrational

kisvegabor, 2019-02-06:
Then `lv_hal_tick.c` can be modified like this:
```c
#if LV_TICK_CUSTOM_INCLUDE
#  include LV_TICK_CUSTOM_INCLUDE
#endif
```

Is it working well?

ruslanproject, 2019-02-06:
it is well, but problem when update
I do not pretend to my decision, this is just a solution that is understandable and without any hacks.

kisvegabor, 2019-02-06:
If it works well than it can be added to the library. This way it will be fully backward compatible.

closed 2019-03-06 by stale[bot]