lvgl/lvgl #798: spinbox issue

ScarsFun, 2019-02-05:
Hi,
I'm testing spinbox object with stm32 and ili9341, but results are the same with simulator.
In spinbox digit count and decimal format settings lv_spinbox_set_digit_format:
setting "separator_position" to 0, decimal point is always shown and cursor don't wraps from last digit to first.

kisvegabor, 2019-02-06:
@AloyseTech can you have look at it?

kisvegabor, 2019-02-12:
@ScarsFun 
I pushed some fixes for spinbox. It is working well now?

AloyseTech, 2019-02-12:
Hi guys,
Sorry I’m not able to help right now. I have a lot of work on my side.

kisvegabor, 2019-02-13:
@AloyseTech No problem, thank you for the feedback!

ScarsFun, 2019-02-13:
by setting "separator_position" to 0 decimal point is no more showed but cursor stay fixed an less significative digit.
There is a problem with ranges too: by setting, for example, min to -99 and max to 999, I'm able to set the most significative digit to -1 turning the encoder with cursor on hundreds digit and decreasing from a positive value.

kisvegabor, 2019-02-15:
I reviewed the spinbox's core code and made some updates and fixes. I tested with several different settings and it seemed good to me.

The updates are in the `master` branch.

ScarsFun, 2019-02-15:
Works well now.
Thanks!

kisvegabor, 2019-02-15:
Awesome!

closed 2019-02-15 by kisvegabor

AloyseTech, 2019-02-15:
Thanks @kisvegabor for dealing with this